Hieromon/AutoConnect #209: Credentials in EEPROM overwritten by custom settings

TKTF50, 2020-04-21:
Hi Hieromon,

First, thanks for creating this library!

I'm using the boundaryOffset setting to leave space for my own configuration settings in EEPROM, but I've run into a problem. What I found is that the AutoConnect credentials get erased if I update my configuration settings. In other words, here is the sequence of steps (using an AutoConnect portal with custom web pages):

1. Write custom configuration settings to EEPROM.
2. Write credentials to EEPROM.

At this point, all of the settings can be read from EEPROM successfully.

3. Go back into the portal and update the custom configuration settings in EEPROM.

Now the credentials, written in step 2, have been erased.

It took me a while to track down that the EEPROM library for ESP8266 erases the entire flash sector when it writes to any part of the sector. This isn't a problem for AutoConnect because it knows how much custom data to preserve (because of the boundaryOffset setting). But I also have to manually preserve the AutoConnect data whenever I update my custom settings.

As a workaround, I added a new method to `AutoConnectCredentialBase` that returns the total size of the space required for AutoConnect's credential storage:

`  virtual uint16_t dataSize(void) { return sizeof(AC_IDENTIFIER) - 1 + sizeof(uint8_t) + sizeof(uint16_t) + _containSize; }`

In my function to store my settings, I then only have to reserve enough extra buffer space to preserve the AutoConnect data:

```
  struct CustomConfig myConfigStruct;
  AutoConnectCredential credentials(sizeof(myConfigStruct));
  EEPROM.begin(sizeof(myConfigStruct) + credentials.dataSize());
  EEPROM.put(0, myConfigStruct);
  bool wasWritten = EEPROM.commit();
  DEBUG_PRINTF(PSTR("Save %s\n"), wasWritten ? "OK" : "FAILED");
  EEPROM.end();
```

Maybe there is something I misunderstood about how custom EEPROM settings are supposed to be managed with the AutoConnect library. If so, please let me know the proper method for preserving the AutoConnect credentials when I update my own settings.

Otherwise, would you consider making a method like the one I proposed above part of the official library?

Thanks.

Hieromon, 2020-04-22:
@TKTF50 Thank you for using AutoConnect.
Exactly, your point is correct. ESP8266's EEPROM library considers the area as a pool, and the I/O unit is the pool unit. That is, the user sketch must handle the area of AutoConnectCredential along with own configuration. I'm dissatisfied with this. (ESP32 arduino core has solved this problem with the Preferences class)

Your workaround is realistic, versatile and applicable depending on the Sketch. I'll grab your solution into the next release as v1.2.0. Thank you for your contribution.

> Otherwise, would you consider making a method like the one I proposed above part of the official library?

I find the above is competent, but do you have other ideas?

TKTF50, 2020-04-22:
@Hieromon,

Thank you for the quick response.

I think your approach using `getEEPROMSize()` is better than what I suggested. But maybe it would be better to call the method `getEEPROMUsedSize()` since you are returning the amount of memory **used** by the configuration, not the total memory available.

Also, if this method does not work in the ESP32 environment, how about surrounding the declaration and implementation with `#if defined(ARDUINO_ARCH_ESP8266)`. I believe you are already using this in other places in your code (or maybe just in the sample code).

Arigatō

Hieromon, 2020-04-22:
@TKTF50,

I staged the patch for `getEEPROMUsedSize` on branch [patches/v120](https://github.com/Hieromon/AutoConnect/tree/patches/v120).
You can evaluate it.

TKTF50, 2020-04-22:
@Hieromon,

I tested the patch with my code and it works fine.

I looked at the new code and the example you created and I have a few suggestions, mostly about wording. Please see my attachment.

Thank you! :)
[getEEPROMUsedSize.txt](https://github.com/Hieromon/AutoConnect/files/4518705/getEEPROMUsedSize.txt)

Hieromon, 2020-04-23:
@TKTF50,

Thank you very much for your suggestions! It ’s more precious to me than PR.
Describing legible and accurate documentation is sometimes more important than code fulfillment. It can guide users to the desired place in the shortest distance. Without an accurate map, users may lose their way.
I'm sorry that the AutoConnect documentation is written in poor English, which discourages users from reading it. I' keenly aware of my lack of English skills but will continue to practice diligence English learning.

This topic will close automatically when the [patches](https://github.com/Hieromon/AutoConnect/tree/patches/v120) branch merged.
Thank you.

TKTF50, 2020-05-04:
Hi @Hieromon,

Actually, I think you've done a very good job with the documentation you created. The obvious effort you put into the documentation is one of the reasons I chose to use your library in my project.

If you'd like some help with the wording, I'd be willing to donate some time to it.

Hieromon, 2020-12-02:
Merged #280

closed 2020-12-02 by Hieromon