Hieromon/AutoConnect #210: AP+Station mode with AutoConnect.begin()

TKTF50, 2020-04-22:
Hi Hieromon,

Here's another question/suggestion for you:

I'm using your AutoConnect library with a project that uses the ESP-NOW communication protocol. The nodes (currently, all ESP8266's) all need to both send and receive messages using ESP-NOW. Some of the nodes also connect to the WLAN in my house. I've found that the two kinds of communication (ESP-NOW and WLAN) only work together if the nodes connect to the LAN in the combined AP+STA mode. I want to use your library to make the WLAN connection, but it switches to STA mode in the `begin` method. I've modified the begin method to use the `enableSTA` method instead of `mode` (see below) and this makes the library work for my purposes.

```
bool AutoConnect::begin(const char* ssid, const char* passphrase, unsigned long timeout) {
  bool  cs;

  // Overwrite for the current timeout value.
  _connectTimeout = timeout;

  // Start WiFi connection with station mode.
  WiFi.softAPdisconnect(true);
  cs = WiFi.enableSTA(true);		// >> changed from: WiFi.mode(WIFI_STA);
  AC_DBG(PSTR("Enable WIFI_STA: %s\n"), cs ? "Ok" : "Failed");
  delay(100);
```

Is there a reason why you chose to set the WiFi to only STA mode in `begin`?

I haven't found any problem with either the portal or just connecting to the network with this change but I'm only using ESP8266's. Also, since I use a button to control whether or not the module goes into configuration mode, I'm only testing the cases:

1. `autoRise = false, immediateStart = false` (connect to WLAN without allowing portal to start in the event of a failure)
2. `autoRise = true,  immediateStart = true`  (start portal without attempting to first connect to WLAN)

If the above code change is a problem, would you consider adding a new option to the library to allow AP+STA mode in the `begin()` method?

Thanks.

Hieromon, 2020-04-22:
@TKTF50 Hello,
> Is there a reason why you chose to set the WiFi to only STA mode in begin?

I intentionally inserted it but forgot why. :)
Before designing AutoConnect, I investigated the behavior of the ESP8266 using a [tool](https://github.com/Hieromon/ESPShaker) that can interactively operate. This is because the transition conditions for WIFI mode were ambiguous in the Espressif documentation.
Then I found that there are some conditions that do not enter the STA mode properly after booting depending on the mode and the WiFi connection state immediately before resetting the module.
But I forgot what it was and am attempting reproducing it now.

The other is that the ESP8266 keeps broadcasting the probes for soft APs. This has side effects.
As a result, ESP8266 may momentarily lose connectivity with the connected router. Then the SDK is secretly reconnecting it without raising an event.  It can be found in the trace from the core debug log. At that moment the return value of WiFistatus is not WL_CONNECTED, it will be WL_IDLE_STATE.

I wanted to lock the ESP module in the desired state before the AutoConenct::begin sequence started.
To be honest, I don't trust much APSTA mode. It will work fine for the uses Espressif envisions, especially for their provided protocols. (eg ESP-MESH etc) But if not, I think it's wise to avoid the situation of staying in APSTA mode as much as possible.
I hope AutoConnect is used by many people. But not everyone has a deep insight that you have. While some people overestimate SoftAP, so AutoConnect prioritized stability, my experience.

> If the above code change is a problem, would you consider adding a new option to the library to allow AP+STA mode in the begin() method?

Sure, please PR you have. Thank you for your contribution.

TKTF50, 2020-04-22:
@Hieromon Hi,

You caught me by surprise. I don't have a code change prepared and I've only ever used GIT to clone a repository. I've never created a PR (or even a repository). :)

But I'd be happy to give it a try. It will take me a few days (or more) but I guess there is no hurry.

closed 2020-12-02 by Hieromon

Hieromon, 2020-12-02:
Merged #280

commit e0334635 (2020-05-13): Merge pull request #217 from TKTF50/preserve-AP-Mode

Added AutoConnectConfig::preserveAPMode setting.
resolves #210
PR #220 (2020-05-13): Pr/217
Fixes #210