Hieromon/AutoConnect #211: GET on _ac cause error 104

dumarjo, 2020-04-22:
Hi,

Does anybody have this error when browsing to the _ac page ?

```
[D][WiFiClient.cpp:509] connected(): Disconnected: RES: 0, ERR: 128
[V][WebServer.cpp:286] handleClient(): New client
[V][Parsing.cpp:113] _parseRequest(): method: GET url: /_ac search: 
[V][Parsing.cpp:219] _parseRequest(): headerName: Host
[V][Parsing.cpp:220] _parseRequest(): headerValue: 192.168.2.12
[V][Parsing.cpp:219] _parseRequest(): headerName: Connection
[V][Parsing.cpp:220] _parseRequest(): headerValue: keep-alive
[V][Parsing.cpp:219] _parseRequest(): headerName: User-Agent
[V][Parsing.cpp:220] _parseRequest(): headerValue: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/81.0.4044.113 Safari/537.36
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept
[V][Parsing.cpp:220] _parseRequest(): headerValue: image/webp,image/apng,image/*,*/*;q=0.8
[V][Parsing.cpp:219] _parseRequest(): headerName: Referer
[V][Parsing.cpp:220] _parseRequest(): headerValue: http://evmodule/_ac/open
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Encoding
[V][Parsing.cpp:220] _parseRequest(): headerValue: gzip, deflate
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Language
[V][Parsing.cpp:220] _parseRequest(): headerValue: en-US,en;q=0.9,fr-CA;q=0.8,fr;q=0.7
[V][Parsing.cpp:247] _parseArguments(): args: 
[V][Parsing.cpp:230] _parseRequest(): Request: /_ac
[V][Parsing.cpp:231] _parseRequest():  Arguments: 
[E][WiFiClient.cpp:392] write(): fail on fd 61, errno: 104, "Connection reset by peer"
```
Jonathan

Hieromon, 2020-04-23:
It is the first phenomenon I see and has been cut off TCP connection from the browser.
There is a strange request in your posted HTTP header.
```
[V][Parsing.cpp:220] _parseRequest(): headerValue: image/webp,image/apng,image/*,*/*;q=0.8
```
AutoConnect sends HTML using `send` method of `WebServer` class, and the header generated by ESP32 Arduino core library is as follows. (It is from ESP32 DevKIT-C of mine with ESP32 arduino core v1.0.4 release)
```
[V][Parsing.cpp:220] _parseRequest(): headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
```
In the first header, the content is interpreted as an image with an unknown length. As a result, it seems that the connection has been disconnected from the browser. But your posted log alone can't identify anything.
If you need to investigate this behavior, please post the information needed to reproduce it following [this guide](https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository).

Hieromon, 2020-12-02:
Closed due to no activity

closed 2020-12-02 by Hieromon