Hieromon/AutoConnect #212: WebServer and esp_https_ota

DavidTruyens, 2020-04-25:
Hi,

I use the esp-https-ota in my project and would like to add AutoConnect. But I get an error when trying to add the WebServer:

`.platformio/packages/framework-arduinoespressif32/tools/sdk/include/nghttp/http_parser.h:102:6: error: redeclaration of 'HTTP_OPTIONS'`

The redeclaration is for all HTTP functions like:
* HPPT_PUT
* HTTP_HEAD
* HTTP_GET
* ...

Any suggestions? 

I like the update to use HTTPS as it's a critical component. If I would switch to the AutoConnectUpdate it'll be using HTTP, right?

Thanks,
David

Hieromon, 2020-04-26:
I had no idea to make WebServer class coexist with the ESP HTTP  (similarly for HTTPS） client component used by esp-https-ota.
Have you asked on [espressif/arduino-esp32](https://github.com/espressif/arduino-esp32)?

Hieromon, 2020-09-28:
Closed due to no activity.

closed 2020-09-28 by Hieromon