Hieromon/AutoConnect #214: MDNS Access in local browser.

hamidrasyid, 2020-05-01:
Hi, thanks for your very good library. It's very useful for my project.

I just want to ask, I used the combination with mDNS method to access Webserver home or AC config, so I don't have to remember the IP address when it connects to the Wifi in STA. Instead, I can used <mDNS>.local . I used this example:

`if (Portal.begin()) {
    if (MDNS.begin("esp32")) {
      MDNS.addService("http", "tcp", 80);
    }
  }`

But after I tested, if I directly access esp32.local in my browser, it can't connect. And if I tried to connect with the IP address, it works. After I tried with IP Address, and I tried again access esp32.local immediately, it works. After a while it back to cannot connect, and I must to access IP address first in order to got success in mDNS address too.

Thanks.

Hieromon, 2020-05-04:
I don't think AutoConnect is the cause of the problem.
1. Which host machine is running the browser?
2. If you use a sketch other than AutoConnect, can your host machine resolve the name with mDNS?

Hieromon, 2020-10-20:
Closed due to no activity for more than a month.

closed 2020-10-20 by Hieromon

Elton-JJ, 2024-06-07:
Hi, Thanks for the Library.

I'm facing the same issue too, been banging my head the whole night to get it figured out.

Been using this following code

`#include <WiFi.h>
#include <WebServer.h>
#include <AutoConnect.h>
#include <ESPmDNS.h>

WebServer Server(80);
AutoConnect Portal(Server);

void setup() {
  Serial.begin(115200);
  delay(100);
  
  if (Portal.begin()) {
    if (MDNS.begin("esp32")) {
      MDNS.addService("http", "tcp", 80);
      Serial.println("mDNS responder started");
      
    }
  }
}

void loop() {
  Portal.handleClient();
}`

I'm able to access the portal for LocalIP/_ac, and not at esp32.local

I'm on a Windows 11 system using Chrome browser and Arduino IDE 1.8.19 and running ESP32 core 2.0.17

The following example code works fine, using WiFi.h

`#include <WiFi.h>
#include <WebServer.h>
#include <ESPmDNS.h>

/*Put your SSID & Password*/
const char* ssid = "Airtel_Zerotouch";  // Enter SSID here
const char* password = "Airtel@123";  //Enter Password here

WebServer server(80);

void setup() {
  Serial.begin(115200);
  delay(100);

  Serial.println("Connecting to ");
  Serial.println(ssid);

  //connect to your local wi-fi network
  WiFi.begin(ssid, password);

  //check wi-fi is connected to wi-fi network
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected..!");
  
  // Initialize mDNS
  if (!MDNS.begin("esp32")) {   // Set the hostname to "esp32.local"
    Serial.println("Error setting up MDNS responder!");
    while(1) {
      delay(1000);
    }
  }
  Serial.println("mDNS responder started");
  
  server.on("/", handle_OnConnect);
  server.onNotFound(handle_NotFound);

  server.begin();
  Serial.println("HTTP server started");
}
void loop() {
  server.handleClient();
}

void handle_OnConnect() {
  server.send(200, "text/html", "<!DOCTYPE html><html><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1.0, user-scalable=no\"></head><body><h1>Hey there!</h1></body></html>"); 
}

void handle_NotFound(){
  server.send(404, "text/plain", "Not found");
}`

Thank You for taking out the time to respond.