Hieromon/AutoConnect #218: Compiling in Arduino IDE (1.8.12) I get following messages

msillano, 2020-05-09:


msillano, 2020-05-09:
Compiling in Arduino ID (1.8.12)  with esp8266\2.7.1 (d1_mini) I get the messages:
```
D:\User\Documents\progetti2020\arduino\libraries\AutoConnect\src\AutoConnect.cpp: In member function 'bool AutoConnect::begin(const char*, const char*, long unsigned int)':

D:\User\Documents\progetti2020\arduino\libraries\AutoConnect\src\AutoConnect.cpp:112:46: warning: suggest braces around empty body in an 'if' statement [-Wempty-body]

       AC_DBG("Current:%.32s\n", current.ssid);

                                              ^

In file included from D:\User\Documents\progetti2020\arduino\libraries\AutoConnect\src\AutoConnectAux.cpp:14:0:

D:\User\Documents\progetti2020\arduino\libraries\AutoConnect\src\AutoConnectElementBasisImpl.h: In member function 'bool AutoConnectFileBasis::attach(ACFile_t)':

D:\User\Documents\progetti2020\arduino\libraries\AutoConnect\src\AutoConnectElementBasisImpl.h:114:41: warning: 'SPIFFS' is deprecated (declared at C:\Users\Sillano\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266/FS.h:269): SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]

     handlerFS = new AutoConnectUploadFS(SPIFFS);

                                         ^

In file included from D:\User\Documents\progetti2020\arduino\libraries\AutoConnect\src\AutoConnectAux.cpp:16:0:

D:\User\Documents\progetti2020\arduino\libraries\AutoConnect\src\AutoConnectElementJsonImpl.h: In member function 'void AutoConnectElementJson::_setMember(const JsonObject&)':

D:\User\Documents\progetti2020\arduino\libraries\AutoConnect\src\AutoConnectElementJsonImpl.h:96:96: warning: suggest braces around empty body in an 'else' statement [-Wempty-body]

       AC_DBG("Warning '%s' loading, unknown posterior '%s'\n", name.c_str(), posterior.c_str());

                                                                                                ^

D:\User\Documents\progetti2020\arduino\libraries\AutoConnect\src\AutoConnectAux.cpp: At global scope:

D:\User\Documents\progetti2020\arduino\libraries\AutoConnect\src\AutoConnectAux.cpp:464:14: warning: unused parameter 'args' [-Wunused-parameter]

 const String AutoConnectAux::_insertStyle(PageArgument& args) {

              ^

```
They are only warning, but...
Best regards
m.s.

Hieromon, 2020-05-11:
The AutoConnect master is currently in the process of evaluating the ESP8266 arduino core 2.7.0 and does not support yet.
ESP8266 core recommends migrating from SPIFF to the LittleFS, but AutoConnect does not support LittleFS alone while maintaining backward compatibility. AutoConnect needs to branch the source code pass according to the core version at compile-time, but it does not support yet.
https://arduino-esp8266.readthedocs.io/en/2.7.1/filesystem.html?highlight=littleFS#spiffs-and-littlefs

Hieromon, 2020-12-02:
Merged #280

closed 2020-12-02 by Hieromon