Hieromon/AutoConnect #219: Using as offline setup tool with different pages to load/save elements

ArminPP, 2020-05-09:
Hello, I would like to use this great library as an setup tool only at (first) startup. 
In my application I have +30 elements which I would like to store and load from EEProm.
  
I use „Elements.ino“ as base and was able to include two or more pages, but I am totaly lost in saving/loading the elements in/from EEProm.
I was not able to save the elements from the 2nd page to EEProm, also loading from EEProm doesn’t work. (1st Page {PAGE_GLOBAL_SETTING} load/save is OK, but has some side effects to page 2…)

```
[static const char PAGE_GLOBAL_SETTINGS[] PROGMEM = R"(
{
  "uri": "/global",
  "title": "Global Settings",
  "menu": true,
  "element": [
    {
      "name": "isBATTERY",
      "type": "ACCheckbox",
      "value": "isBATTERY",
      "label": "Gerät im Batteriebetrieb",
      "labelposition": "infront",
      "checked": false
    },
      {
        "name": "lastline",
        "type": "ACElement",
        "value": "<hr><br><br>"
      },
    {
      "name": "load",
      "type": "ACSubmit",
      "value": "Load",
      "uri": "/global"
    },
    {
      "name": "save",
      "type": "ACSubmit",
      "value": "Save",
      "uri": "/save"
    },
    {
      "name": "adjust_width",
      "type": "ACElement",
      "value": "<script type=\"text/javascript\">window.onload=function(){var t=document.querySelectorAll(\"input[type='text']\");for(i=0;i<t.length;i++){var e=t[i].getAttribute(\"placeholder\");e&&t[i].setAttribute(\"size\",e.length*.8)}};</script>"
    }
  ]
}
)";
static const char PAGE_MPU[] PROGMEM = R"(
{
  "uri": "/MPU",
  "title": "MPU Settings",
  "menu": true,
  "element": [
    {
      "name": "MPU_2_FILTER",
      "type": "ACSelect",
      "option": [
        "OFF",
        "ON"
      ],
      "label": "dig. Tiefpass Filter [Hz]",
      "selected": 0
    },
      {
        "name": "lastline",
        "type": "ACElement",
        "value": "<hr><br><br>"
      },
    {
      "name": "load_1",
      "type": "ACSubmit",
      "value": "Load_1",
      "uri": "/MPU"
    },
    {
      "name": "save_1",
      "type": "ACSubmit",
      "value": "Save_1",
      "uri": "/save"
    },
    {
      "name": "adjust_width",
      "type": "ACElement",
      "value": "<script type=\"text/javascript\">window.onload=function(){var t=document.querySelectorAll(\"input[type='text']\");for(i=0;i<t.length;i++){var e=t[i].getAttribute(\"placeholder\");e&&t[i].setAttribute(\"size\",e.length*.8)}};</script>"
    }
  ]
}
)";

static const char PAGE_SAVE[] PROGMEM = R"(
{
  "uri": "/save",
  "title": "SAVE",
  "menu": false,
  "element": [
    {
      "name": "caption",
      "type": "ACText",
      "format": "Elements have been saved to %s",
      "style": "font-family:Arial;font-size:18px;font-weight:400;color:#191970"
    },
    {
      "name": "validated",
      "type": "ACText",
      "style": "color:red"
    },
    {
      "name": "echo",
      "type": "ACText",
      "style": "font-family:monospace;font-size:small;white-space:pre;"
    },
    {
      "name": "ok",
      "type": "ACSubmit",
      "value": "OK",
      "uri": "/global"
    }
  ]
}
)";

AutoConnectConfig config;
AutoConnect portal;
AutoConnectAux page_global_settings;
AutoConnectAux page_mpu;
AutoConnectAux page_save;

void setup() 
{
  page_global_settings.load(PAGE_GLOBAL_SETTINGS);
  page_global_settings.on([] (AutoConnectAux& aux, PageArgument& arg) {
    if (portal.where() == "/global") {
       Serial.println(F("LOAD /global"));
      SPIFFS.begin();
      File param = SPIFFS.open(PARAM_FILE, "r");
   if (param) {
        aux.loadElement(param, {
      "isBATTERY" 
      });
        param.close();
      }
      SPIFFS.end();
    }
    return String();
  });  
  
  page_mpu.load(PAGE_MPU);
  page_mpu.on([] (AutoConnectAux& aux, PageArgument& arg) {
    if (portal.where() == "/MPU") {
        Serial.println(F("LOAD /MPU"));
     SPIFFS.begin();
      File param = SPIFFS.open(PARAM_FILE, "r");
      if (param) {
        aux.loadElement(param, {
      "MPU_2_FILTER" 
      });
        param.close();
      }
      SPIFFS.end();
    }
    return String();
  });
  
  page_save.load(PAGE_SAVE);
  page_save.on([] (AutoConnectAux& aux, PageArgument& arg) {
    aux["caption"].value = PARAM_FILE;

#if defined(ARDUINO_ARCH_ESP8266)
    SPIFFS.begin();
#elif defined(ARDUINO_ARCH_ESP32) 
    SPIFFS.begin(true);
#endif
    File param = SPIFFS.open(PARAM_FILE, "w");
    if (param) {
      Serial.println(F("SAVE ALL"));
      // Save as a loadable set for parameters.
      page_global_settings.saveElement(param, { 
      "DEVICENAME", 
      "isBATTERY", 
      "INTERVALL", 
      "MPU_1", 
      "MPU_2", 
      "TEMP", 
      "CURRENT", 
      "INPUT" 
      });
      page_mpu.saveElement(param, { 
      "MPU_1_NAME",
      "MPU_1_FILTER",
      "MPU_1_ACC_RANGE",
      "MPU_2_NAME",
      "MPU_2_FILTER",
      "MPU_2_ACC_RANGE",
      "FREQUENCY_MPU",
      "SAMPLES_MPU"
      });
      param.close();
      // Read the saved elements again to display.
      param = SPIFFS.open(PARAM_FILE, "r");
      aux["echo"].value = param.readString();
      param.close();
    }
    else {
      aux["echo"].value = "SPIFFS failed to open.";
    }
    SPIFFS.end();
    return String();
  });

  
  portal.join( { page_global_settings, page_mpu, page_save});
  config.menuItems = AC_MENUITEM_DISCONNECT | AC_MENUITEM_RESET | AC_MENUITEM_HOME;
  portal.config(config);
  portal.begin();

```

At the moment there are on every page load/save buttons, an it seems that everyone handles all the elements of each page. That would be OK for me, if the storing and loading will work…

May I ask for some advice or a simple demo code acording to the topic of my question?
Thank you,
Armin

p.s. 
I am totally new to Github, so I hope this is the correct way to ask a question, otherwise I like to apologize in advance...

Hieromon, 2020-05-13:
Part of the Sketch you posted doesn't contain some elements specified with the saveEelement that should be saved. But I let you know assuming these exist on the complete Sketch you own.

1. About you mentioned that "I was not able to save the elements from the 2nd page to EEProm".
  Have you tried printing PARAM_FILE to diagnose what happened?
  I need diagnostic information from you in order to analyze your sketch.

2. Sketch you posted that doesn't match your wants:
  The following code is a quote for your posted sketch. The saveElement function saves the specified element as **one JSON document**. Your code stores **two different JSON documents** in one PARAM_FILE. However, when Sketch loads it, your code tries to treat it as one JSON document. This is wrong.
    ```cpp
    File param = SPIFFS.open(PARAM_FILE, "w");
    if (param) {
      Serial.println(F("SAVE ALL"));
      // Save as a loadable set for parameters.
      page_global_settings.saveElement(param, {
        "DEVICENAME",
        "isBATTERY",
        "INTERVALL",
        "MPU_1",
        "MPU_2",
        "TEMP",
        "CURRENT",
        "INPUT"
      });
      page_mpu.saveElement(param, {
        "MPU_1_NAME",
        "MPU_1_FILTER",
        "MPU_1_ACC_RANGE",
        "MPU_2_NAME",
        "MPU_2_FILTER",
        "MPU_2_ACC_RANGE",
        "FREQUENCY_MPU",
        "SAMPLES_MPU"
      });
      param.close();
      // Read the saved elements again to display.
      param = SPIFFS.open(PARAM_FILE, "r");
      aux["echo"].value = param.readString();
      param.close();
    }
    ```

ArminPP, 2020-05-13:
Thank you very much for your help!

> Part of the Sketch you posted doesn't contain some elements specified with the saveEelement that should be saved. But I let you know assuming these exist on the complete Sketch you own.

You are right, I reduced the code, because I thougth it is to long. 

> 2. Sketch you posted that doesn't match your wants:
The following code is a quote for your posted sketch. The saveElement function saves the specified element as one JSON document. Your code stores two different JSON documents in one PARAM_FILE. However, when Sketch loads it, your code tries to treat it as one JSON document. This is wrong.

I thought there was a 'hidden' append function so that I could put all the settings in one JSON file ...

After a while I tried it as you mentioned above, but the side effects are still there...
So I shrinked my latest code to show it here, an surprisingly it works :-)
The side effect was/is, that after I pressed the save button, I only got a blank page. 
Sometimes I got a blank page after I hit the load button...

The only thing I was wondering, what happend if i use some unsupportet HTML Tags in the json code, would this cause some side effects?
```
     "name": "DEVICENAME",
      "type": "ACInput",
      "label": "Gerätename",
      "style": "font-weight:bold",        <-------  this isn't supported, I think?

```

Now I played around with the shrinked Code - it works like a charm, the larger one sometimes showed me a blank page again, but after pressing the "return" button (the triangle on the phone), I was able to save/load the parameters.

I hope you don't mind if I put the whole code in here:

```
#define Version "Autoconnect_Test_V0.8a"
#include "C:\Users\Armin\OneDrive\ARDUINO\LIBRARIES\AP_Settings.h"
#define __FILENAME__ (strrchr(__FILE__, '\\') ? strrchr(__FILE__, '\\') + 1 : __FILE__)
#define VERBOSE_MODE    // Alle Debug Prints ausführen 

// ********************************
// ***   Autoconnect Settings   ***
// ********************************
#if defined(ARDUINO_ARCH_ESP8266)
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
using WebServerClass = ESP8266WebServer;
#elif defined(ARDUINO_ARCH_ESP32)
#include <WiFi.h>
#include <WebServer.h>
#include <SPIFFS.h>
using WebServerClass = WebServer;
#endif
#include <FS.h>
#include <AutoConnect.h>

#define PARAM_FILE_GLOBAL      "/PARAM_GLOBAL.json"
#define PARAM_FILE_MPU         "/PARAM_MPU.json"
// #define PARAM_FILE_CURRENT     "/PARAM_CURRENT.json"

// ***********************************
// *** Globale Variablen allgemein ***
// ***********************************
unsigned long previousMillis = 0;

static const char PAGE_GLOBAL_SETTINGS[] PROGMEM = R"(
{
  "uri": "/global",
  "title": "Global_Settings",
  "menu": true,
  "element": [
    {
      "name": "text",
      "type": "ACText",
      "value": "Wählen Sie ihre globalen Einstellungen, und speichern sie diese ab.",
[... 527 lines cut ...]
    }
    SPIFFS.end();
    return String();
  });

  portal.join({ AuxGLOBAL, AuxMPU, saveGLOBAL, saveMPU});
  config.ticker = true;     //LED die blinkt, wenn STA Mode!

 // MENÜAUSWAHL !!!!!!!!!!!!!!!!!!!!!!!!!
  config.menuItems = AC_MENUITEM_DISCONNECT | AC_MENUITEM_RESET; // | AC_MENUITEM_HOME;

  portal.config(config);
  portal.begin();
}

//#########################################################################
void loop() 
//#########################################################################

{
 Serial.print(".");
 delay(1000);
  portal.handleClient();
}
````

Hieromon, 2020-05-13:
> The only thing I was wondering, what happend if i use some unsupportet HTML Tags in the json code, would this cause some side effects?

AutoConnect ignores the undefined attribute for each element.

> Now I played around with the shrinked Code - it works like a charm, the larger one sometimes showed me a blank page again, but after pressing the "return" button

The cause of this behavior is probably a lack of heap. The HTML generated from JSON is too large and does not have enough memory. If you are using the ESP8266 you should suspect insufficient heap area. You can find it in the debug log output by uncommenting [**PB_DEBUG**](https://github.com/Hieromon/PageBuilder/blob/22c0fd08994a11bf8f974478c0606504d1ccc151/src/PageBuilder.h#L34) contained in [PageBuilder.h](https://github.com/Hieromon/PageBuilder/blob/master/src/PageBuilder.h).
If the cause is an insufficient memory then you will see the following message on the serial monitor:
```
[PB] Failed building, free heap: %d (A result of the ESP.getFreeHeap)
```

ArminPP, 2020-05-14:
> If you are using the ESP8266 you should suspect insufficient heap area.

You are right, I use Wemos D1 mini and Wemos D1 mini Pro modules...

Today I made some tests with active PB_DEBUG switch and I got some results...

```
08:18:21.995 -> [PB] at leaving build: 20960 free		---> START UP 
08:18:21.995 -> [PB] Free heap:22960, content len.:6073
08:18:21.995 -> [PB] Res:200, Chunked:1
08:18:22.198 -> [PB] at leaving build: 30040 free		>--- START UP end

08:19:32.434 -> [PB] Free heap:29304, content len.:unknown	---> choose PAGE_GLOBAL_SETTINGS 
08:19:32.434 -> [PB] Res:200, Chunked:2
08:19:32.434 -> [PB] Transfer-Encoding:chunked
08:19:32.434 -> [PB] at leaving build: 16944 free		>--- PAGE_GLOBAL_SETTINGS end

08:20:48.431 -> [PB] Free heap:28376, content len.:unknown	---> SAVE Button in PAGE_GLOBAL_SETTINGS 
08:20:48.431 -> [PB] Res:200, Chunked:2
08:20:48.431 -> [PB] Transfer-Encoding:chunked
08:20:48.465 ->    SAVE /GLOBAL
08:20:48.565 -> [PB] at leaving build: 15128 free		>--- SAVE Button in PAGE_GLOBAL_SETTINGS end

08:22:47.873 -> [PB] Free heap:26888, content len.:unknown	---> OK Button in PAGE_SAVE_GLOBAL
08:22:47.873 -> [PB] Res:200, Chunked:2
08:22:47.873 -> [PB] Transfer-Encoding:chunked
08:22:47.906 -> [PB] **Failed building, free heap:17800**      <--------------- ERROR ---------
08:22:47.906 -> [PB] at leaving build: 17824 free		>--- OK Button in PAGE_SAVE_GLOBAL

Now I got a blank screen with a header named "Global_Settings"

08:25:40.069 -> [PB] Free heap:27264, content len.:unknown	---> choose PAGE_MPU
08:25:40.069 -> [PB] Res:200, Chunked:2
08:25:40.069 -> [PB] Transfer-Encoding:chunked
08:25:40.069 -> [PB] Failed building, free heap:18176
08:25:40.069 -> [PB] at leaving build: 18200 free		>--- PAGE_MPU end

Now I got a blank screen with a header named "Vibration Settings"
```
I switched 4-6 times between global/vibration and always got a blank screen...
I also tried the triangle on my phone several times, it did not work either ...

I recompiled the code an uploaded it once more (without any changes, only to try it a 2nd time from scratch):

Sometimes (but not always) it take quite a longe time after pushing "Save" button and showing up "PAGE_SAVE_GLOBAL" site:

```
08:33:34.149 -> [PB] Free heap:28376, content len.:unknown	---> Pushing "SAVE" in PAGE_GLOBAL_SETTINGS
08:33:34.149 -> [PB] Res:200, Chunked:2
08:33:34.149 -> [PB] Transfer-Encoding:chunked
-> 08:33:34.149 ->    SAVE /GLOBAL
-> 08:35:08.208 -> [PB] at leaving build: 14568 free		---> showing up the PAGE_SAVE_GLOBAL site
(it took ~90 sec to show the PAGE_SAVE_GLOBAL site...)

08:37:50.149 -> [PB] Free heap:27080, content len.:unknown	---> pushing "OK" Button in PAGE_SAVE_GLOBAL site
08:37:50.149 -> [PB] Res:200, Chunked:2
08:37:50.149 -> [PB] Transfer-Encoding:chunked	
08:37:50.149 -> [PB] Failed building, free heap:17992		<--------------- ERROR ---------
08:37:50.149 -> [PB] at leaving build: 18016 free

Now I got the blank screen again...
```

It is interesting that the memory always goes down when the Elements are stored in the filesystem 
and before showing up a site which was loaded before ...

Hieromon, 2020-05-14:
There are at least two parts to improve.

1. `delay(1000)` in `loop()`
  Do not use the delay if you do the periodical process in the loop. It prevents the ESP8266WeServer to work properly. This restriction is not for AutoConnect only, it's true even if you handle the ESP8266WebServer class natively.
  ESP8266WebServer realizes communication by the event from the TCP stack that operates in the loop thread. If the delay is used in the loop, the event cannot be processed normally.
  This part can be improved as follows:
    ```cpp
    unsigned long tm;
    ...
    void setup() {
    ...
      portal.begin();
      tm = millis();
    }

    void loop() {
      if (millis() - tm > 1000) {
        Serial.print('.');
        tm = millis();
      }
      portal.handleClient();
    }
    ```
  
2. Using `readString` in AUX page handler
  If the parameter size to save is large, it is better to avoid using the readString in the AUX page handler. However, it's not a must, is better.
    
    The execution timing of the AUX page handler is inside ESP8266WebServer event processing when the HTTP packet is received, and the memory is consumed to the maximum at that time.
On the other hand, the ESP8266 arduino core String class allocates a buffer in the global heap. So, the ESP8266 arduino core String class is very well considered and is designed to keep the number of buffer expansions as small as possible.
However, since C++ is a native language and not a runtime interpreter for Java, Python, etc., there is no garbage collector. Therefore the Sketch is responsible for the final memory management.
Since the readString will repeat buffer expansion according to the read size, the free and the realloc occur each time, and fragmentation of the heap area is induced.
I deliberately adopted the readString with Elements.ino example because I wanted to give the example as plain as possible, but using it in your sketch may not be appropriate.
    
    The storage parameter size in your sketch is over 2000 bytes, which exceeds 8 times for a buffer of the readStging as 256 bytes. I recommend to drop the transition to the echo back page by the SAVE button and return to the input page.

ArminPP, 2020-05-14:
Thank's again for your time and help!

> Do not use the delay if you do the periodical process in the loop.

I am aware of this evil function and I always use the non blocking version, as you described above.
So I am sorry, I didn't wat to confuse you, this is only my naive way to check if the library works in setup() only. At now I comment out `// portal.handleClient();` in loop() to be sure, that everything works in setup() only. (And it does!)
As you can imagine I am not very deep into system programming, so this is how I handle things sometimes ...

Accoording to your suggestions, I loaded the code without any changes in to an ESP32 Module, and it works without any failure.
Then I removed `aux["echo"].value = param.readString();` in saveGLOBAL/MPU and  also in the Json Part. Then I loaded the code into the Wemos ESP8266 again.

Now there is much more free heap space :
(Startup - open Global Page - save Global - 2x Load Global)
```
15:21:15.449 -> [PB] Free heap:25088, content len.:6072
15:21:15.449 -> [PB] Res:200, Chunked:1
15:21:15.549 -> [PB] at leaving build: 32168 free
15:21:21.723 -> [PB] Free heap:31624, content len.:unknown
15:21:21.723 -> [PB] Res:200, Chunked:2
15:21:21.723 -> [PB] Transfer-Encoding:chunked
15:21:21.723 -> [PB] at leaving build: 19264 free
15:21:35.025 -> [PB] Free heap:30312, content len.:unknown
15:21:35.025 -> [PB] Res:200, Chunked:2
15:21:35.025 -> [PB] Transfer-Encoding:chunked
15:21:35.025 ->    SAVE /GLOBAL
15:21:47.474 -> [PB] at leaving build: 21072 free
15:21:56.511 -> [PB] Free heap:30960, content len.:unknown
15:21:56.511 -> [PB] Res:200, Chunked:2
15:21:56.511 -> [PB] Transfer-Encoding:chunked
15:21:56.511 -> [PB] at leaving build: 18600 free
15:22:30.507 -> [PB] Free heap:29864, content len.:unknown
15:22:30.507 -> [PB] Res:200, Chunked:2
15:22:30.507 -> [PB] Transfer-Encoding:chunked
15:22:30.507 -> LOAD /global
15:22:30.560 -> [PB] at leaving build: 18288 free
15:22:33.318 -> [PB] Free heap:29560, content len.:unknown
15:22:33.318 -> [PB] Res:200, Chunked:2
15:22:33.318 -> [PB] Transfer-Encoding:chunked
15:22:33.318 -> LOAD /global
15:22:33.372 -> [PB] at leaving build: 18096 free
```
In my project I would like to use the Wemos devices, but I still have to check if there is enough free heap space, if the other code is also integrated.
For the future I will consider to change to ESP3 Modules.

Thanks again,
nice greetings from Austria

Hieromon, 2020-05-14:
Close due to resolved.

closed 2020-05-14 by Hieromon