Hieromon/AutoConnect #222: Captive portal is still sending SSID after successful Wifi re-connection

rowifi, 2020-05-17:
Apologies if I'm doing something wrong. I have a button on my ESP32 that erases the Wifi credentials. Once this is done, the captive portal begins and I can access the server to enter new credentials.
Once this is done, the wifi connection is re-established, but the captive portal is still shwoing up on my mobile device. I cannot log into it, but I don't think it should be showing.
If I do a full reset with the valid credentials, then of course the captive portal is not seen at all.
M y code is simple and based on the examples with the addition of an 'erase button' 
Code here:
https://paste.ofcode.org/jXp5pzMFQJdFgCr72QTvNL

Hieromon, 2020-05-18:
> Once this is done, the wifi connection is re-established, but the captive portal is still shwoing up on my mobile device. I cannot log into it, but I don't think it should be showing.

It is normal behavior. AutoConnect does not automatically stop SoftAP immediately after connecting to the WiFi router from the captive portal unless you reset the module.
Refs: https://hieromon.github.io/AutoConnect/gettingstarted.html#connection-establishment
If you want to stop SoftAP without resetting after establishing a WiFi connection, follow the code below:
```cpp
void setup() {
...
  if (Portal.begin()) {
    Serial.println("HTTP server:" + WiFi.localIP().toString());
    WiFi.softAPdisconnect(false);
  }
...
}  
```

closed 2020-05-18 by rowifi

Hieromon, 2020-05-18:
@rowifi,
NOTE: The above code cannot display the successful connection screen after it is established.

rowifi, 2020-05-18:
OK. No problem. I just thought I was doing something wrong. After it established the Wifi connection, I couldn't connect to the captive portal again but I could still see the SSID.
 When I tried to access it nothing happened - or so I think nothing happened it seemed to be intermittent and I was not sure why that was happening.  For sure most of the time it would not show me the menu again.

Hieromon, 2020-05-18:
Your experience is proof that the AutoConnect is working properly.
AutoConnect shuts down DNS after establishing the connection, but not SoftAP. SoftAP is still active so it maintains a WiFi data link with the client device and TCP is reachable. However, DNS is down and the client device is unable to resolve the FQDN for the request to detect the captive portal. As a result, ERR_NAME_NOT_RESOLVED occurs.

rowifi, 2020-05-18:
Thank you Hieromon for your time!

rowifi, 2020-05-19:
I noted that the above code snippet still leaves the ESP32 SSID visible but with a corrupted name.
To remove the local AP completely I changed  'false' to 'true'.

```void setup() {
...
  if (Portal.begin()) {
    Serial.println("HTTP server:" + WiFi.localIP().toString());
    WiFi.softAPdisconnect(true);
  }
...
} ```

Hieromon, 2020-05-19:
Yah, you are right.
I confirmed now that its parameter indicates to stop SoftAP.
https://github.com/espressif/arduino-esp32/blob/49b76649f13acdb8abaf795015d2740bc5f6b340/libraries/WiFi/src/WiFiAP.cpp#L200-L202
Thanks for letting me know.