Hieromon/AutoConnect #223: Support for ESP32-S2 modules

sundar19, 2020-05-19:
I am planning to use ESP32-S2-WROOM in my project. Does AutoConnect support this? Just trying to ensure before placing an order.
Thanks!!!

Hieromon, 2020-05-19:
To be precise, is the ESP32 arduino core support ESP32-S2? synonymously.
I think I'm still working on it.
https://github.com/espressif/arduino-esp32/pull/3990

sundar19, 2020-05-19:
Yes , oh ok but still I might be able to use this autoconnect with s2 series if iam right? 
Thanks!

Hieromon, 2020-05-19:
Well, it depends on the implementation of the arduino core for ESP32-S2. Since AutoConnect does not use a process other than loop thread, it will work if backward compatibility will be ensured for the core libraries especially WebServer class.
However, in the current work, the WiFi class has changed for both AP and STA. I won't know without testing how it will affect AutoConnect.

sundar19, 2020-05-19:
Thank you so much @Hieromon , Good luck for your Esp32-S2 implementation! Hope you notify in your repo after its over !
Thanks!

Hieromon, 2020-05-19:
@sundar19 Thank you for increasing my homework. :-)

PR #406 (2021-09-30): Enhance/v130
Fix #403
Fix #387
Fix #361 
Fix #356
Fix #331 
Fix #325
Fix #296 
Fix #223
closed 2021-09-30 by Hieromon