Hieromon/AutoConnect #224: OTA Issue with Update Server and Spiffs

JemRF, 2020-05-19:
I get the following error when doing an OTA update using the remote updateserver.py

test.ino.esp32.bin from 192.168.2.xxx:xxxx
Updating...
Could not start (0):

The problems seems to be linked to the spiffs file where I am storing my MQTT settings. I find the update works fine when there is no file save in the Spiffs, but does not work after I save my MQTT settings in the Spiffs. Then if I erase the Spiffs files the OTA updatre works again.

I have tried various file partitions schemes (defaily & minimal Siffs) with same results. 

Any ideas what I am doing wrong?

Hieromon, 2020-05-19:
@JemRF Thank you report. I want to reproduce it and fix it before v120 release.
Could you provide the minimum code that can reproduce the problem?

JemRF, 2020-05-19:
When I reduced the code to minimal code the problem went away and I was not able to replicate it. Sorry for the false alarm if the problems returns, and I can consistently replicate, I'll report back to you. By the way this is a great project and I think it's going to become very popular, if not already.

closed 2020-05-19 by JemRF