Hieromon/AutoConnect #225: multiple ESP32 in different sketches and networks lost their AP config simultanously

valki2, 2020-05-20:
Hi Hieromon, 
strange thing happend: after more than 1 year of running flawlessly, 4 ESP32s in different applications just lost their AP settings (and I had to reconfigure via AutoConnects own hotspot). 
Any idea?  Is an alien invasion upcoming or my neighbour playing with an EMP? :-)

Hieromon, 2020-05-20:
Oh, What happened? Four modules at the same time?
I wonder what...
It might be the result of the millis() rounded up caused over the bit width?...No, 1 year is too long.
Has the module been reset during that period?

valki2, 2020-05-21:
hehe... you see me clueless - no idea :) No, its not the migration and no millis() hassle as I avoid that problem within the code. (but yes, that was my first idea too - including a possible "sync" between the different modules by a short house-wide power out for example).... 
 Ill be watching that :)

closed 2020-05-21 by valki2