Hieromon/AutoConnect #227: PlatformIO Vs OTA (Browser) update: not enough space ?

fbattarra, 2020-05-30:
First of all, thank you so much for your incredibly useful job! I love AutoConnect and I am looking forward for the v1.2 for 
- the AUTH  implementation #171 ,
- a safer cohabitation between saved credentials and other custom saved settings #209 ,
- (why not), overcoming the deprecation of SPIFFS (although it's working pretty fine).

I'm working with PlatformIO and I recently tried to move from a WeMos D1 mini to an ESP-01S, stumbling into an issue which is not so trivial to me.

when uploading firmware via PlatformIO / USB, `esptool write_flash` parameters (afaik) compresses the binary (`Compressed 477952 bytes to 333143`), while when I just compile my code in order to have a .bin to upload OTA the compression seems not among the triggered tasks and the final file has the _uncompressed_ size (that's 3.79 Mbit for a 4Mbit board).

I don't know whether I could be able, by better investigating how to compress my build via PlatformIO, I eventually would be able to overcome the issue I am facing: my code (which at the moment is sightly more than an _HelloWorld_ ) nicely fits when uploaded via the IDE / USB, but on an OTA / Browser update the "not enough space" appears.

On the WeMos board I don't have any issue, but it's obvious as there I can count on 32Mb.

I was relying a lot on the OTA update for ESP-01S board, as I need to remove its header pins once programmed for the first time; do you have any hint on what I am actually doing wrong? :)

All my gratitude, once more, for your fantastic job!

[EDIT]: although I declared esp01_1m on my .ini PIO file, the /_ac homepage for the ESP-01S is yelling it has but 4194304 bit ! What am I doing wrong?

Hieromon, 2020-06-01:
@fbattarra Thank you for using AutoConnect.
Bloating of the .bin size is a serious concern. I know the description for uploading a compressed binary file or filesystem in the same section you mentioned.
https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html#compression
Also, there are ideas on how to implement the above method with PlatformIO in the following forum topics of PlatformIO.
https://community.platformio.org/t/esp8266-ota-upload-vs-compressed-ota-upload/13487/5
However, I haven't tried this method yet. For now, I think these methods are the hope to rescue the OTA from the bloated size of the .bin.

fbattarra, 2020-06-01:
@Hieromon Heartfelt thanks for your reply and your hints!

As from the forum topics, as I'm already using Expressif8266@2.5.1, I eventually needed nothing but gzipping (-9 for maximum compression) the bin once compiled with PlatformIO. 

I obtained 333135 bytes instead of 477952 (which was stated as 62.2% of 761840 availables), but it remains still too big for an OTA update (where I should not exceed the remaining 37.8% / 288048).

Please note that, so far, I am only adding  `ESP8266mDNS` in my sketch, so there really is no room for saving space by trying to remove something.

That said, what about the other possible [solution](http://www.letscontrolit.com/downloads/ESPEasyUploaderMega.zip), uploading OTA a first, very lightweight firmware containing only what's needed to perform another OTA update (without any AutoConnect) and picking on the EEPROM the saved credential for the last working WiFi connection ? A tinier fw should free a bunch of flash, allowing then to upload, in the 2nd step, the actual new firmware (gzipped).

It seems a nerdy, brainy way to overcome a board's flash limits for OTA update, where the fw doesn't leave enough space to store a newer version of itself, but it could indeed be the smarter solution available.

Could you please, then, point me towards the code where it reads the EEPROM looking for previously stored SSIDs ? I'd then try to take inspiration from the _.ino_ offered for ESPEasy for the same purpose and modify it in order to work for AutoConnect... :)

closed 2020-06-02 by fbattarra

Hieromon, 2020-06-03:
@fbattarra Thank you for the investigation.
The method you presented as a solution is not really much different from the implementation of the AutoConnectOTA class. I think it will be revealed if you explore the AutoConnectOTA.cpp.

The location of the firmware to replace in the bootstrap based on the [ESP8266 flash layout](https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html?highlight=flash%20layout#flash-layout) is deterministic. And the source binary that the bootstrap copies must fit in the OTA update area.
To implement your proposed method, AutoConnect seems to need to write the binary uploaded to the file system directly to the boot area of the Sketch without bootstrapping, but is it possible to do it by the Sketch itself without using esptool.py?

fbattarra, 2020-06-03:
@Hieromon in the meanwhile I found even [another "tiniest fw for OTA updates"](https://github.com/G6EJD/ESP8266-Over-The_air-OTA-Updating), supposed to let as much free space as possible in the flash for uploading OTA firmwares that are more than half the size of the flash. It somewhat works, even though I found no proper error management there, so you start with your project containing AutoConnect leaving not enough room for an OTA update of itself, then you use the OTA update feature to update a more lightweight firmware and, eventually, you upload your real new firmware OTA because not OTA update area is... bigger.

Therefore, it seems that after the override of `esp01_1m` board specifications for PIO I don't need anymore a _firmware-in-the-middle_ so I won't investigate further the topic - even because 1MB boards are already the bare minimum for projects using AutoConnect and, so far, I am cool with the solution working on PIO IDE.

I am sorry to have not an answer to your question because of my lack of expertise: you mention "OTA update area" and, apart from what I find on [the documentation](https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html#update-process-memory-view) (and, I admit, I haven't yet read carefully) it's unclear to me how to juggle with the memory management. 

I'm now with  550096 KBytes free (although I cannot tell how many for the "OTA area") and a 333289 KB firmware to upload... I feel plenty of room and not anymore in the need of having an intermediate fw to upload up to 640KB worthy firmwares OTA :)
That's why I closed the _issue_ myself :)

Thank you so much once more for the amazing library and your active support!