Hieromon/AutoConnect #228: Redirect on success not working

UpperAlex, 2020-05-31:
Hey,

I am using AutoConnect on port 81 and ESP UI on port 80. 
I did not test all features, but when connecting to an SSID and it is successfull the redirect points to the wrong port. I think it is AutoConnect.cpp line 971. _currentHostIP seems to not include the port.

Would be great to hear if i configured something wrong or if its a bug.

Hieromon, 2020-06-01:
@UpperAlex 
> I think it is AutoConnect.cpp line 971. _currentHostIP seems to not include the port.

To my knowledge, some operating systems for client devices limit the captive portal port to 80 only. But it's worth trying out your way.
Have you tested with the modified code? Please give a PR if your diagnosis is correct and can be resolved by a change code.

UpperAlex, 2020-06-01:
No I didn't test if I can fix it. If I have time I will check that, but I don't have that much knowledge of your code so it's not that easy. 
I am running into another problem. I start the AutoConnect portal after booting and end it, close and stop the server. After that I start another server. When the user triggers it I stop the second server again. Until here it seems to work fine. While trying to add the configuration to the Portal I get an exception (Portal.config(ApConfiguration)). Do you maybe have an idea what I do wrong?

`

  char Ap[] = "asdfasdf";
  char ApPassword[] = "asdfasdf";
  IPAddress ApIp(123, 123, 123, 123);
  ESP8266WebServer Server(80);
  AutoConnect Portal(Server);
  AutoConnectConfig ApConfiguration;

void startAutoConnectPortal(bool immediateStart)
  {
    ApConfiguration = AutoConnectConfig(Ap, ApPassword);
    ApConfiguration.autoReconnect = true; // Enable auto-reconnect
    if (immediateStart)
    {
      ApConfiguration.immediateStart = true;
      ApConfiguration.autoRise = false;
    }
    else
    {
      ApConfiguration.immediateStart = false;
      ApConfiguration.autoRise = true;
    }
    ApConfiguration.apip = ApIp;

    Serial.println("Add configuration to autoconnect portal");
    Portal.config(ApConfiguration);

    Serial.println("Begin autoconnect portal");
    Portal.begin();
    // Portal.begin(WifiSsid, WifiPassword);

    delay(10);

    Serial.println("Close autoconnect portal");
    Portal.end();
    delay(10);
    Serial.println("Close server.");
    Server.close();
    Server.stop();
    Serial.println("autoconnect portal finished.");
  }
`

Hieromon, 2020-06-08:
@UpperAlex 
> While trying to add the configuration to the Portal I get an exception (Portal.config(ApConfiguration)).

Could you post the stack dump? I will investigate.

Hieromon, 2020-10-15:
Closed, no reproducible and no response.

closed 2020-10-15 by Hieromon