Hieromon/AutoConnect #229: Ap starts button press with additional features

Nagaraj4896, 2020-06-02:
While using button press AP starts example  due to immediate start=true the device starts AP at every power cycle.but I need the AP starts when I press button but it should not starts AP every powercycle . The device need to connect WiFi with existing credentials in flash .in loop if I pressed button it should starts AP then I need to connet with WiFi and in the example the device will reconnect fails is noted..but I need to reconnect if WiFi  fails in loop..

Hieromon, 2020-06-08:
@Nagaraj4896 
> While using button press AP starts example due to immediate start=true the device starts AP at every power cycle.but I need the AP starts when I press button but it should not starts AP every powercycle .
> Ans sometimes while access the Autoconnect credentials the device code is uploading it says packet header timed out or MDF file is not match....due to the button AP not working as for my purpose.i planned to delete all credentials and then reset the board...but it doesn't upload... please help to button press AP mode...

I can't grab the intent of your requirement from the topic posted. Can you change it to a more concise expression? Especially, what does mean the power-cycle? Is it a general term or your unique word for this phenomenon? 
So, let us be clear your requirement step by step

First of all, do you want to launch the captive portal with on-demand when you press the external switch?

naga-master, 2020-06-08:
@Hieromon  yes
1.once the device started the esp32 will connect previous credentials stored in eeprom.
2.if button pressed means then AP mode with captive portal should start.

In your example due to immediatestart the AP mode starts initially without connecting previous WiFi..

Hieromon, 2020-06-08:
@Nagaraj4896 Does the approach below meet your requirements?
https://github.com/Hieromon/AutoConnect/issues/145#issuecomment-544916950

naga-master, 2020-06-08:
> @Nagaraj4896 Does the approach below meet your requirements?
> [#145 (comment)](https://github.com/Hieromon/AutoConnect/issues/145#issuecomment-544916950)

but this not working for me .because  AutoConnect* portal = nullptr; is used but i have used autoconnect portal(); how can impleament this with my code.

Hieromon, 2020-06-09:
@naga-master 
Why you change an account?

How do you implement your sketch? The code presented in the comments I mentioned is proven.
Also
> .because AutoConnect* portal = nullptr; is used but i have used autoconnect portal(); how can impleament this with my code.

It's not difficult to allocate an instance dynamically and handle it. There is no problem with using that method. It seems that your problem has nothing to do with the functionality of the AutoConnect library itself, and it due to C++ programming techniques.
Here is not a place to support the general programming techniques of C++, but I may be able to amend your code if you provide code that implements your minimum requirements.

naga-master, 2020-06-09:
> @naga-master
> Why you change an account?
>
> How do you implement your sketch? The code presented in the comments I mentioned is proven.
> Also
>
>
> It's not difficult to allocate an instance dynamically and [...]

Sorry actually I logined wrongly to another my account...

I used the handle client example code 
 Simply I need to add ondemond button press with handleclient.ino file without Config immediatestart in the setup function.....
 
Please help me to sort out this.. @Hieromon

Hieromon, 2020-06-11:
So, I have summarized your requirements in the list below, does it match your needs?

1. The ESP module uses AutoConnect to attempt a WiFi connection. However, if no known access points are found, the captive portal will not start automatically.
This process is performed in the **setup** function.
2. Execute **AutoConnect::handleClient** in the **loop** function. It also detects the pressing state of an external switch.
3. When the ESP module detects the external switch pressing state during the **loop** function, it will start the captive portal. In this scene, you can access the ESP module via SoftAP.
4. If you can be connected to any access point using the captive portal, SoftAP by the ESP module will automatically stop. Also, it will stop the SSID broadcast and you will be able to access the ESP module's web server via the access point you just established.
5. If you cannot be connected to nearby access points, SoftAP will continue to operate, but the ESP module has no connection with other networks. Of course the internet too, it is still standalone.
6. You will press the external switch when the ESP module is in the state of 5. above, SoftAP has already launched, the ESP module will ignore the pressed switch.
7. Manual reset is required to reset the ESP module in cycles 1-5 above. It also does not perform a reset autonomously.

I have questions about the listed requirement as above. How does the ESP module of your Sketch behave if there is no WiFi connection? According to the requirements listed above, you lose access way to the ESP module via the network in the absence of a WiFi connection, can you accept this restriction?

naga-master, 2020-06-11:
> So, I have summarized your requirements in the list below, does it match your needs?
>
> 1. The ESP module uses AutoConnect to attempt a WiFi connection. However, if no known access points are found, [...]

@Hieromon  Thank you ..Yes exactly this is I want.
In loop 
1.if WiFi automatically disconnected in loop and if the esp cannot able to reconnect means..it should not start SOFTAP captive portal ...it should try to connect using saved WiFi credentials in EEPROM.
2.After Pressing external switch it should start SOFTAP but if the ESP cannot be able to detect any of new WIFi means it should stay in SOFTAP mode...

Hieromon, 2020-06-11:
@naga-master 
> 1.if WiFi automatically disconnected in loop and if the esp cannot able to reconnect means..it should not start SOFTAP captive portal ...it should try to connect using saved WiFi credentials in EEPROM.

Its request looks strange to me. Are you understand TCP/IP built over WiFi datalinks enough? It's like connecting to WiFi automatically when you find a known AP with your cellphone carrying around. At this time, your cellphone is a WiFi client. On the other hand, the ESP module acts as a server. Your requirements depend on how you implement that server.
Once the WiFi data link is lost, the IP connection bound to it will be lost. This means that the ESP8266WebServer instance will be disabled. Therefore, if you repeatedly disconnect and establish the WiFi data link with different APs (that is, the IP address will become valid as different), you also need to repeatedly start and stop ESP8266WebServer and AutoConnect without resetting the ESP module.
Do you understand these requirements for sketching as a server?

naga-master, 2020-06-12:
> @naga-master
> 
> > 1.if WiFi automatically disconnected in loop and if the esp cannot able to reconnect means..it should not start SOFTAP captive portal ...it should try to connect using saved WiFi credentials in EEPROM.
> 
> Its request looks strange to me. Do you understand TCP/IP built over WiFi datalinks enough? It's like connecting to WiFi automatically when you find a known AP with your cellphone carrying around. At this time, your cellphone is a WiFi client. On the other hand, the ESP module acts as a server. Your requirements depend on how you implement that server.
> Once the WiFi data link is lost, the IP connection bound to it will be lost. This means that the ESP8266WebServer instance will be disabled. Therefore, if you repeatedly disconnect and establish the WiFi data link with different APs (that is, the IP address will become valid as different), you also need to repeatedly start and stop ESP8266WebServer and AutoConnect without resetting the ESP module.
> Do you understand these requirements for sketching as a server?
> 
> [EDIT]
> What specific use do you envision for the ESP module implemented by the above requirements? (Especially with the constraint that the module never reset)

Because whenever I need to configure the esp to new router at the timeof without knowing IP address of esp connected wifi .so I want to press a button in esp and captive portal should starts and I will conmect to new router...but I don't want to use retainportal = true... ..

ACTUALLY I implemented my requirement below I will tell you how I implemented please tell me is that right or not..if it is not right means please suggest a good solution.
1.in setup  autoconnect added with handle client setup function.
2.in loop all will works as handle client loop function and I am checking the pin for low if that pin is low for 3 seconds I assigned the config.immediatestart = true portal(&config) and then starts captive portal with portal begin like in setup..
 So when I press the button for 3seconds it will start AP mode.
And another one if WiFi is connected for 10mins continuously means I taken the all enteries from flash using AC.load then connect the WiFi as I want...

Is this approach is correct?please provide suggestions @Hieromon  
Thank you

mgfossatti, 2020-06-12:
Hi

mgfossatti, 2020-06-12:
Hi @Hieromon , great project !!! 

I'm trying to deploy a similar funcionality with a Press-Button attached to ESP32.
Let me explain it:
1 - Boot
2 - Check for existing credentials already saved
2.a - If credentials available, try to connect
2.b - If no credentials available, wait for button to be pressed
3.a - Connection successfull, continue with sketch
3.b - Connection not successfull, retry untill button get pressed.
4 - If button pressed, erase previous credentials and run AP

I'm a newby so perhaps you can help me with some example code.
Tks

Hieromon, 2020-06-13:
@naga-master @mgfossatti You are too particular on the integrated features of AutoConnect.

All of your requirements are made possible by implementing the following routine by your Sketch. Also, it can be called from the loop.
- Scan WiFi signal using the **WiFi.scanNetworks** **asynchronous** with **periodically** for broadcasted SSIDs.
- Match detected SSID with AutoConnect credentials.
- If there is a match, try to connect using WiFi.begin natively
- If the WiFi connection could be established, Sketch starts AutoConnect with AutoConnect::begin. The WebServer will be started inside AutoConnect.

Also, the following steps should be implemented by loop.
- Stop AutoConnect when the WiFi station mode status changes to WL_DISCONNECTED.<br>However, if AutoConnect is already activated by an external switch, do not stop it. This judgment process depends on the specification of the user who decides the sketch motion. That is something you should decide.
- Detect the SoftAP start switch. Do not use the **delay** for detection. It should be timeout detection using the **millis**. The **delay** for a few seconds in the loop will hinder TCP communication. ESP8266WebServer is not asynchronous.
- Sketch starts AutoConnect according to the detection result of the switch. It also controls the AutoConnect enable status by some flags.
- If AutoConnect is activated, execute handleClient. If it hasn't started yet, skip it.

As you can see, all of these are programming techniques related to WiFi connectivity. It has nothing to do with the library function of AutoConnect.

naga-master, 2020-06-14:
> @naga-master @mgfossatti You are too particular on the integrated features of AutoConnect.
> 
> All of your requirements are made possible by implementing the following routine by your Sketch. Also, it can be called from the loop.
> 
> * Scan WiFi signal using the **WiFi.scanNetworks** **asynchronous** with **periodically** for broadcasted SSIDs.
> * Match detected SSID with AutoConnect credentials.
> * If there is a match, try to connect using **WiFi.begin** natively
> * If the WiFi connection could be established, Sketch starts AutoConnect with AutoConnect::begin. The WebServer will be started inside AutoConnect.
> 
> Also, the following steps should be implemented by loop.
> 
> * Stop AutoConnect when the WiFi station mode status changes to WL_DISCONNECTED.However, if AutoConnect is already activated by an external switch, do not stop it. This judgment process depends on the specification of the user who decides the sketch motion. That is something you should decide.
> * Detect the SoftAP start switch. Do not use the **delay** for detection. It should be timeout detection using the **millis**. The **delay** for a few seconds in the loop will hinder TCP communication. ESP8266WebServer is not asynchronous.
> * Sketch starts AutoConnect with the **immediateStart** according to the detection result of the switch. It also controls the AutoConnect enable status by some flags.
> * If AutoConnect is activated, execute handleClient. If it hasn't started yet, skip it.
> 
> As you can see, all of these are programming techniques related to WiFi connectivity. It has nothing to do with the library function of AutoConnect.
> So, what's the problem? Which of the above steps corresponds to your problem?

Thank you I will let you know after I try.

mgfossatti, 2020-06-14:
> @naga-master @mgfossatti You are too particular on the integrated features of AutoConnect.
>
> All of your requirements are made possible by implementing the following routine by your Sketch. Also, it [...]

Thanks, will try a solution in this way.

Hieromon, 2020-06-18:
@naga-master Please use the code quotes. I find it painful to review content that doesn't separate code blocks like your post.
Please read at least here. Thank you.

- [Reports the issue to AutoConnect Github repository](https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository)
> the Sketch code with formatted by the code block tag (Reduce to the reproducible minimum code for the problem)   
- [Creating and highlighting code blocks](https://help.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks)

Your code does not implement the following logic I mentioned above. It's almost incomplete code.
- Scan WiFi signal using the WiFi.scanNetworks asynchronous with periodically for broadcasted SSIDs.
- If there is a match, try to connect using WiFi.begin natively.

Also, the logic you have added is inappropriate as a structure to execute inside the loop(). This logic is not fulfilling my advice. (Especially it doesn't ensure asynchrony)

1. It just loads the credentials but does not pass them to AutoConnect::begin for reconnect. There is no logic to find the matched SSID. You need a separate routine that will launch WiFi.scanNetworks when it detects WL_DISCONNECTED.

    ```cpp
      for (int8_t i = 0; i < count; i++) {
        // Loads all crendentials.
        ac.load(i, &entry);
        ssid[i] = (char *)entry.ssid;         //WiFi SSID Names
        pass[i] = (char *)entry.password;     //WiFi Passwords
        //Serial.println(F("Wifi is not connected"));
      }
    ```

2. This detection logic for an external switch will stop the loop() execution. It's synchronous, should be asynchronous. Change to measure elapsed time without stopping loop execution.

    ```cpp
        if (digitalRead(TRIGGER_PIN) == LOW) {
          unsigned long tm = millis();
          while (digitalRead(TRIGGER_PIN) == LOW) {
            yield();
          }
    ```

3. This is probably a simple mistake. AutoConnect::begin always will execute regardless of the state of the external switch.

    ```cpp
      if (millis() - tm > HOLD_TIMER)
        Serial.println("entered in switch ");
      Config.immediateStart = true;
      portal.config(Config);
      portal.onDetect(atDetect);

      if (portal.begin()) {
    ```

4. This does not meet your requirements. You need to pass the credential to AutoConnect::begin that matches the SSID detected by WiFi.scanNetworks.  
And it cannot keep up with IP address changes because it has not stopped the WebServer instance.

    ```cpp
      if (portal.begin()) {
        WebServerClass& server = portal.host();
   ```

#### Note

AutoConnect is not intended to black-box what you expect. It replaces establishing a WiFi connection process with WiFi.begin but does not provide automatic access point detection and automatic WiFi connection while under executing the loop, like some OSs. It doesn't connect "arbitrarily" "behind the usual sketch you write" according to the WiFi radio signal conditions.

Of course, it is possible to implement similar functionality by Sketch, but first of all, I recommend you to implement an ESP8266/ESP32 compliant WiFi-AP search and connection utility before using AutoConnect.

naga-master, 2020-06-18:
@Hieromon Sorry for my post, and simply yesterday i tried on demand button press code it through error at the if(portal) line ,
```
AutoConnect* portal = nullptr;

bool detectSwitch() {
  /*
  Returns true if an external switch to configure is active.
  */
}

bool connectWiFi(const char* ssid, const char* password, unsigned long timeout) {
  WiFi.mode(WIFI_STA);
  delay(100);
  WiFi.begin(ssid, password);
  unsigned long tm = millis();
  while (WiFi.status() != WL_CONNECTED) {
    if (millis() - tm > timeout)
      return false;
  }
  return true;
}

void setup() {
  AutoConnectCredential credt;
  station_config_t  config;
  for (int8_t e = 0; e < credt.entries(); e++) {
    credt.load(e, &config);
    if (connectWiFi(config.ssid, config.password, 30000))
      break;
  }
  if (WiFi.status() != WL_CONNECTED) {
    // Here, do something when WiFi cannot reach.
  }
}

void loop() {
  if (detectSwitch()) {
    AutoConnectConfig config;
    config.immediateStart= true;
    if (!portal) {
      portal = new AutoConnect;
    }
    portal->config(config);
    if (portal->begin()) {
      portal->end();
      delete portal;
      portal = nullptr;
    }
  }
  // Here, ordinally sketch logic.
}
```
**no match for 'operator!' (operand type is 'AutoConnect')**

you said it is all about  wifi technique ,but i have a doubt why i cannot be able to start captive portal with AP mode using autoconnect  by pressing external switch in loop. i want to integrate above technique into handleclient. my main concept is that esp32 should work like tasmota where we can start AP mode by pressing 4 short presses of external button , because in universally the home automation system has the button to configure to new wifi.I want to implement it by autoconnect.all the beautiful work is already done by autoconnect but this is the only part is missing i think , please sort this problem.i don't have in depth knowledge.Thank you for your consistent responses .

Hieromon, 2020-06-18:
@naga-master 
Tosmota itself contains "arbitrarily" "behind the usual sketch you write". That is, it implements the OS-like essence I mentioned. it does not assume Arduino core and is an alternative firmware as Tosmota declares itself.
AutoConnect is the Arduino library. You have misunderstood and need to understand the basic structural characteristics of the Arduino framework that is implemented by the setup and the loop.  I have already presented a processing method that can be implemented.
Its result will lead us to a more constructive discussion.

This topic does not have it. You are just placing an order and requesting that you bring the Tosmata firmware into the Arduino framework. The two are different. Please provide specific specifications to make it possible with the Arduino framework without losing the user's sketch flexibility.

I would like to actively incorporate specifically library specification suggestions. But it has to be feasible. Mere "what I want" is not a specification.
> no match for 'operator!' (operand type is 'AutoConnect')

I cannot reproduce the error in both Arduino IDE and PlatformIO. Its syntax is acceptable.
I asked you to read the [Reports the issue to AutoConnect Github repository](https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository). It contains the information I need to analyze the problem. But you only post "I got an error message". Why can't my request be accepted?
On the other hand, you indicate your request to me. It's just a unilateral request, I think that is an unacceptable act for the open-source community.
Here is not a customer service center.

Hieromon, 2020-09-25:
Closed due to no valid response

closed 2020-09-25 by Hieromon