Hieromon/AutoConnect #230: Delayed deactivation of the portal after connection to AP established

Tost69, 2020-06-08:
Hi Hieromon,

first: thank you very much for your library.

I have some questions regarding your examples HandleClient, HandlePortal and HandlePortalEx (tested on a Wemos D1).

In the documentation (part How to embed) you mentioned that it's better to use AutoConnect::handleClient instead of handleRequest.

But the example HandleClient (using handleRequest) is the only example (of this three) where the Home menu is working fine in Captive portal:
HandlePortal and HandlePortalEx both adds the Root and the GPIO page after the Wifi connection has been established, so Home is not working in the Captive portal.

How can the behaviour in HandlePortal and HandlePortalEx be changed that the Home is working in the Captive portal?

HandleClient (using handleRequest) does not close the Captive portal after the connection to AP has been established, but the displayed IP switches from Captive portal IP to the IP assigned by AP DHCP.
Closing the Captive portal with WiFi.softAPdisconnect(true) after portal.begin works but then there is not enough time to show the assigned IP (statistics).

Is it possible change the behaviour after connecting to the AP to show the statistics page for a while with a count down and then call the reset page (_ac/reset) ?
How could this be done?

Thanks a lot and kind regards, Tom

Hieromon, 2020-06-09:
@Tost69 Thank you for using AutoConnect, also thanks to mentioning the incompleteness of HandlePortal and HandlePortalEx.
However, it is my intention. The biggest difference between HandleClient.ino and HandlePortal.ino is the exact timing of the registration for the page handlers. (ie. handleRoot and handleGPIO) The registration of handleRoot and handleGPIO in HandlePortal.ino (also HandlePortalEX.ino) is after AutoConnect::begin. What that means is that as you mentioned, under the captive portal, HTTP requests for "/" and "/io" are not handled. I presented it as examples in response to the request "I don't want to process anything other than AutoConnect pages in the captive portal state". It also shows what kind of restrictions occur when you select "Use ESP8266WebServer instance allocation by AutoConnect". That is, the HTTP request will be lost because the page handler cannot be registered until the returns from AutoConnect::begin.
You can enable handleRoot and handleGPIO even under the captive portal state by modifying the following steps based on these examples:

- Current:
https://github.com/Hieromon/AutoConnect/blob/d39545ce13c12c3e8fddef599b7325b276707ac9/examples/HandlePortal/HandlePortal.ino#L106-L117

- Modified:
  ```cpp
  ESP8266WebServer server;
  AutoConnect portal;
  
  void setup() {
    ...
    server.on("/", handleRoot);
    server.on("/io", handleGPIO);
  
    // Starts user web site included the AutoConnect portal.
    portal.onDetect(atDetect);
    if (portal.begin()) {
    //  WebServerClass& server = portal.host();
      Serial.println("Started, IP:" + WiFi.localIP().toString());
    }
    else {
      Serial.println("Connection failed.");
      while (true) { yield(); }
    }
    ...
  }
  ```

Tost69, 2020-06-09:
Hi Hieromon,

thanks for your quick support. Regarding your suggestion 'Modified': that's exactly, what I've done and this was working.

I changed this a little bit more: 

1. Within atDetect: set a global bool variable wasInAPMode to true in case the Captive portal is running and added page handlers for Root andf GPIO
2. After portal.begin: condition depending on wasInAPMode:
 - if true: close the Captive portal with WiFi.softAPdisconnect(true); -> no reboot, Root+GPIO now also works in the WiFi net
 - if false: added page handlers for Root andf GPIO

This also worked fine, AutoConnect is a quite flexible solution :-)

But I wanted to use the WebServerClass instantiation by AutoConnect and use portal.handleClient instead. So I tried this with WebServerClass& server = portal.host().
While this is working after portal-begin it's not within atDetect (because you start the webserver after onDetectExit in AutoConnect.cpp).

Would it be possible to allow HTTP requests for "/" and "/io" under the captive portal by changing the order of this two statements?

I have another (probably stupid) question: is it possible to unjoin an AUX page or to deactivate the page handler for one of the URIs previously passed (the opposite of server.on / portal.join)?

I ask this because I want to achieve the following:
- I would like to provide a page in the Captive portal with some instructions/explanations on how to use this and why the portal disappears after successful Wifi connection.
- I want to avoid restarting the device; it should continue to run regardless of whether Wi-Fi connections are currently established or the Captive portal / Wifi connection is available or unavailable.
- I want to provide a page with the sketch content only when a Wifi connection has been established and then the first page (instructions) should no longer be available.

Thank you very much for your help, best regards, Tom

Hieromon, 2020-06-11:
@Tost69 You are very familiar with the behavior of AutoConnect, and it encouraged me.

> Would it be possible to allow HTTP requests for "/" and "/io" under the captive portal by changing the order of this two statements?

To be honest, I was wondering about the onDetectExit call timing when implementing it. It limits what you can do in your exit routine as you mentioned.
My intention is to protect the registration of the page handler for AutoConnect by executing the exit routine before starting the WebServer. AutoConnect uses a **404 handler** (which is known as registration by **ESP8266WebServer::onNotFound**) for the captive portal impersonation. If the user sketch registers a 404 handler at the exit routine it will overwrite the registered AutoConnect handler and it caused the deactivation of the captive portal. Of course unable auto pop-up too. You can see its mechanism by exploring the **AutoConnect::_startWebServer** in AutoConnect.cpp.
However, Sketch can still use the ESP8266::onNotFound after AutoConnect::begin, so my intention is not enough. Whether it's effective or not, I feel a little better.
If accepting your change request has a useful effect on your Sketch, I would not hesitate to do so. Need change?

> is it possible to unjoin an AUX page or to deactivate the page handler for one of the URIs previously passed (the opposite of server.on / portal.join)?

It accurately pointed out the improvement of the AutoConnect. ESP8266WebServer cannot dynamically activate or deactivate the registered page handler. But I might be able to do it only for AutoConnectAux. I will try it. 
Would you give me some time to think?

Tost69, 2020-06-14:
Hi Hieromon,

thank you very much for your reply. I have already noticed the upcoming version 1.20 and also noticed the changes regarding BASIC/DIGEST authentication.

The new AUX append and detach methods looks very promising. I think this really closes a gap in the ESP8266WebServer/WebServer.

What I'm not sure about is whether the destruction of the instance is really necessary in all cases: with the portal methods enableMenu, disableMenu, both the menu items are displayed/hidden in the luxbar and the URIs are activated/disabled.

Why not have something similar for AUX? An AutoConnectAux page is more than a single page to display some values (there may also be another URI to read changed values).
But couldn't it be used together with PageBuilders dynamic page functionality so that the content of a page depends on Sketch values and then the registered handler could distinguish between different content variants? (Maybe then another new method is needed to rename a page in the menu.)

To my request to start the web server before the onDetectExit in AutoConnect.cpp: I am not sure at the moment: On the one hand, you are right, this is something that ensures that the Captive portal works. But on the other hand, it will also be the case that any user of your library finds a way that it will not work (e.g. as described by you by using ESP8266::onNotFound after AutoConnect::begin in the Sketch).
Fortunately, it is usually not the intention of a sketch programmer to make something not work :-).

You already spent a lot of time developing this library; and it's already very good.

I'm just trying to use AutoConnect's Wifi management, along with the built-in PageBuilder functionality, as a holistic concept for all web pages provided by a sketch.

AutoConnect pages and sketch pages should behave similarly and look similar. From the user's point of view, there should be one single web application which provides multiple features: 
- the Wifi Management (AutoConnect)
- probably additional Sketch settings (AutoConnect AUX)
- the Sketch data (AutoConnect AUX or custom Pages sharing the same AutoConnect menu bar).

Hopefully this is not too ambitious.

Kind regards, Tom

Hieromon, 2020-06-16:
@Tost69 
> the portal methods enableMenu, disableMenu, both the menu items are displayed/hidden in the luxbar and the URIs are activated/disabled.
> Why not have something similar for AUX? 

What you've pointed out is correct. There is no consistency between the specification of **AutoConnect::enableMenu** and **AutoConnectAux::menu**. AutoConnect::enableMenu/disableMenu disables the URI too, but AutoConnect::menu just controls the visibility of menu items, URI is still valid as its name implies.

However, there is still room for improvement in this API. Its change may resolve the following things you mentioned:
> But couldn't it be used together with PageBuilders dynamic page functionality so that the content of a page depends on Sketch values and then the registered handler could distinguish between different content variants?

#### Improvement

1. Add a default argument to **AutoConnectAux::menu** to enable/disable the URI.

    ```cpp
    void AutoConnectAux::menu(const bool post, const valid bool = true)
    ```
    If the `valid` is true, the AUX URI is valid and it can respond to HTTP requests. If the `valid` is false, the URI will be temporarily invalid. An HTTP request to the URI will respond 404.
    It also ensures backward compatibility by making it the default argument.

2. The above argument affects the control of AutoConnectAux page generation. Specifically, add the validity evaluation of URI to the following logic.

    https://github.com/Hieromon/AutoConnect/blob/d39545ce13c12c3e8fddef599b7325b276707ac9/src/AutoConnectAux.cpp#L487-L491

    Changed: (**_valid** will be assigned as a private member to AutoConnectAux class)

    ```cpp
    if (uri != String(_uri) && _valid) {
      if (_next) {

How about this?

PS.
> AutoConnect pages and sketch pages should behave similarly and look similar. From the user's point of view, there should be one single web application which provides multiple features:

I also agree with you on this. I should be this and had aimed for this.
But the idea of AutoConnectAux came later. If I had noticed that from the beginning, the AutoConnect structure would have been more consistent.

Tost69, 2020-06-26:
@Hieromon,

I tested the upcoming version 1.2.0 and found that authentication is working fine.

I also checked the append and detach functions:

1. Append (without handler)
* global: `AutoConnectAux *dynAux;`
* within another handler function: 
```
dynAux = Portal.append("/test", "Title");
dynAux->load(AUX_DYN);
```
* result:
   - menu item added (the loaded AUX_DYN **can overwrite title and uri**)
   - menu item calls uri
   - uri is working, shows page content (with Luxbar)

2. Detach
* within another handler function:
`Portal.detach("/test"); `
* result:
  - menu item removed
  - uri is not working anymore
  -> in case the loaded content in AUX_DYN has overwritten the uri this **new uri has to be used**

In the beginning I've had some issues:
- I re-used your example Simple and modified it to test the new features. The homepage uses a script to reload the page every second, so the page is less responsive and my button to dynamically enable / disable the Aux did not worked as expected: the menu item was removed, but the uri was still working, I first fixed this with a `delete dynAux;` after detach. But when it was clear why this was happening, I removed the reload script and it worked as expected (without delete).

- I tried to use the Append with handler but it was not working for me (I probably misunderstood something here):
`dynAux = Portal.append("/test", "Title", handleDyn);`
* result:
   - menu item added
   - menu item calls uri
   - uri is working, **but page has no content** (but Luxbar is visible)
   - the handler was not called
- the subsequent `Portal.detach("/test"); ` has had the following result:
   - menu item removed
   - **uri is still working** and now the page content is shown (but without Luxbar, it's not an Aux anymore)
   - the handler is called now

However, beside this two little things it's working well. I also think it would be a good idea to have the additional argument in the AutoConnectAux::menu function.

Thanks a lot for your time and help, best regards, Tom

Hieromon, 2020-09-17:
@Tost69 Thank you for your reports and so sorry for the late reply.
Actually, I was suddenly forced to join another project and hadn't afforded to access the repository. It's a barren battle that has nothing to do with software development and doesn't contribute to the open-source community. It is a fetter to in-law and responsibility.
I'm still not quite free from the bondage, but I let handle new issues with v120 that you've come across.

Could you please provide your test sketch including `handleDyn` for me to investigate the v120 enhanced code?

Hieromon, 2020-12-02:
Merged #280

closed 2020-12-02 by Hieromon