Hieromon/AutoConnect #232: Right Setup for Start with AP (about 60sec) and then change to STA if possible

WilliamFrasson, 2020-06-14:
Hi Hieromon
i'm tring to setup the AutoConnect to provide always a first step in AP mode and only after start the STA.  The AP mode should start always also when the SSID is detected 
There is a way?

here my code :
 Server.on("/", rootPage);
//m_ACconfig.immediateStart = false;  //start AP always at startup
//m_ACconfig.autoRise = true;
//m_ACconfig.portalTimeout = 60000;  //after 60 seconds stop AP and work only in STA mdoe
m_ACconfig.autoReconnect = true;
m_ACconfig.title = "HOME_12345";
m_ACconfig.apid = "HOME_12345"
m_ACconfig.psk = "home";
m_ACconfig.hostName = "HOME_12345";
Portal.config(m_ACconfig);

auxConfig.load(AUX_CONFIG);
auxConfig.on(getConfig);
auxRestart.load(AUX_RESTART);
auxRestart.on(setConfig);
Portal.join({auxConfig, auxRestart});
Portal.begin();

Hieromon, 2020-06-16:
Regarding the behavior of **AutoConnect::begin**, it describes in the [**documentation**](https://hieromon.github.io/AutoConnect/lsbegin.html), so please refer to it.
However, your request is different from the standard behavior (AutoConnect does not perform STA-connect after launching SoftAP), so AutoConnectConfig's configuration settings cannot do that. You can modify the library as needed.

Hieromon, 2020-09-24:
Close due to no activity record for more than one month.
Reopen when new reports are posted.

closed 2020-09-24 by Hieromon