Hieromon/AutoConnect #234: Reconnect without blocking

TelleRen2k, 2020-06-15:
Hi,

I'm playing around with the AutoConnect library and I have it working fine. When initially setting up an ESP32 I reach the captive portal and I can configure the system to connect to an existing router. The following startups happen with automatic connection to the router. All very good.

But what to do when the connection for some reason is lost (maybe the router is reset) - it seems I can then again by code start the portal - but it will start in a blocking way instead of reconnecting to the router which is now again available?

Is it possible to reconnect (in loop() ) without blocking code execution?

Thank you in advance.

Hieromon, 2020-06-16:
If you only want to reconnect to the router that was established previously, just run WiFi.reconnect. The solution for you is not dependent on AutoConnect, is provided by the ESP32 WiFi library.
```cpp
AutoConnect portal;

void setup() {
  if (portal.begin()) {
    WiFi.setAutoReconnect(true);
    WiFi.persistent(true);
  }
}

void loop() {
  if (WiFi.status() != WL_CONNECTED) {
    WiFi.reconnect();
  }
  portal.handleClient();
}
```
Even if not, since the ESP32 core WiFi library originally considers reconnection, it will reconnect automatically when it becomes available for connection without implementing special processing in Sketch.

However, if the ESP32 may have different IP address than it did before, such as a router reboot, you are worth trying to evaluate the following Sketch code:
```cpp
AutoConnect portal;
String  ssid;
bool    reconnecting = false;

void reconnectWiFi(void) {
  if (ssid.length() && !reconnecting) {
    AutoConnectCredential credt;
    station_config_t  config;

    credt.load(ssid.c_str(), &config);
    WiFi.begin((const char*)config.ssid, (const char*)config.password);
    reconnecting = true;
  }
}

void setup() {
  Serial.begin(115200);
  Serial.println();
  if (portal.begin()) {
    ssid = WiFi.SSID();
  }
}

void loop() {
  if (WiFi.status() != WL_CONNECTED) {
    reconnectWiFi();
  }

  portal.handleClient();
}
```

TelleRen2k, 2020-06-22:
Hmmm, just saw that my comment from last week does not show.

Thank you very much for the feedback :-)

Hieromon, 2020-09-17:
Did your problem go away?
Can I close this topic?

Hieromon, 2020-12-02:
Merged #280

closed 2020-12-02 by Hieromon

ark02, 2021-01-14:
https://github.com/Hieromon/AutoConnect/issues/234#issuecomment-644667512

Just a small comment on this reconnect code

In some cases, if the loop is not entered yet, this will fail to reconnect 
```cpp
void loop() {
  if (WiFi.status() != WL_CONNECTED) {
    WiFi.reconnect();
  }
```

I found that creating a WiFi event trigger allows the reconnect to work in more scenarios thus making it more stable for mission-critical applications 

```cpp

void WiFi_events(WiFiEvent_t event, WiFiEventInfo_t info){
  Serial.printf("[WiFi-event] event: %d\n", event);
  switch (event) {
    case SYSTEM_EVENT_STA_DISCONNECTED:
            Serial.println("Disconnected from WiFi access point. Attempting to reconnect");
            WiFi.reconnect();
            break;
  }

}

void setup(void) 
{
    WiFi.onEvent(WiFi_events);
}
```

Hieromon, 2021-01-15:
@ark02
Your way is a common way for reconnecting, but esp8266 and esp32 have subtle conditions for firing the STA_Disconnect event. (Especially around WiFi.begin)
I didn't dare to adopt this approach because AutoConnect needs to provide consistent behavior even if the platform is different.
Since v1.2.0, Combining AutoConnectConfig::autoReconnect and AutoConnectConfig::reconnectInterval should provide consistent reconnect behavior without blocking. Please check out there, [**Automatic reconnect (Background)**](https://hieromon.github.io/AutoConnect/adconnection.html#automatic-reconnect-background).