Hieromon/AutoConnect #236: Allow AutoConnectOTA upload to SPIFFS

jmcasimar, 2020-06-19:
I am testing the AC_OTA_BUILTIN option. Neverthless, It surprised me that I cannot upload the SPIFFS.

In your file [AutoConnectOTA.cpp](https://github.com/Hieromon/AutoConnect/blob/master/src/AutoConnectOTA.cpp) the function AutoConnectOTA::_open() specifies that only supports upload to Flash.

So first, why only supports flash and not SPIFFS? Have you ever test it trying to support SPIFFS?

I have a couple of days exploring this option and I want to share my results. All my test were made in an ESP32 and I cannot test it on ESP8266 right now. I used the Arduino IDE to compile my sketches.

First, I tried to upload my spiffs.bin directly and I get the Wrong Magic Byte error, so in your function I add a conditional statement to detect if the file contain the word "spiffs" with this statement I can change the default U_FLASH to U_SPIFFS if it is necessary.

Then I test it again and I get the Not Enough Space Error.  I realized that I was using the default 4MB partition scheme, with this scheme I have 1.2MB for the app and 1.5MB for the SPIFFS. The data I want to upload was about 12KB before compiling it, but the compiler adjust it to filled the 1.5MB, so when I tried to upload it was impossible. So I change to the Minimal SPIFFS partition scheme, that gives me 1.9MB for the app and just 190KB for the SPIFFS. More than enough for my actual files.

I test it again, but I get the Bad Size Given error, so I realized that using your maxSketchSpace variable it won´t work when I need to upload the SPIFFS. So finally, I also change this variable depending the binary filename. 

My last test was succesful I can upload even a firmware.bin or a spiffs.bin to my ESP. The function code is this:
```
bool AutoConnectOTA::_open(const char* filename, const char* mode) {
  AC_UNUSED(mode);
  _binName = String(strchr(filename, '/') + sizeof(char));
  uint32_t  maxSketchSpace;
  uint8_t cmd;
  if (_binName.indexOf("spiffs") >= 0){
    cmd = U_SPIFFS;
    maxSketchSpace = UPDATE_SIZE_UNKNOWN;
  } else {
    cmd = U_FLASH;
    maxSketchSpace = (ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000;
  }
#ifdef ARDUINO_ARCH_ESP8266
  WiFiUDP::stopAll();
#endif
  // It only supports FLASH as a sketch area for updating.
  if (Update.begin(maxSketchSpace, cmd)) {
    if (_tickerPort != -1)
      pinMode(static_cast<uint8_t>(_tickerPort), OUTPUT);
    _status = OTA_START;
    AC_DBG("%s updating start\n", filename);
    return true;
  }
  _setError();
  return false;
}
```

jmcasimar, 2020-06-19:
I forgot to mention that in order to succeed, we need to upload the first sketch in the same partition table that will be used always. It seams that is possible change the partition table via OTA but the process it is too much more complicated.

Hieromon, 2020-06-20:
I didn't think it was so important to provide the file system as an upload destination option. You should have seen the comments in the code in AutoConnectOTA.cpp. In other words, I intentionally supported only OTA via FALSH.

https://github.com/Hieromon/AutoConnect/blob/d39545ce13c12c3e8fddef599b7325b276707ac9/src/AutoConnectOTA.cpp#L123

SPIFFS is deprecated in ESP8266 and will be replaced by LittleFS. Therefore, I have to test OTA via three areas: SPIFFS, LittleFS, FLASH, and also validation on two platforms, ESP8266, ESP32. I wanted to release AutoConnectOTA as soon as possible rather than spending time testing OTA via the filesystem.

Actually, Update class only changes the address of the updater to be read according to the **command** argument. ESP32 does this reasonably than ESP8266. The error you encountered is reliable because the ESP32 Update class uses esp_partition_find_first to position the updater.
However, the ESP8266 Update class has a different implementation, which is not a reliable method. (Especially when it is not possible to determine at runtime whether the file system is LittleFS or SPIFFS)

Do you know what the bootloader is doing to update the firmware from the file system? I tested over a hundred patterns for OTA updates, but I'm not familiar with the ESP8266 bootloader constraints. Therefore, I am not fully aware of what happens as a result of firmware updates from the file system. Please let me know if you know how it works. (I read the Espressif documentation carefully)

After all, is it so important to provide the option of where to OTA upload to the Sketch? What is that use case for?
Of course, OTA support via the file system is not so difficult. (maxSketchSize can also be calculated),

jmcasimar, 2020-06-20:
Ok , let me explain my case. 

I have an array of ESP inside a system recolecting data from different sensors and is pretty difficult to access them. Actually, I need about 6 people and it takes about 5-6 hours to take the structure  out and to put it in again. I am using MQTT as my primary communication protocol, and the normal program flow is try to connect to internet, if succeed then stablish communication with the mqtt broker, if not then create the portal. Right now, my portal has some customs pages that I storage as json files in the file system. Until here everything is ok. 

My problem is that recently I realized that my WiFi signal is too weak that most of the time the esp failed to connect and when they are not connected I am losing the data they are collecting. I do not have this problem with the esp that are near to shore, but with the others I have been trying to improve my WiFi signal with multiples repeaters, but It seams that the structure works like a Faraday´s box making difficult to connect them to WiFi.

To solve this problem I though to start using esp-now as an alternative communication protocol and look for some improvement in the collecting data system. So I am developing this solution to test it, but I need to add a couple of extra keys in my customs pages in order to make it work. Like my customs pages are in the file system and I have AutoConnectOTA I thought the easiest way was to upload the new spiffs.bin and firmware.bin files with OTA. Neverthless, It was not that easy and that is why I opened this issue.

At this point, I think I have 3 solutions to my problem:
* First and worst, to take the structure out just to upload the new firmware and customs pages to all the esp.
* Second, to enable OTA to work with SPIFFS and Flash just like I did.
* Third, to stop using file system to storage my customs pages and insert them in directly in Flash.

Having said that, I understand the complicity to develop AutoConnectOTA and integrate them into 2 different plataforms. The fact that SPIFFS is deprecated and the new LittleFS just make it worst. I used to work with esp32, and  I am not familiar with esp8266, that is why I cannot test it in both platforms right now. I am going to buy a couple to esp8266 to test and share my results if you think that may be usefull.

I think that if AutoConnect gives the option to create custom pages and save them in the file system it would be nice if AutoConnectOTA let you modify this customs pages. It is not completely necessary add this feature, but as you said it is not so difficult.

If you think that is not important to support this feature I am completely fine, my only recomendation would be to add a note in [Customs Pages docs](https://hieromon.github.io/AutoConnect/acintro.html#write-the-custom-web-page-with-json) where you clarify that in case to use filesystem to save customs pages cannot be updated with OTA.

Hieromon, 2020-06-22:
@jmcasimar Thank you for the detailed explanation.

In this situation, if you use the AutoConnectFile element (which can upload files to SPIFFS), you need to modify a page that contains AutoConnectFile. And its action may lead to a loss of continuity of UI operation because AutoConnectFile will replace itself which is the own custom web page.

In this case, the sketch would be more flexible if AutoConnectOTA could upload both the bin and the json that described the custom web pages in the same operation. I understand the situation you encountered.

I have some ideas upon considering this feature enhancement:

1. Add the following options to the built-in page of AutoConnect OTA (Its actual body is a custom web page made from AutoConnectAux).

    [image]

    - Add an updater file type like as above.
    - AutoConnectOTA handles whether to output to the file system or update the firmware via the Update class according to the file type.

2. Or, AutoConnectOTA will **automatically identify** the destination by the **file extension**.

- If the file extension is **.bin**, AutoConnectOTA will update the firmware using the Update class.
- The choice of whether Flash or Filesystem is the recipient of the update binary is identified by the radio button like as above.
- With this specification, it may not be necessary to upload new firmware to the filesystem. (the destination can be always **FLASH**)

The above two candidate specifications will provide the ability to easily upload files without using **AutoConnectFile**. (It may be better to change the menu title from "**Update**")

Is this idea consistent with your intent?

jmcasimar, 2020-06-22:
Perfect! That idea is what I was looking for.

Personally, I like the idea that AutoConnectOTA identify the destination by the file extension, because it reduces the number of interactions with the user and also it reduces the number of errors or exceptions that AutoConnectOTA have to handle.

It does not matter what way you decide to go, both will work. Thank you @Hieromon

Hieromon, 2020-06-23:
@dmaixner Hello, If you follow this topic, can I have your opinion?

I planned to integrate a regular file upload facility into AutoConnectOTA. This enhancement will allow user sketches to update files on the ESP module's file system via OTA without using AutoConnectFile.

Since you have [previously commented](https://github.com/Hieromon/AutoConnect/issues/204#issue-601058751) for AutoConnectOTA, I assume you are continuing to use AutoConnectOTA. So I'm concerned that this enhancement will affect the current functionality you know.
I intend to keep backward compatibility with AutoConnectOTA, but cannot guarantee it completely. So if possible, can you help me with any advice about the specification restriction I should care?

The followings are candidate specifications for the AutoConnectOTA Enhancement. I will release it in v1.2.0.

1. Add function to upload a regular file to Filesystem on the ESP module.
2. No changed to the AutoConnectOTA UI page. (Its substance is AutoConnectAux), need to change the menu title? ( which is now "**Update**"), Also I will revise the explanation text on the top of the  AutoConnectOTA page.
3. Whether the upload destination file system is **SPIFFS** or **LittleFS** is decided at the time of sketch compilation.
4. Usage declaration of AutoConnectOTA will specify by **AutoConnectConfig::ota**. (no change)
5. If the OTA file has the "**.bin**" extension, AutoConnectOTA will update the ESP module firmware. (the file is considered a binary sketch)  
**Due to this specification, the file which has a ".bin" extension cannot be uploaded as the regular data file**.
6. If the OTA file has **no** "**.bin**" extension, AutoConnectOTA will write it to the flash on the ESP module via the filesystem. It will create if the file does not exist, otherwise, the file will be replaced.
7. If you upload a .bin file, the module will be rebooted after the uploading completion. On the other hand, if you upload a regular file, the module will not restart automatically.  
Perhaps, I might add the option of auto-restart to the AutoConnectOTA page. I care about the case where a user uploads a JSON file of a custom web page.

@jmcasimar The above is the current specification candidate. I wait for your correction.

Thank you for your contributions.

dmaixner, 2020-06-24:
Hello @Hieromon,
yes, still using it, thank you for letting me know. My use case is rather simple -- I just wanted to upload new firmware by selecting file (I'm not using updateserver to provide the firmware). I'm not uploading files to FS at the moment (and still, I suppose I would be using separate Aux with AutoConnectFile for this, because it is kind of strange to upload regular files via page designed to update firmware). But I understand there can be such a request and original OTA web server (from ESP lib) has this possibility. If I remember correctly, there are two open-file-dialog buttons, one for firmware and one for FS data. If you would use this approach, it would be great to have the possibility to configure the combination -- if just uploading firmware, or just FS or both and customize the page accordingly -- but I like your approach better, it seems it should consume less memory.

Here are my notes to your specification:
1) ok
2) ok, Update is fine, you can still change it in labels (which I need to be able to localize every visible string)
3) awesome, this is needed -- I suppose this same selected FS is going to be used with other features as well (like saving/loading elements to/from file)
4) ok
5) ok, this should work I suppose. Maybe it would be nice if it's configurable, e.g. by some regex with default value set to "*.bin".
6) ok, no problem
7) there is still possibility to restart device from menu and to specify your own description label on the OTA page. This should be enough to let the user know he needs to restart. What if he wants to upload more files and restart only after all those files are uploaded? I would rather keep it simple and not add another elements.

Those are just my thoughts, I hope it helps. :)

David

jmcasimar, 2020-06-24:
My notes to your specification:

1. Perfect
2. I agree with @dmaixner  **"Update"** is ok while you can still change the title in labels.
3. Perfect
4. Ok
5.  Ok. I like that approach. I am not sure if someone maybe want to upload a **".bin" file** into the file system, it would be weird but as long as nobody need it, It will work.
6. Perfect
7. Ok, I think the auto-restart option is a good idea, in particular it would be a good option for my use case. Neverthless, I let the default restart only when **".bin" file** was uploaded.

Thank you!

Hieromon, 2020-10-10:
@jmcasimar, @dmaixner I implemented the file upload feature via AutoConnect OTA and staged it to the [enhance/v120](https://github.com/Hieromon/AutoConnect/tree/enhance/v120) branch. It still lacks coverage for tests. Please evaluate before release if possible.
Thank you for waiting.

Hieromon, 2020-12-02:
Merged #280

closed 2020-12-02 by Hieromon