Hieromon/AutoConnect #237: AC Input Type for Password

hyper-wave, 2020-06-29:
How can we configure a ACInput on a custom webpage for using a password? Currently the API lists no mention of "password" as an element type.

josjimenezja, 2020-07-28:
Were you able to do this setup? i need too

Hieromon, 2020-09-17:
Currently, the only way is to place the `input type="password"` element natively using AutoConnectElement.  The following custom web page json includes an `input type="password"` element as a `pass`.

```json
"element": [
  {
    "name": "pass",
    "type": "ACElement",
    "value": "<label for=\"pass\">Password:</label><input type=\"password\" id=\"pass\" name=\"pass\" minlength=\"8\" required>"
  }
]
```

The value of pass can be retrieved by qualifying the ACElement name `pass` with square brackets in the AUX page handler.

```cpp
String pass = aux["pass"].value;
```

I've been aware of the need for the password element for some time, but I thought it wasn't very frequent.
I will add the **ACPassword** element at the request of some people. It since v120.

josjimenezja, 2020-11-10:
Hi @Hieromon  i try to used your code but when i  print the password in the serial port afer read it, the print `"<label for=\"mqttpass\">Password</label><input type=\"password\" id=\"pass\" name=\"pass\" minlength=\"8\" required><br>" `

Hieromon, 2020-11-11:
Of course, that's exactly what AutoConnectElemet owns.  You need to understand the passing of arguments by the ESP8266WebServer library (WebServer library for ESP32) in order to receive the response with the `input type="password"` tag I presented. The [documentation](https://hieromon.github.io/AutoConnect/achandling.html#custom-field-data-handling) has a description of it.  See [How you can reach the values](https://hieromon.github.io/AutoConnect/achandling.html#how-you-can-reach-the-values) of a documentaion.

Since AutoConnectElement is a native HTML element, to get the POST value, access the page arguments passed by the ESP8266WebServer (or WebServer for ESP32) library directly. We use the [ESP8266WebServer::arg](https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer#getting-information-about-request-arguments) function for that, also PageBuilder owns an equivalent function too called [PageArgument::arg](https://github.com/Hieromon/PageBuilder#arguments-of-invoked-user-function). You can use PageArgument::arg within the AUX page handler.

```json
{
  "name": "pass",
  "type": "ACElement",
  "value": "<label for=\"pass\">Password:</label><input type=\"password\" id=\"pass\" name=\"pass\" minlength=\"8\" required>"
}
```

```cpp
String aux_page_hadler(AutoConnectAux &aux, PageArgument &arg) {
  Serial.println(arg.arg("pass"));
}
```

This means that there is no particular reason to implement for `input type="password"` as AutoConnectElement. (Maybe call it AutoConnectPassword?) So I didn't implement it, we can always take an alternative.
To be honest, I think AutoConnectPassword is not so useful. The disadvantage of increasing memory by implementing it is greater than the convenience of the AutoConnectPassword.

Hieromon, 2020-12-02:
Merged #280

closed 2020-12-02 by Hieromon