Hieromon/AutoConnect #238: HTML page with table using AutoConnectAux

samiron7567, 2020-07-04:
I am newbie in AutoConnect and JSON, With below example as given, it is possible to achieve the requirement on a simple HTML page with table inserted inside. However I am not able to attach this page with AutoConnectAux, so I can get the AutoConnect Header with the custom web page. 

Also it is very difficult to form a table in ArduinoJSON with stylesheet attached to it. I searched all your web page but unable to find such example that describes defining stylesheet and calling into such table or any content with  AutoConnectAux and ArduinoJSON. Can you please redirect me to the correct documentation or if not available, can you guide me, if that is possible then how to do that.

```
void handleRoot() {
  String page = PSTR(
"<html>"
"</head>"
  "<style>"
    "table {"
      "font-family: arial, sans-serif;"
      "border-collapse: collapse;"
      "width: 100%;"
      "color: black;"
    "}"
    "td, th {"
      "border: 1px solid #dddddd;"
      "text-align: left;"
      "padding: 8px;"
    "}"
    "tr:nth-child(even) {"
      "background-color: #dddddd;"
    "}"  
  "</style>"
"</head>"
"<body>"
  "<header>"
    "<div class=\"menu\">" AUTOCONNECT_LINK(BAR_24) "</div>"
    "<p><b>Hello World</b></p>"
  "</header>"
  "<p>Put some text here</p>"
"<table>"
    "<tr>"
      "<th>List1</th>"
      "<th>List2</th>"
    "</tr>"
    "<tr>"
      "<td>option1</td>"
      "<td>option2</td>"
    "</tr>"
  "</table>"
"</body>"
"</html>"
  );
  server.send(200, "text/html", page);
}

void setup() {
  server.on("/", handleRoot);
}
```

Hieromon, 2020-09-18:
You can define shorter CSS using the [ACStyle](https://hieromon.github.io/AutoConnect/acelements.html#autoconnectstyle) element as long as you don't break the style of the pre-defined AutoConnectAux page.

```json
...
  "element": [
    {
      "name": "tbstyle",
      "type": "ACStyle",
      "value": "table {font-family: arial, sans-serif;border-collapse: collapse;width: 100%;color: black;}td, th {border: 1px solid #dddddd;text-align: left;padding: 8px;}tr:nth-child(even) {background-color: #dddddd;}"
    },
    {
      "name": "table",
      "type": "ACElement",
      "value": "<table><tr><th>List1</th><th>List2</th></tr><tr><td>option1</td><td>option2</td></tr></table>"
    },
```

However, additional styles based on this way should not interfere with the style of the AutoConnectAux page. You should separate the scope from AutoConnectAux style by making it a unique style class. Avoid style definitions that have a global effect as described above.

Hieromon, 2020-10-20:
Closed, the solution has mentioned above.

closed 2020-10-20 by Hieromon