Hieromon/AutoConnect #239: invalid types 'AutoConnectAux*[const char [5]]' for array subscript

soemarko, 2020-07-12:
From the [Tips for data conversion](https://hieromon.github.io/AutoConnect/datatips.html), I can do `int value = aux["INPUT"].value.toInt();` as shorthand.

But when I tried:

```
		AutoConnectAux* auxPage = portal.aux("/app-config");
		auxPage->loadElement(file, { "key", "server", "port" });
		Serial.println(auxPage["port"].value.toInt());
```

The above error appears. No error when I tried the long way. 1 extra line is not a big deal. I'm using the esp32 dev board if that matters.

Hieromon, 2020-09-18:
This indicated matter contains a little misleading. Due to the interpretation of C++ syntax.
An `auxPage` is a pointer, so syntactically valid ways of calling its operator[] would be:

```cpp
Serial.println((*auxPage)["port"].value.toInt());
```

or

```
Serial.println(auxPage->operator[]("port").value.toInt());
```

Hieromon, 2020-10-15:
Closed, not AutoConnect library issue.

closed 2020-10-15 by Hieromon