Hieromon/AutoConnect #241: Disable items in statistic page ? This is not an issue

manuprendlair, 2020-07-17:
Hello,

Is it possible to remove some item from the statistic page without altering source code ?
Is it possible to do that ?

Because I found a way by removing items in the _PAGE_STAT[] definition ( in AutoConnectPage.cpp).

Thanks

Hieromon, 2020-09-21:
There is no other way to modify the _PAGE_STAT other than by editing the library source code directly. If you just want to remove items, you can use the **#ifndef** directive with the preprocessor to remove the _PAGE_STAT table item.
The code below shows the basic stance of removing items with #ifndef.

https://github.com/Hieromon/AutoConnect/blob/d39545ce13c12c3e8fddef599b7325b276707ac9/src/AutoConnectPage.cpp#L592

```cpp
"<table class=\"info\" style=\"border:none;\">"
  "<tbody>"
#ifndef NO_STATS_ESTABLISHEDCONNECTION
  "<tr>"
    "<td>" AUTOCONNECT_PAGESTATS_ESTABLISHEDCONNECTION "</td>"
    "<td>{{ESTAB_SSID}}</td>"
  "</tr>"
#endif
#ifndef NO_STATS_MODE
  "<tr>"
    "<td>" AUTOCONNECT_PAGESTATS_MODE "</td>"
    "<td>{{WIFI_MODE}}({{WIFI_STATUS}})</td>"
  "</tr>"
#endif
#ifndef NO_STATS_IP
  "<tr>"
    "<td>" AUTOCONNECT_PAGESTATS_IP "</td>"
    "<td>{{LOCAL_IP}}</td>"
  "</tr>"
#endif
...
```

Then specify the definition of the items to be removed with platformio.ini of PlatformIO IDE.

```ini
build_flags =
  -DNO_STATS_ESTABLISHEDCONNECTION
  -DNO_STATS_MODE
  -DNO_STATS_IP
```

Hieromon, 2020-11-09:
Closed due to no activity for more than a month.

closed 2020-11-09 by Hieromon