Hieromon/AutoConnect #242: Issue- connecting to WiFi

joy-sen, 2020-07-21:
Hello,

My initial intension is just to connect the esp32 device with the wifi router. But i don't want to hard code the router Ssid and password in the program. So i want to do it by using Autoconnect.

Here is the below program i am using to do the so. The problem i am facing is, i am unable to connect the esp32 device with any wifi router using Autoconnect library.  
At the same time if i am hard coding the wifi Ssid and Password, the esp32 device is getting connected to the router.

/******************************** Smart Plug Source Code ESP32 *********************************/

/*=======================================================================================*/
/*                                 Header File Declaration                               */
/*=======================================================================================*/
#include <WiFi.h>
#include <AutoConnect.h>
#include <WebServer.h>

/*=======================================================================================*/
/*                                    Object Declaration                                 */
/*=======================================================================================*/
WebServer Server;
AutoConnect Portal(Server);
AutoConnectConfig Config;

void Config_Page()
{
String  content =
"<!DOCTYPE html>"
"<html>" 
"<head>" 

"<style>" 
"h1{text-align: center;}" 
"h3{text-align: center;}" 
"p {text-align: center;}" 
"div {text-align: center;}" 
"img { display: block; margin-left: auto; margin-right: auto;}" 
"</style>" 

"<meta name=\"viewport\" content=\"width=device-width, initial-scale=1\">" 
"<body style=\"background-color:powderblue;\">" 
"<h1 >WELCOME TO CONFIGURATION PAGE</h1>" 
"<p id =\"p1\"style=\"font-size:130%\"></p>" 
"<p id =\"p2\"style=\"font-size:130%\"></p>" 

"<script type=\"text/javascript\">"
"var today = new Date();" 
"var date = today.getFullYear()+'/'+(today.getMonth()+1)+'/'+today.getDate();" 
"document.getElementById('p1').innerHTML=date;" 

"function startTime(){"
"var today=new Date();"
"var h=today.getHours();"
"var m=today.getMinutes();"
"var s=today.getSeconds();"
"m=checkTime(m);"
"s=checkTime(s);"
"document.getElementById('txt').innerHTML=\"Time : \" + h + \":\" + m + \":\" + s;"
"var t=setTimeout(startTime,500);}"

"function checkTime(i) {if (i < 10){i = \"0\" + i}; return i;}"
"var day = today.getDay();" 
"var daylist =[\"Sunday\",\"Monday\",\"Tuesday\",\"Wednesday \",\"Thursday\",\"Friday\",\"Saturday\"];" 
"document.getElementById('p2').innerHTML=\"Day : \" + daylist[day];" 
"</script>" 

"<body onload=\"startTime()\">" 
"<div id=\"txt\"style=\"font-size:140%\"></div>" 
"<h3> Click the settings icon to access Autoconnect Menu</h3>" 
"<p></p><p style=\"padding-top:15px;text-align:center\">" AUTOCONNECT_LINK(COG_24) "</p>" 
"[image]"
"</head>"  
"</html>";

Server.send(200, "text/html", content);

}

void setup(){

/*********************************** Common Setup ***********************************/
	delay(1000);
	Serial.begin(115200);
	Serial.println();

/************************************************************************************/

	Config.autoReconnect = true;
	Config.hostName = "esp32-01";
  
	Server.on("/", Config_Page);

/****************************** Connecting To Wifi Network **************************/      

	if(Portal.begin())                 
  	{
    		Serial.println("Server Started, IP:" + WiFi.localIP().toString());     
  	}
  	else
  	{
    		Serial.println("Connection Failed...!!!");
    
    		while(true)
    		{
      			yield();
    		} 
  	}

/*************************************************************************************/

}

  void loop(){

	Portal.handleRequest();
	if(WiFi.status()==WL_IDLE_STATUS)
	{
  		ESP.restart();
  		delay(1000);
	}
}

Please suggest how i can overcome this issue as soon as possible as i am stuck at very beginning step. For information i am following all the steps mentioned in this
[Link](https://hieromon.github.io/AutoConnect/index.html) 

Thanks.

TryingToGITthis, 2020-07-23:
When you connect to the captive portal, are you seeing a list of available WiFi access points?

joy-sen, 2020-07-28:
Yes i am able to see the available access points while connecting to the captive portal.
The good news is now i am able to connect the ESP-32 device with my router. There is some issue with the android mobile i was using when i changed the android device it's started working as expected. 

Thank you for your kind attention.

joy-sen, 2020-08-12:
My problem is resolved... So i am closing the issue.
Thanks.

closed 2020-08-12 by joy-sen