Hieromon/AutoConnect #243: File Browser and LittleFS filenames missing first character, no root directory

monarchit, 2020-07-23:
I have encountered an issue with the display of filenames on the FS WebServer file manager edit page.

I have used the example sketch and switched to LittleFS and it would allow me to upload and view files (not in the \ directory but just empty directory for some reason) also, the filenames are missing an initial character thus I am unable to download any files at all. I am not sure if the files are being corrupted when they get uploaded as I have no way to check that.

Is there any way to patch this issue with filenames and download feature?

Thanks a lot.

Hieromon, 2020-09-21:
It is unclear how the indicated behavior is related to AutoConnect.
The FSBrowser code you mentioned is based on the ESP8266 WebServer library's example and the upload part has not been modified for the AutoConnect case as of v1.15 release. I recommend checking that you confirm behaver the same with the FSBrowser in the ESP8266 library.
After that, if it works as you expected, provide me with information by referring to [**this page**](https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository) to let me start the investigation. The location of the cause cannot be decomposed based on your posted information alone. Thank you.

Hieromon, 2020-10-15:
No response, closed

closed 2020-10-15 by Hieromon