Hieromon/AutoConnect #244: Watchdog triggered during captive portal

trycoon, 2020-07-30:
To make my ESP32 firmware a bit more robust and restart the microcontroller in case it gets stuck I have enabled the builtin watchdog support:   esp_task_wdt_init(WDT_TIMEOUT_SEC, true);
If the program fail to reset the watchdog within 60 secs the ESP32 reboots, the reset logic is called within the main-loop.
This works great, except when AutoConnect is in the Captive portal-mode (to configure my WiFi), then it seems to not run the loop()-function at all, and the device will reboot every 60 secs which is very annoying.
Is there a alternative loop-called when in Captive portal-mode that I can jack into and run the reset watchdog calls? Or do you have other suggestions? Thanks!

Kashyapkoshti, 2020-08-28:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 

I am getting this error in my esp32 devkit v1
How to solve it?

Hieromon, 2020-09-21:
I think that it can be realized by adding logic that calls the exit routine in the following part of AutoConnect::begin.

https://github.com/Hieromon/AutoConnect/blob/d39545ce13c12c3e8fddef599b7325b276707ac9/src/AutoConnect.cpp#L226-L235

Change to like this:

1. Prepare the exit routine register method to AutoConnect class

**AutoConnect.h**
```cpp
class AutoConnect {
public:
  ...
  void AutoConnect::whileCaptivePortal(std::function<bool(void)>);
  ...
protected:
  ...
  std::function<bool(void)>  _whileCaptivePortal;
  ...
}
```

**AutoConnect.cpp**

```cpp
void AutoConnect::whileCaptivePortal(std::function<bool(void)> fn) {
  _whileCaptivePortal = fn;
}
```

2. Call an exit routine while in the captive portal if the exit is available

**AutoConnect.cpp:L226**

```cpp
while (WiFi.status() != WL_CONNECTED && !_rfReset) {
  handleClient();
  // By an exit routine to escape from Captive portal
  if (_whileCaptivePortal) {
    if (!_whileCaptivePortal())
      break;
  }
  // Force execution of queued processes.
  yield();
  // Check timeout
  if ((hasTimeout = _hasTimeout(_apConfig.portalTimeout))) {
    AC_DBG("CP timeout exceeded:%ld\n", millis() - _portalAccessPeriod);
    break;
  }
}
```

3. An exit routine can reset the WDT and support escape from the captive portal

**example.ino**

```cpp
AutoConnect  portal;

bool resetWDT(void) {
  esp_task_wdt_init(WDT_TIMEOUT_SEC, true);
  return true;  // If false, exit from the captive portal
}

void setup(void) {
  ...
  portal.whileCaptivePortal(resetWDT);
  portal.begin();
  ...
}
```

If the above approach meets your needs, and from the point of view you mentioned out, it may help other users who have a headache to escape from the captive portal.

Escape from the captive portal depends on a timeout. However, this exit routine may be useful in situations where time-constrained situations cannot be selected. This assumes a different convenience than the background of your request, but do you think this extension will be useful to other u

trycoon, 2020-09-22:
Yes, thanks you! Essentially the reset of watchdog has the same use-case as "yield()", something that needs to be run while busy-waiting for WiFi to be active. Nice touch to be able to exist loop without having to wait for timeout.

Hieromon, 2020-12-02:
Merged #280

closed 2020-12-02 by Hieromon

zekageri, 2022-09-08:
### Captive restart still happening

```
assert failed: tcp_recved IDF/components/lwip/lwip/src/core/tcp.c:980 (don't call tcp_recved for listen-pcbs)

Backtrace:0x40083cbd:0x3ffb37800x4008ece5:0x3ffb37a0 0x400944d9:0x3ffb37c00 
x40130a66:0x3ffb38f0 0x4019cbb2:0x3ffb3910 0x4012d318:0x3ffb3930 

  #0  0x40083cbd:0x3ffb3780 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:402
  #1  0x4008ece5:0x3ffb37a0 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:128
  #2  0x400944d9:0x3ffb37c0 in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:85
  #3  0x40130a66:0x3ffb38f0 in tcp_recved at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c:979
      (inlined by) tcp_recved at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c:969  
  #4  0x4019cbb2:0x3ffb3910 in _tcp_recved_api(tcpip_api_call_data*) at lib/AsyncTCP/src/AsyncTCP.cpp:423
  #5  0x4012d318:0x3ffb3930 in tcpip_thread_handle_msg at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:172
      (inlined by) tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:154

ELF file SHA256: 0000000000000000

Rebooting...
```