Hieromon/AutoConnect #245: Samples don't compile

Greece2, 2020-08-05:
I tried all samples, but they don't compile.
Here are some of the error messages:

> W:\libraries\AutoConnect\src\AutoConnectUploadImpl.h:153:38: note: in expansion of macro 'FILE_WRITE'
>        uint8_t oflag = *mode == 'w' ? FILE_WRITE : FILE_READ;
>                                       ^
> In file included from W:\libraries\AutoConnect\src\AutoConnectAux.cpp:13:0:
> W:\libraries\AutoConnect\src\AutoConnectUploadImpl.h:153:60: error: expected ':' before ';' token
>        uint8_t oflag = *mode == 'w' ? FILE_WRITE : FILE_READ;
>                                                             ^
> W:\libraries\AutoConnect\src\AutoConnectUploadImpl.h:153:60: error: expected primary-expression before ';' token

IDE is 1.8.13, AutoConnect lib-version is 1.1.7

Hieromon, 2020-09-21:
I couldn't reproduce the issue. Setting the IDE's Preferences to log details of compiler messages can help you know what's going on in your environment.

Hieromon, 2020-10-15:
Closed due to no activity.

closed 2020-10-15 by Hieromon