Hieromon/AutoConnect #247: ESP boots faster than Router

Bagunceiro, 2020-08-16:
I have AutoConnect operating on a couple of ESP8266s - it generally works very well, but I have a problem. In the case of power restoration after an outage, for example, the router takes a lot longer than the ESP to reboot and the ESP enters into its SoftAP mode (as expected). However, I would have expected it to connect to the router when that becomes available but it does not appear to. Is this expected behaviour or do I have a fault somewhere?

ark02, 2020-08-24:
Also having the same issue here. Looking through other issues, there are easy ways to call WiFi.reconnect() when connection is dropped however there is not an easy way to get the device to search for and connect to past-connected AP's after it goes into SoftAP mode

There is one other issue with a potential solution:
https://github.com/Hieromon/AutoConnect/issues/153#issuecomment-557396970

However this seems clunky with all the ESP.restarts and I'm not sure how it will affect the default flow of the AutoConnect library

Unless a more creative solution is found I'm thinking to just watch for the APSTOP event triggers or something similar and then try calling WiFi functions to connect from AutoConnect creds

Hieromon, 2020-09-21:
@Bagunceiro Your encountered phenomenon is an assumed problem with AutoConnect::begin, and I recognize that its solution is a future subject.
A useful measure is to continue a valid SSID scan even in the captive portal and try to reconnect automatically inside AutoConnect if found known SSID. And I presented the skeleton of the implementation in the past, as ark02 commented.

We need to further consider the consistency with other options to determine the specification of this option, but I plan to enhance it with after v120.

Hieromon, 2020-11-09:
@Bagunceiro I prepared the AutoConnectConfig::autoReconnect option with enhancement in v120.
By combining it with AutoConnectConfig::portalTimeout, AutoConnect will catch the SSID broadcast after the router restarts. AutoConnect executes this reconnection asynchronously in the handleClient loop.
If possible, you should rate it and let me know what I need to improve.

ark02, 2020-11-09:
@Hieromon Thank you for this update! We have been struggling with this bug on our devices for quite some time 

To confirm, no additional setup is needed on our end to test this? Just download the latest AutoConnect Library and test router resets? 

Also, I assume we should remove the following: 
```
 void WiFi_events(WiFiEvent_t event, WiFiEventInfo_t info){
  Serial.printf("[WiFi-event] event: %d\n", event);
  switch (event) {
    case SYSTEM_EVENT_STA_DISCONNECTED:
            Serial.println("Disconnected from WiFi access point");
            WiFi.reconnect();
            break;
  }
}
```

Hieromon, 2020-11-10:
@ark02 You need to pull the [enhance/v120](https://github.com/Hieromon/AutoConnect/tree/enhance/v120) branch newest commit to your local repository. It has not been fully tested yet, but it is at a level that can be evaluated in simple use cases. 
The skeleton of the test sketch looks like this:

```cpp
AutoConnect portal;
AutoConnectConfig config;

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();

  config.autoReconnect = true;
  config.reconnectInterval = 6;  // Time interval to try to connect, a unit is 30[sec] by default. So, this setting actually specifies 3[min].
  portal.config(config);
  if (portal.begin()) {
    config.autoRise = false;  // Avoid launching the captive portal during handleClient loop.
    portal.config(config);
    Serial.printf("AP connected:%s, ", WiFi.SSID().c_str());
  }  
  Serial.println("Exit from the setup, rush into the loop");
}

void loop() {
  // Here, the router may restart during the handleClient loop.
  // Attempts reconnection every 3 minutes.
  portal.handleClient();
}
```

If retainPortal and autoRise are enabled when a WiFi disconnect is detected inside the handleClient loop, then SoftAP and the DNS server will be dynamically launched to start the captive portal. If you don't want this behavior, disable those settings as the false.
Also, please turn on [AC_DEBUG](https://github.com/Hieromon/AutoConnect/blob/8651c1a90390ad70947c5d8ea3378260fb759b45/src/AutoConnectDefs.h#L14) macro definition and get a debug trace.

Hieromon, 2020-12-02:
Merged #280

closed 2020-12-02 by Hieromon