Hieromon/AutoConnect #252: Keep ESP32 AutoConnect Captive Portal Always Running

TryingToGITthis, 2020-09-01:
I have an ESP32 that is running in softap mode with AutoConnect. I'm trying to get the captive portal to always be active, even if the client portion is connected to an AP. The current problem is once the ESP32 establishes the client connection with the router or after the initial timeout period, the captive portal goes away and you have to manually enter the IP address of the of the softap in the web browser.

Is there a setting I'm missing that allows this to happen?

`Config.autoRise = true;
  Config.immediateStart = true;
  Config.portalTimeout = 30 * 1000
  Config.retainPortal = true;
  Config.title = "ESP32 AP"
  Config.homeUri = "/";
  portal.config(Config);`

Hieromon, 2020-09-21:
AutoConenct has the effect of stopping captive portal detection (a trick that uses a DNS server) once it connects to the router as a client. Therefore, even if you specify retainPortal, the automatic pop-up will not be displayed when the timeout occurs.
It is possible to continue the internal DNS server by changing the meaning of the retainPortal option. However, some users do not want such behavior.

You may not want to modify the library code privately, but if you modify AutoConnect.cpp as follows, your request will be met for the time being.

**Current code**
https://github.com/Hieromon/AutoConnect/blob/d39545ce13c12c3e8fddef599b7325b276707ac9/src/AutoConnect.cpp#L239-L243

**Modify to**
```cpp
if (cs) { // Comment out the following 3 lines
//   _dnsServer->stop(); 
//   _dnsServer.reset(); 
//   AC_DBG("DNS server stopped\n"); 
} 
```

TryingToGITthis, 2020-09-22:
I made the modification locally but nothing happened. The pop-up never happens. Even if the device doesn't connect to the router, it doesn't pop-up after the config period. If it is connected, it still doesn't pop-up.

Hieromon, 2020-09-24:
In that state, the captive portal is valid. But, your expectation is that the captive portal screen will pop up automatically, right? If so, if the ESP32 module is already connected to the router as STA (that is, the Local IP has already been assigned), then auto-pop will not occur.

TryingToGITthis, 2020-09-24:
Thank you for the quick response. Right now it's not even popping up when it isn't connected. Is it possible I could modify the code to do that or it just can't be done?

Hieromon, 2020-09-25:
Producing the automatic pop-up as expected comes down to how it is implemented with the client device OS. AutoConnect only provides a trigger to launch its behavior.

The client device OS that implements automatic pop-up will begin to verify the internet transparency for an introduced TCP connection when it does not have any TCP connection. That's the moment you choose esp32ap on your device's WiFi selection screen.

Generally, when the OS has a new TCP connection, it issues an HTTP-GET to a unique URL and determines that it is Internet transparent if there is a default response. If the response is not the default content, it will auto-pop up its returned HTML as the captive portal content.

AutoConnect is trapping this mechanism. AutoConnect::begin internally launches a DNS server and responds to all name resolution requests with SoftAP IP address. Then, the HTTP-GET for Internet transparency verification issued from the client device OS arrives at the ESP module, and AutoConnect responds HTML as the captive portal screen. It will be popped up automatically.

This trapping mechanism has the theme to realize your wishes and is occurred by the _captivePortal function in AutoConnect.cpp.

https://github.com/Hieromon/AutoConnect/blob/d39545ce13c12c3e8fddef599b7325b276707ac9/src/AutoConnect.cpp#L771-L783

If the HTTP request arriving at SoftAP is Internet transparency verification from the client device, it will be redirected to the local IP of the ESP module.
In other words, the theme is how to redirect HTTP requests from clients connected to SoftAP. Can you specify it explicitly? Of course, it should not interfere with HTTP requests from other connected clients.

Hieromon, 2020-11-09:
Closed due to no activity more one month.

closed 2020-11-09 by Hieromon