Hieromon/AutoConnect #253: Best practice to show dynamic values as label

Summer88, 2020-09-02:
First of all thank you for sharing this great library!

I'm triyng to dynamize a label on a Button on the web page, started from this [example](https://github.com/Hieromon/AutoConnect/blob/master/examples/HandleClient/HandleClient.ino):
```

String StatoPin() // should return the label
{
String content = "Hello";
content += String(F(") : <span style=\"font-weight:bold;color:"));
content += digitalRead(BUILTIN_LED) ? String("Tomato\">ON") : String("SlateBlue\">OFF");
content += String(F("</span>"));
return content;
}
 
ACSubmit(testbutton, "StatoPin", "/io?v=low"); //compile fine as it is a fixed string
ACSubmit(testbutton, StatoPin(), "/io?v=low"); // trow error on compilation
ACSubmit(testbutton, StatoPin_00, "/io?v=low"); // trow error on compilation

AutoConnectAux mqtt_save(URI, "Page Label", false, {
  testbutton,
  clear
});

void loop{
StatoPin_00=StatoPin();
...
}

```
I'm looing at the API [doc](https://hieromon.github.io/AutoConnect/apielements.html#autoconnectsubmit)  but didn't understand why names differs from examples i.e. ACSubmit is the same as AutoConnectSubmit?
Any advice is really appreciated.

Hieromon, 2020-09-24:
> ACSubmit is the same as AutoConnectSubmit?

The entity shown by those two is the same. ACSubmit is expanded by the macro to a variable of AutoConnectSubmit type.
https://hieromon.github.io/AutoConnect/acelements.html#how-to-coding-for-the-elements

The button label of the submit button on the web page is taken from the [**value**](https://hieromon.github.io/AutoConnect/acelements.html#value_6) of AutoConnectSubmit. To change the label dynamically, access the **value** member of the AutoConnectSubmit variable declared by ACSubmit.

```cpp
ACSubmit(testbutton, "ON", "/io?v=low");
...
void buttonOff(void) {
  testButton.value = "OFF";
  testButton.uri = "/io?v=high";
}

void buttonOn(void) {
  testButton.value = "ON";
  testButton.uri = "/io?v=low";
}
```

It's unclear what kind of behavior you specifically envision as "dynamize a label", but if you want to change the label on the web page without causing page transitions, you need to embed JavaScript for that using AutoConnectElement.

Hieromon, 2020-11-09:
Closed due to no activity for more than a month.

closed 2020-11-09 by Hieromon