Hieromon/AutoConnect #254: ISSUES WITH SAVING CREDENTIALS!

Tejas-MD, 2020-09-02:
Hi, as the title suggests, the library fails to save credentials. This is the reason it fails to reconnect upon portal timeout and reboot(Power Re-connection). I am sure it's the credentials saving issue cause I tried `Wifi.begin()` at the time of re-connection and it too fails.

Surprisingly, it reconnects pretty well when reset. Please look into this.

And Yes, I have :

`Config.autoReconnect = true;`

and:

`Config.autoSave=AC_SAVECREDENTIAL_AUTO;`

OptumCS, 2020-09-03:
I found this flowchart helpful in figuring out how to configure AutoConnect and when it tries to reconnect using last good credentials, etc.  https://hieromon.github.io/AutoConnect/lsbegin.html

Also, I created a simple function to dump the credentials from EEPROM for troubleshooting.  Maybe this can give you some insight or verify what is being saved (or not saved) in the EEPROM:
```
#include <AutoConnectCredential.h>

#define CREDENTIAL_OFFSET 0

//--------------------------------------------------------------
// Dump the MAC Address from a 6 byte array

String getMACAddress(byte bssid[6]) {
    
    return  String(bssid[5], HEX) + ":" + 
            String(bssid[4], HEX) + ":" + 
            String(bssid[3], HEX) + ":" + 
            String(bssid[2], HEX) + ":" +  
            String(bssid[1], HEX) + ":" + 
            String(bssid[0], HEX);

} // END: getMACAddress

//----------------------------------------------------------------
// Dump the AutoConnect WiFi credential entries from EEPROM

void dumpCredentials(void) {
    
    AutoConnectCredential acCredential(CREDENTIAL_OFFSET);
    station_config_t acConfig;

    // Get the number of config entries
    uint8_t count = acCredential.entries(); 

    Serial.println(F("======="));     
    
    if ( count > 0 ) {

        for (int8_t i = 0; i < count; i++) {  
            
            acCredential.load(i, &acConfig);
            
            Serial.print(F("Credential Number: "));
            Serial.println(i + 1);
            Serial.print(F("SSID: "));
            Serial.println(String((char*)acConfig.ssid));
            Serial.print(F("Password: "));
            Serial.println(String((char*)acConfig.password));
            Serial.print(F("BSSID: "));
            Serial.println(getMACAddress(acConfig.bssid));

            if (acConfig.dhcp == 0) {  // NOTE: 0 = DHCP, 1 = Static IP
                Serial.println(F("Using DHCP"));
            }
            else {
                Serial.println(F("STATIC IP ------->"));
                Serial.print(F("Static IP: "));
                Serial.println(IPAddress(acConfig.config.sta.ip));
                Serial.print(F("Static Gateway: "));
                Serial.println(IPAddress(acConfig.config.sta.gateway));
                Serial.print(F("Static Netmask: "));
                Serial.println(IPAddress(acConfig.config.sta.netmask));
                Serial.print(F("Static DNS1: "));
                Serial.println(IPAddress(acConfig.config.sta.dns1));
                Serial.print(F("Static DNS2: "));
                Serial.println(IPAddress(acConfig.config.sta.dns2));
            }
            
            if (i < (count - 1)) {
                Serial.println(F("======="));     
            }
        }
    }
    else {
        Serial.println(F("No WiFi Credentials Found"));
    }
    Serial.println(F("======="));   

} // END: dumpCredentials
```

Tejas-MD, 2020-09-04:
Hi, thanks for your fantastic solution, but it was PlatformIO's bug, showed me a new update today that fixed everything... 
Close the issue please.