Hieromon/AutoConnect #255: HTML Numeric input field

Summer88, 2020-09-02:
For new versions can you please implement also an HTML numeric input field? 

Something like:
`ACInputNumber(numerictag, "Label", "min", "max")`

Thank you

Hieromon, 2020-09-24:
I think we have two options for enhancing this element.
One is to add a new element that accepts only numeric attributes as your suggestion. The other is to extend the current AutoConnectInput to add an option to enable only numeric attributes. it will generate the `<input type="number" min="..." max="...">` tag in any case.

However, the number type HTML element allows input such as `1+1+1` depending on the browser. So, it comes right down to validate the input value with the sketch.

In conclusion, it is better to use the [**pattern**](https://hieromon.github.io/AutoConnect/acelements.html#pattern) attribute of AutoConnectInput and give a regular expression that allows only numbers. In this way, the sketch works only requires validation of the range of values, and does not require numerical character checking of the input data.

Hieromon, 2020-10-15:
This request has no advantage over the method of constraining the value of AutoConnectText by using a regular expression with the pattern attribute.

closed 2020-10-15 by Hieromon