Hieromon/AutoConnect #256: Custom AP name and passphrase not working

rn00175, 2020-09-03:
Here's a snippet of my sketch:
```
const char* ap = "customap";
const char* pass = "custompass";

WebServer server;
AutoConnect Portal(server);
AutoConnectConfig Config(ap, pass);

void handle_rootPage() {
  server.send(200, "text/html", rootHtml());
}

void setup() {
  // put your setup code here, to run once:
  
  Config.menuItems = AC_MENUITEM_CONFIGNEW | AC_MENUITEM_OPENSSIDS | AC_MENUITEM_DISCONNECT | AC_MENUITEM_UPDATE | AC_MENUITEM_HOME;
  Config.ota = AC_OTA_BUILTIN;
  Config.autoReset = true;
  
  Portal.config(Config);
  
  delay(1000);
  Serial.begin(115200);
  Serial.println();

  server.on("/", handle_rootPage);
  if (Portal.begin()) {
    Serial.println("WiFi connected: " + WiFi.localIP().toString());
  }
}

void loop() {
  // put your main code here, to run repeatedly:
  Portal.handleClient();
  //delay(1000);
```

Initializing `AutoConnectConfig` wit default parameters works just fine. But when I set custom AP name and passphrase, the network shows up as gibberish when scanning with my phone's WiFi and won't let me reconnect. Any clues on what I'm doing wrong?

OptumCS, 2020-09-03:
The following works for me:

```
// Device specific web server
ESP8266WebServer g_webServer;

// AutoConnect captive portal object
AutoConnect acPortal(g_webServer);

// AutoConnect captive portal configuration object
AutoConnectConfig acConfig;

#define CAPTIVE_PORTAL_SSID "ThisLanIsMyLan";
#define CAPTIVE_PORTAL_PASSWORD "Not1234";

void setup() {

	// Lots of setup code here...
	
	acConfig.apid = CAPTIVE_PORTAL_SSID;
	acConfig.psk = CAPTIVE_PORTAL_PASSWORD;
	
	// Other AutoConnect Configuration here...
	// See https://hieromon.github.io/AutoConnect/apiconfig.html
	
        // **** Set the configuration to AutoConnect 
	acPortal.config(acConfig);
	
	// Start the captive portal
        if (acPortal.begin()) {
		// ---> WiFi Connect SUCCESS
	}
	else {
		// ---> WiFi Connect FAIL
	}

} // END: setup()
```
I don't set them when I construct the config object.  I set then like all other config properties.  Hope this helps!

rn00175, 2020-09-03:
> The following works for me:
>
> ```
> // Device specific web server
> ESP8266WebServer g_webServer;
>
> // AutoConnect captive portal object
> AutoConnect acPortal(g_webServer);
>
> // AutoConnect captive portal [...]

Didn't think about this one. I'll give it a shot. Thanks for the suggestion!

rn00175, 2020-09-03:
@OptumCS Tried your suggestion and a couple other ways to try and configure it. Doesn't seem to work. The problem persists whenever I don't use the default configuration, whether it's setting the AP and PSK, or just AP, or just PSK. Wondering if it has to do with my DevKit.

This all feels weird considering the default constructor and custom constructor is identical apart from the AP and PSK section. If it did have something to do with my DevKit I would assume the default constructor wouldn't even work in the first place.

v-c, 2020-09-04:
How much space is your sketch using? What are your partition sizes? Are you sure you aren't running out of heap space and nothing else is corrupting the eeprom?

OptumCS, 2020-09-04:
@rn00175 All good points.  Yes, using the constructor or using the properties of the config object should be the same.  Sounds like something is getting munged in memory.  I'm using PlatformIO/Arduino framework with ESP8266 Wemos D1 mini and LOLIN D1 mini Pro boards.

rn00175, 2020-09-04:
@v-c The barebones sketch (Only implementing autoconnect) uses 917862 bytes (70%) of program storage space, where maximum is 1310720 bytes. The sketch that I integrate it to uses 943718 bytes (72%). Not a lot of difference.

Partition scheme is default, though I'm not entirely sure how big that is.

@OptumCS it does sound like it. Do you perhaps have any suggestions on how to take a peek at the memory to try and find where it went wrong?
A simple fix would probably just be modifying the library itself and its defined values but I hate doing that, especially when it shouldn't be necessary.

v-c, 2020-09-04:
How about a bare bones sketch with nothing else but a blinking LED? Does that exhibit similar behavior?

OptumCS, 2020-09-04:
I guess for the simple fix/hack option, you wouldn't really wouldn't need to modify the library.  The password and  SSID are defined in AutoConnect/src/AutoConnectDefs.h with #ifndef.  So as long as you define AUTOCONNECT_APID and AUTOCONNECT_PSK in your sketch before including AutoConnect, it should change the defaults.  Here is what is in /src/AutoConnectDefs.h
```
// SSID that Captive portal started.
#ifndef AUTOCONNECT_APID
#if defined(ARDUINO_ARCH_ESP8266)
#define AUTOCONNECT_APID  "esp8266ap"
#elif defined(ARDUINO_ARCH_ESP32)
#define AUTOCONNECT_APID  "esp32ap"
#endif // !ARDUINO_ARCH_ESP8266
#endif // !AUTOCONNECT_APID

// Password that Captive portal started.
#ifndef AUTOCONNECT_PSK
#define AUTOCONNECT_PSK   "12345678"
#endif // !AUTOCONNECT_PSK
```

rn00175, 2020-09-04:
@v-c nothing else but a blinking sketch uses 207765 bytes (15%). Like so:
```
#define RED_PIN 3

String foo = "bar";

void setup() {
  // put your setup code here, to run once:
  pinMode(RED_PIN, OUTPUT);
  digitalWrite(RED_PIN, LOW);
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(1000);
  digitalWrite(RED_PIN, HIGH);
  delay(1000);
  digitalWrite(RED_PIN, LOW);
}
```

Including `<AutoConnect.h>` without actually using it bumps this up to about 23%. Including all 3 required headers (AutoConnect, WebServer, and WiFi) doesn't seem to change anything.

However when the required objects are initialized i.e
```
WebServer server;
AutoConnect Portal(server);
```
Usage shoots up to 60%. Including the `Portal.handleClient()` in the loop bumps it up again to 69%.

@OptumCS Thanks for the small tip.

v-c, 2020-09-04:
Does that simple sketch show the behavior you see with your other sketch of autoconnect not working? If it does not show the same problem, you should close this issue as it likely has something to do with your sketch.

rn00175, 2020-09-04:
@v-c a bit lost by what you mean as behavior. If you're referring to the custom AP and Passphrase not working then yes, even on the basic blink sketch my microcontroller won't broadcast a proper SSID name and I can't connect to it, even when it's unsecured. The code snippet on the original post only contains code initializing the AutoConnect library.

closed 2020-09-08 by rn00175