Hieromon/AutoConnect #258: Feature Request: Custom Ticker Output

cyberjacob, 2020-09-15:
Would be rather useful if the "ticker" functionality in AutoConnect could be used with more complicated output devices.

For example, the M5Stack Atom doesn't have an onboard LED that can be controlled with a simple GPIO pin. It uses a WS2812b addressable RGB LED which needs a specific data stream to turn on/off and change colour. At the moment, it's not possible to use AutoConnect's ticker output with such devices without adding an external LED.

As a possible solution to this, AutoConnect could add callback functions to set the status LED state?

Hieromon, 2020-09-24:
I've considered the WS2812b datasheet, but I don't think it's enough to drive it, even if it implements a callback from AutoConnect. The AutoConnect's ticker handles the AutoConnectTicker class, which relies on a hardware timer but is already a callback function in itself.

I think using **attachInterrupt** to capture the ticker pin signal gives the sketch more flexibility than expecting a callback from AutoConnect to drive the WS2812b. This way also allows WS2812b to be properly started and terminated by the AutoConnect::begin, AutoConnect::onDetect exit, and AutoConnect::onConnect exit.
How about this?

Hieromon, 2020-10-15:
This topic is not relevant to WiFi connectivity and has presented a way to implement the request, so closed.

closed 2020-10-15 by Hieromon