Hieromon/AutoConnect #259: Automatic switch WIFI

ddonindia, 2020-09-19:
I have two WIFI access points A and B.  First I connected A then B. 
If I restart ESP32 it always try to connect to access point B and if its not available AutoConnect starts in AccessPoint mode. 

I am using https://github.com/Hieromon/AutoConnect/blob/b79028346eeedf8139b2e50467fbd5a21862bb77/examples/Credential/Credential.ino sketch  with `Config.autoReconnect = true` 

Can you please help ?

Hieromon, 2020-09-24:
This is strange. If autoReconnect is enabled, AutoConnect will scan the SSID that is being broadcast if the first WiFi.begin fails with the B access point and match it with the stored credentials. At this time, ESP32 should be able to connect to the A access point you mentioned.

Could you please refer to the documentation and enable [**AC_DEBUG**](https://hieromon.github.io/AutoConnect/faq.html#3-turn-on-the-debug-log-options) to get the log?

If you are using the PlatformIO, set **build_flags** in platformio.ini as below:

```ini
build_flags =
  -DCORE_DEBUG_LEVEL=5
  -DAC_DEBUG
```

For the Arduino IDE, edit the library source [**AutoConnectDef.h**](https://github.com/Hieromon/AutoConnect/blob/b79028346eeedf8139b2e50467fbd5a21862bb77/src/AutoConnectDefs.h#L14) directly and uncomment `// #define AC_DEBUG` line. Then specify `Verbose` at the `Core Debug Level:` in the Arduino IDE's Tools menu.

ddonindia, 2020-09-24:
Thanks I will try and report back.

ddonindia, 2020-09-30:
After erasing and reflash firmware its started working

closed 2020-09-30 by ddonindia