Hieromon/AutoConnect #262: Hardcoded Channel ID in rssiMQTT Examples

gclugston, 2020-09-24:
Great Library and examples,  I am still experimenting with it.  

In the rssiMQTT examples under Void handleRoot you have your ThingSpeak channel ID hard coded into the sketch.  When I start the ESP8266 and navigate to Home, it displays your ThingSpeak chart.  This was easily remedied by changing the hard coding on the sketch to my personal channel ID, but is there a way to read this from SPIFFS and display my ThingSpeak data on startup?

src=\"https://thingspeak.com/channels/454951/charts/1

Thank you

Hieromon, 2020-09-24:
Oops, I overlooked quite this mistake until you pointed out.

To display the correct chart of the channel ID entered on the custom web page, modify the mqttRSSI.ino source code as follows (with editing the handleRoot function):

The channel ID is included in the url constituent string of the `src` attribute in the `iframe` tag. It's `454951` as my channel ID, which is hard-coded as you pointed out. Change this 6-digit channel ID number to `{{CHANNEL}}` string.

```cpp
"<iframe width=\"450\" height=\"260\" style=\"transform:scale(0.79);-o-transform:scale(0.79);-webkit-transform:scale(0.79);-moz-transform:scale(0.79);-ms-transform:scale(0.79);transform-origin:0 0;-o-transform-origin:0 0;-webkit-transform-origin:0 0;-moz-transform-origin:0 0;-ms-transform-origin:0 0;border: 1px solid #cccccc;\" src=\"https://thingspeak.com/channels/{{CHANNEL}}/charts/1?bgcolor=%23ffffff&color=%23d62020&dynamic=true&type=line\"></iframe>"
```
Then, replace the `{{CHANNNEL}}` string with the input channel ID.

```cpp
void handleRoot() {
  String  content = ... ;
  content.replace("{{CHANNEL}}", channelId);  // Insert here
  WiFiWebServer&  webServer = portal.host();
  webServer.send(200, "text/html", content);
}

```

Also, this fix applies to **mqttRSSI_FS.ino** and **mqttRSSI_NA.ino** too, which will be fixed in v1.2.0.

Thank you for your contribution.

gclugston, 2020-09-24:
Thank you for your prompt response.  

Your change works great.

I also noticed the HOME, i.e. handleRoot page does not load completely if you have the sketch doing other things.  I am measuring temperature using DS18B20 and averaging readings.  So I added a 100ms delay at the end of the handleRoot function.  

I could use some help with my application.  If you ever freelance contact me.

Hieromon, 2020-09-25:
> i.e. handleRoot page does not load completely if you have the sketch doing other things.

I can't say for sure because I haven't observed the big picture of your sketch, but I don't think it has anything to do with AutoConnect. Because it is the WebServer library that calls handleRoot in response to an HTTP request. AutoConnect does not intervene in calling handlers registered by WebServer::on.

If you are using the DS18B20 based on mqttRSSI code, the structure of mqttRSSI is probably not appropriate. Does it conflict between TCP receive interrupts and OneWire processing?
In particular, if you just replace the getStrength function with the DS18B20 communication, you will probably have other problems. The timing of OneWire process of ESP8266 will be stable if it is separated from TCP communication as much as possible.

Also, it is not desirable to put a delay of 100 [ms] (which is a very long time for the Web Server) in the loop. It only has to slow down the time it takes to respond to HTTP requests from clients. The mqttRSSI is not a good example to use external sensor modules. Would I come up with a better example, let me include it in the library.

gclugston, 2020-09-25:
Thanks Hieromon,

Thank you for your reply.  I sent a message to your gmail account.  Please look at it and let me know if you are interested.

Hieromon, 2020-12-02:
Merged #280

closed 2020-12-02 by Hieromon