Hieromon/AutoConnect #265: problem accessing the wifi network

alemori, 2020-09-30:
On the screen where the wifi network is selected, the next icon and the apply icon do not respond either.

Hieromon, 2020-10-02:
Try to erase the flash completely, then let re-compile and upload.
Also, there are references to the [documentation](https://hieromon.github.io/AutoConnect/faq.html#still-not-stable-with-my-sketch) that you can see in advance. If the problem is not due to your environment, I need the AC_DEBUG log to proceed with the investigation.

alemori, 2020-10-03:
Thank you very much. How do I get the AC_Debug log?

Hieromon, 2020-10-05:
Uncomment the `#define AC_DEBUG` line in `libraries/AutoConnect/src/AutoConnectDefs.h` file shown as the below, re-compile and upload:

https://github.com/Hieromon/AutoConnect/blob/d39545ce13c12c3e8fddef599b7325b276707ac9/src/AutoConnectDefs.h#L13-L14

Also, your sketch should be open `Serial` with `setup()` function.

```cpp
void setup() {
  Serial.begin(115200);
...
```

Open the serial monitor with Arduino IDE, make sure the baud rate the serial monitor and the Serial are the same.
Then, you will see the AC_DEBUG log with the serial monitor at the sketch running.

alemori, 2020-10-06:
Thank you very much for the explanation. I will now paste the output of the serial monitor for analysis. On the mobile phone screen, even though I pressed the next button, it didn't show me any more networks.
[AC] Host:,URI:/generate_204,ignored
[AC] Detected application, connectivitycheck.gstatic.com, (IP unset)
[AC] Host:connectivitycheck.gstatic.com,URI:/generate_204,ignored
[AC] Detected application, connectivitycheck.gstatic.com, (IP unset)
[AC] Host:connectivitycheck.gstatic.com,URI:/_ac,generated:/_ac, allocated
[AC] Host:172.217.28.1,URI:/_ac,already allocated
[AC] Host:172.217.28.1,URI:/favicon.ico,ignored
[AC] Host:172.217.28.1,URI:/generate_204,ignored
[AC] Detected application, connectivitycheck.gstatic.com, (IP unset)
[AC] Host:connectivitycheck.gstatic.com,URI:/generate_204,ignored
[AC] Detected application, connectivitycheck.gstatic.com, (IP unset)
[AC] Host:connectivitycheck.gstatic.com,URI:/generate_204,ignored
[AC] Detected application, connectivitycheck.gstatic.com, (IP unset)
[AC] Host:connectivitycheck.gstatic.com,URI:/generate_204,ignored
[AC] Detected application, connectivitycheck.gstatic.com, (IP unset)
[AC] Host:connectivitycheck.gstatic.com,URI:/generate_204,ignored
[AC] Detected application, connectivitycheck.gstatic.com, (IP unset)
[AC] Host:connectivitycheck.gstatic.com,URI:/_ac/config,generated:/_ac/config, allocated
[AC] Host:172.217.28.1,URI:/_ac/config,already allocated
[AC] 15 network(s) found

[AC] Host:172.217.28.1,URI:/generate_204,ignored
[AC] Detected application, connectivitycheck.gstatic.com, (IP unset)
[AC] Host:connectivitycheck.gstatic.com,URI:/generate_204,ignored
[AC] Detected application, connectivitycheck.gstatic.com, (IP unset)
[AC] Host:connectivitycheck.gstatic.com,URI:/generate_204,ignored
[AC] Detected application, connectivitycheck.gstatic.com, (IP unset)
[AC] Host:connectivitycheck.gstatic.com,URI:/generate_204,ignored
[AC] Detected application, connectivitycheck.gstatic.com, (IP unset)
[AC] Host:connectivitycheck.gstatic.com,URI:/generate_204,ignored
[AC] Detected application, connectivitycheck.gstatic.com, (IP unset)
[AC] Host:connectivitycheck.gstatic.com,URI:/generate_204,ignored
[AC] Detected application, connectivitycheck.gstatic.com, (IP unset)
[AC] Host:connectivitycheck.gstatic.com,URI:/generate_204,ignored
[AC] Detected application, connectivitycheck.gstatic.com, (IP unset)
[AC] Host:connectivitycheck.gstatic.com,URI:/generate_204,ignored
[AC] Detected application, connectivitycheck.gstatic.com, (IP unset)
[AC] Host:connectivitycheck.gstatic.com,URI:/generate_204,ignored
[AC] Detected application, connectivitycheck.gstatic.com, (IP unset)

Hieromon, 2020-10-07:
This trace says that no request has arrived at AutoConnect. The JavaScript embedded in the AutoConnect config page may not be responding.
What happened if you access the config page from your PC instead of a Cell Phone? Does it work?

alemori, 2020-10-08:
If I run it from my PC, the buttons don't work either. The screen is completely displayed, what doesn't respond are the buttons.
The module I'm using is a 4mb mini pro wemos. I advance this answer, but I'm going to do the complete test and complete the data again. It's probably a matter of memory. I am very grateful for the effort to solve the problem.

alemori, 2020-10-08:
I send you the result. I changed the parameters and put the pagebuilder on

HQA⸮<E+⸮⸮p 8A⸮[AC] Current:
[AC] WiFi.config(IP=(IP unset), Gateway=(IP unset), Subnetmask=(IP unset), DNS1=(IP unset), DNS2=(IP unset))
[AC] DHCP client(STARTED)
[AC] WiFi.begin()
[AC] Connecting............................................................................⸮HpY<>dH⸮⸮i: ⸮⸮[AC] Current:
[AC] WiFi.config(IP=(IP unset), Gateway=(IP unset), Subnetmask=(IP unset), DNS1=(IP unset), DNS2=(IP unset))
[AC] DHCP client(STARTED)
[AC] WiFi.begin()
[AC] Connecting....................................................................................................time out IP:(IP unset)
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
[AC] SoftAP esp8266ap/12345678 Ch(1) IP:172.217.28.1 
[AC] http server started
[AC] DNS server started
[AC] Host:,URI:/connecttest.txt,ignored
[AC] Detected application, www.msftconnecttest.com, (IP unset)
[AC] Host:www.msftconnecttest.com,URI:/connecttest.txt,ignored
[AC] Detected application, www.msftconnecttest.com, (IP unset)
[AC] Host:www.msftconnecttest.com,URI:/redirect,ignored
[AC] Detected application, www.msftconnecttest.com, (IP unset)
[AC] Host:www.msftconnecttest.com,URI:/_ac,generated:/_ac, allocated
[AC] Host:172.217.28.1,URI:/_ac,already allocated
[AC] Host:172.217.28.1,URI:/msdownload/update/v3/static/trustedr/en/authrootstl.cab,ignored
[AC] Detected application, ctldl.windowsupdate.com, (IP unset)
[AC] Host:ctldl.windowsupdate.com,URI:/_ac,generated:/_ac, allocated
[AC] Host:172.217.28.1,URI:/_ac,already allocated
[AC] Host:172.217.28.1,URI:/_ac/config,generated:/_ac/config, allocated
[AC] Host:172.217.28.1,URI:/_ac/config,already allocated
[AC] 15 network(s) found

[AC] Host:172.217.28.1,URI:/connecttest.txt,ignored
[AC] Detected application, www.msftconnecttest.com, (IP unset)
[AC] Host:www.msftconnecttest.com,URI:/connecttest.txt,ignored
[AC] Detected application, www.msftconnecttest.com, (IP unset)
[AC] Host:www.msftconnecttest.com,URI:/connecttest.txt,ignored
[AC] Detected application, www.msftconnecttest.com, (IP unset)
[AC] Host:www.msftconnecttest.com,URI:/connecttest.txt,ignored
[AC] Detected application, www.msftconnecttest.com, (IP unset)
[AC] Host:www.msftconnecttest.com,URI:/board/index.php,ignored
[AC] Detected application, digistump.com, (IP unset)
[AC] Host:digistump.com,URI:/_ac,generated:/_ac, allocated
[AC] Host:172.217.28.1,URI:/_ac,already allocated
[AC] Host:172.217.28.1,URI:/chrome-variations/seed,ignored
[AC] Detected application, clientservices.googleapis.com, (IP unset)
[AC] Host:clientservices.googleapis.com,URI:/connecttest.txt,ignored
[AC] Detected application, www.msftconnecttest.com, (IP unset)
[AC] Host:www.msftconnecttest.com,URI:/connecttest.txt,ignored
[AC] Detected application, www.msftconnecttest.com, (IP unset)

Hieromon, 2020-10-09:
This trace has no PageBuilder log, only with AC_DEBUG. Enable **PB_DEBUG** to get the PageBuilder log as I mentioned https://github.com/Hieromon/AutoConnect/issues/265#issuecomment-704635996.
Reference: https://hieromon.github.io/AutoConnect/faq.html#fn:2

Also, you can also check the HTML content transmitted from the wemos using your PC's client browser. As I can guess from the log traces you posted, you can obtain transmitted HTML using Windows client.
For MS Edge, use the DevTools to dump the transmitted HTML by press `Ctrl` + `Shift` + `I` on the client machine. https://docs.microsoft.com/en-us/microsoft-edge/devtools-guide-chromium
For Chrome, to dump the transmitted HTML by press `Ctrl` + `U` on the page.

The `Apply` button on the Config NEW page is processed by the JavaScript placed at the end of the submitted HTML. The reason the button doesn't respond is that the HTML submission may not be complete. You can check if the HTML sent using the client's DevTools is complete as `<html>` ~ `</html>`.
If it is incomplete, the Apply and Next buttons will not respond, and the cause is most suspected [the insufficient memory with PageBuilder](https://hieromon.github.io/AutoConnect/faq.html#some-autoconnect-page-is-cut-off).

alemori, 2020-10-09:
I made the changes, and now it's not taking the WiFi and after a while it's emitting a time out. Next, I put the debug of the trace that I captured
x40100ac8: malloc at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.cpp line 511
0x40227541: _vsnprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vsnprintf.c line 73
0x40100d20: calloc at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.cpp line 820
0x401003dc: pvPortZalloc at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/heap.cpp line 301
0x40105da8: spi_flash_read at ?? line ?
0x4024faed: wifi_param_save_protect_with_check at ?? line ?
0x40100205: __wrap_spi_flash_read at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_phy.cpp line 309
0x4024fbdb: system_param_save_with_protect at ?? line ?
0x4024fbbe: system_param_save_with_protect at ?? line ?
0x402500be: wifi_station_ap_number_set at ?? line ?
0x40100205: __wrap_spi_flash_read at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_phy.cpp line 309
0x402502de: wifi_station_ap_number_set at ?? line ?
0x4025038e: wifi_station_ap_number_set at ?? line ?
0x4025fa40: node_remove_from_list at ?? line ?
0x4025035d: wifi_station_ap_number_set at ?? line ?
0x402503c4: wifi_station_set_config at ?? line ?
0x40203630: ESP8266WiFiSTAClass::disconnect(bool) at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 385
0x4020360c: ESP8266WiFiSTAClass::disconnect(bool) at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 378
0x40105d00: spi_flash_erase_sector at ?? line ?
0x40215d94: IPAddress::toString() const at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/IPAddress.cpp line 173
0x401007f4: umm_free_core at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.cpp line 316
0x40100a93: free at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.cpp line 362
0x4024f7d6: wifi_get_opmode at ?? line ?
0x401003ec: vPortFree at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/heap.cpp line 308
0x40250619: wifi_station_set_auto_connect at ?? line ?
0x40250665: wifi_station_set_auto_connect at ?? line ?
0x40204635: AutoConnect::_disconnectWiFi(bool) at C:\Users\Alejandro\Documents\Arduino\libraries\AutoConnect\src/AutoConnect.cpp line 1184
0x402078b7: AutoConnect::begin(char const*, char const*, unsigned long) at C:\Users\Alejandro\Documents\Arduino\libraries\AutoConnect\src/AutoConnect.cpp line 180
0x40223318: WiFiUDP::read() at ?? line ?
0x40215d94: IPAddress::toString() const at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/IPAddress.cpp line 173
0x401007f4: umm_free_core at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.cpp line 316
0x40100ac8: malloc at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.cpp line 511
0x40219ced: uart_flush at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/uart.cpp line 544
0x40219fd5: uart_init at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/uart.cpp line 676
0x40219ff7: uart_init at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/uart.cpp line 668
0x40216e14: String::copy(char const*, unsigned int) at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/WString.cpp line 214
0x40216ff8: String::operator=(String const&) at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/WString.cpp line 262
0x40207b6e: AutoConnect::begin() at C:\Users\Alejandro\Documents\Arduino\libraries\AutoConnect\src/AutoConnect.cpp line 62
0x40201296: setup at C:\Users\Alejandro\Documents\Arduino\sordo_autoconnect/sordo_autoconnect.ino line 160
0x402018a4: handleRoot() at C:\Users\Alejandro\Documents\Arduino\sordo_autoconnect/sordo_autoconnect.ino line 104
0x4022194c: std::_Function_base::_Base_manager ::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\alejandro\documents\arduino\hardware\esp8266com\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\4.8.2/functional line 1934
0x40221938: std::_Function_handler ::_M_invoke(std::_Any_data const&) at c:\users\alejandro\documents\arduino\hardware\esp8266com\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
0x402018a4: handleRoot() at C:\Users\Alejandro\Documents\Arduino\sordo_autoconnect/sordo_autoconnect.ino line 104
0x4022194c: std::_Function_base::_Base_manager ::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\alejandro\documents\arduino\hardware\esp8266com\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\4.8.2/functional line 1934
0x40221938: std::_Function_handler ::_M_invoke(std::_Any_data const&) at c:\users\alejandro\documents\arduino\hardware\esp8266com\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
[... cut ...] /functional line 1934
0x40221938: std::_Function_handler ::_M_invoke(std::_Any_data const&) at c:\users\alejandro\documents\arduino\hardware\esp8266com\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
0x40218024: loop_wrapper() at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 177
0x40100da5: cont_wrapper at C:\Users\Alejandro\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/cont.S line 81

Hieromon, 2020-10-09:
> I made the changes, and now it's not taking the WiFi and after a while it's emitting a time out. Next, I put the debug of the trace that I captured

What trace is this? Such information has no meaning as I haven't seen your sketch.
As I have repeatedly asked, please refer to the [documentation](https://hieromon.github.io/AutoConnect/faq.html#still-not-stable-with-my-sketch) and provide the information necessary for the analysis. The following is a quote from the [documentation](https://hieromon.github.io/AutoConnect/faq.html#still-not-stable-with-my-sketch).

>  please make your question comprehensively, not a statement. Include all relevant information to start the problem diagnostics as follows:
>
> - Hardware module
> - Arduino core version Including the upstream commit ID if necessary
> - Operating System which you use
> - Your smartphone OS and version (Especially for Android)
> - Your AP information (IP, channel) if related
> - lwIP variant
> - Problem description
> - If you have a STACK DUMP decoded result with formatted by the code block tag the Sketch code with formatted by the code block tag (Reduce to the reproducible minimum code for the problem)
> - Debug messages output (Including arduino core)

This place is for improving the AutoConnect facility and quality. Sometimes, I may also help debug a particular individual, but it is to improve the AutoConnect library, which can benefit many other users. However, Here is not a free user support center.
From the logs you provided, AutoConnect seems to be working fine. If you need to proceed with the problem analysis, do the problem isolation yourself. In other words, it degenerates your sketch to the minimum sketch code that can reproduce the problem. And if you can reproduce the problem, please post the code and debug log.

Thank you.

Hieromon, 2020-10-15:
Closed due to not a library issue.

closed 2020-10-15 by Hieromon