Hieromon/AutoConnect #267: DNS troubles when AP_STA mode

eldiamond2, 2020-10-21:
Hello! Here's my issue:
After configuration thru web-browser at _ac/config page, ESP32 stays in APSTA mode.
PubSubClient when trying to connect catch error:
```
Reconnecting to MQTT 
Client id: [%Client-name]
[E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for mqtt.[hostname].com
```

Hieromon, 2020-10-22:
Use `WiFi.mode(WIFI_STA)` after WiFi client connection established. Generally, it can be determined by the return code of `AutoConnect::begin`.

However, Why does WiFi.hostByName fail in AP_STA mode?
After establishing a connection with the access point as a WiFi client (that is, the IPs have configured), AutoConnect disconnects SoftAP and shuts down the internal DNS server. Does WiFi.hostByName fail in this state?
I tested minimal Sketch using WiFi.hostByName without the pubsubclient, it worked fine.

Hieromon, 2020-10-23:
Closed as it is not considered to be a library issue.
Post the reduced code which can be reproduced the problem, if necessary.

closed 2020-10-23 by Hieromon