Hieromon/AutoConnect #268: ¿Autoconnect just like AP? & ¿AutoConnect and Ethernet?

josjimenezja, 2020-11-05:
Hi @Hieromon , firts of all thank you very much for this great work.
I have two questions about it
1) Can we use the library just like AP ? This because we are sending data for another method that are no wifi (internet) in this case, but we need this library for the AP to configure the device.
2) If we have an ehternet module and we connecte to internet with ehternet, is it possible to use AutoConnect to configure through the ethernet connection?

Hieromon, 2020-11-09:
@josjimenezja I don't understand the meaning of the question you mentioned.
> we need this library for the AP to configure the device.

That said, you're probably misunderstanding AutoConnect. AutoConnect establishes an IP data link. TCP configuration is the role of the WiFi library. It does not join data links via Ethernet.

josjimenezja, 2020-11-09:
Hi @Hieromon,
1) Could be use this library to configure a device but not to connect to a wifi network? This is because we are sending data by radiofrequency and we need to configure the device, but the library have a problem because if we don't connect the device to a wifi network, the program was be block in the setup().
2) About the second question, this is because we are created another device like a gateway and we are consider it to connections to internet options, the firts is wifi and the second is ethernet. The idea to have this options surge because in somecase we can't to connect to a wifi network but it is possible to connect to a ethernet network so i would like to be able to configure or access autoconnect through the IP tathe the ethernet connection provides and that the program does not block in the setup.

Hieromon, 2020-11-09:
I'm still not sure.
What is the device?, Is it an ESP module that implements a sketch that incorporates AutoConnect?
And, Is the WiFi network you mentioned the wireless LAN with an existing access point as an infrastructure?
It seems to me that you are confusing IP data links with TCP connections. If you only want to extract the web interface that AutoConnect has, there is no easy way to separate it.
AutoConnect is open source. You can analyze from the source code how AutoConnect forms the web interface. I think that's enough.

> the program was be block in the setup().

There is also a way to not block. It depends on what conditions you want to open the captive portal. Your question is too vague.

Regarding the second question, 

>  i would like to be able to configure or access autoconnect through the IP tathe the ethernet connection

Configure to what?
How does the TCP stack associated with an Ethernet connection combine with the WebServer instance formed by AutoConnect? SoftAP and STA of Espressif's ESP module are different components. They have individual IPs and are simply combined in software with a higher layer. They do not interface with each other. It's not a router.
The communication model you envision is more like a master-slave as ESP-NOW or ESP-MESH than a server-client. I recommend that you first investigate the ESP32 (even ESP8266) datasheets and SDK documentation. 

This topic is not suitable as an issue and will be closed

closed 2020-11-09 by Hieromon