Hieromon/AutoConnect #269: Run Function on button press in the web interface

1JonaB1, 2020-11-06:
Hello, I'm wondering what is the best way to detect a html button press? The documentation is saying nothing about it.
Thanks!

Hieromon, 2020-11-09:
No, the documentation mentions it. 
https://hieromon.github.io/AutoConnect/acelements.html#autoconnectbutton

AutoConnectButton willl generate an html tag as `<button type="button">` with an `action` attribute which is transfered from the action member.
If you want to use AutoConnectButton to send some HTTP requests to your own page just write such JavaScript code natively and embed it as AutoConnectElement. (Maybe it will be code using an XMLHttpRequest object) This is not the territory of how to use AutoConnectAux, rather the know-how of web applications. The usage for that is outside the scope of AutoConnect guidance.

Hieromon, 2020-11-09:
I have answered, so closing once.
To reopen, I need specific requirements or questions that Should be discussed, please.

closed 2020-11-09 by Hieromon