Hieromon/AutoConnect #270: ESP32 - Guru meditation - Save params of custum page in json file

Di3s3l, 2020-11-07:
Try to build a json custom page, for collect smtp username&password, but my esp32 crash and reboot after push "save" button of custom page. I have follow some of the example on sample sketch folder.... but don't understand where is the problema (bad hardware? or bad code?)

[MRCF_v6.zip](https://github.com/Hieromon/AutoConnect/files/5505601/MRCF_v6.zip)

Thank you

Hieromon, 2020-11-09:
I don't debug for any particular person. Please read the documentation [here](https://hieromon.github.io/AutoConnect/faq.html#still-not-stable-with-my-sketch) carefully to identify the problem. I will not diagnose unless there is a minimum code that can reproduce the problem.

Here is not the User Support Center. Also, uploading an unidentified file (moreover, zip!) Is extremely dangerous.
Closing immediately.

Di3s3l, 2020-11-09:
Sorry, the code was in the zip file, bocuse are sketch and json file.
I have check the documentation and the crash of esp is when execting that part of code

  File param = SPIFFS.open(PARAM_FILE, "w");
  Portal.aux("/mail_setting")->saveElement(param, { "mail_server", "mail_user", "mail_password", "mail_sender", "mail_to", "mail_prt", "mail_SSL" });
  param.close();

The element of mail_setting are loaded from a json file

Hieromon, 2020-11-09:
Read [here](https://hieromon.github.io/AutoConnect/faq.html#i-cannot-see-the-custom-web-page) and [here](https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository).
Turn on [AC_DEBUG](https://hieromon.github.io/AutoConnect/faq.html#3-turn-on-the-debug-log-options), and take the trace. Also, you should check your JSON documents with [JSON ASSISTANT](https://arduinojson.org/v6/assistant/).

Please provide the minimum code that can reproduce the problem. (Reduce to the reproducible minimum code for the problem.) In this case, reduce the number of AutoConnect Elements saved by saveElement to the minimum number that the problem can be reproduced. The Sketch code and the trace log with formatted by the code block tag.

You may not be using the AutoConnect API correctly. As a result, the JSON document is not loading properly, or anyone of the AutoConnect Elements is lost. You should check for errors properly.

Di3s3l, 2020-11-09:
Thank you. I will look at your link and provide more information as soon as possible :)

Di3s3l, 2020-11-10:
Hi,
I have attached sketch, json file of two aux page (settings collecetor and setting saver) and serial out of the esp32 with debug enabled.

[MRCF_v6.ino_.txt](https://github.com/Hieromon/AutoConnect/files/5519672/MRCF_v6.ino_.txt)
[mail_setting.json_.txt](https://github.com/Hieromon/AutoConnect/files/5519668/mail_setting.json_.txt)
[mail_save.json_.txt](https://github.com/Hieromon/AutoConnect/files/5519669/mail_save.json_.txt)
[log dump.txt](https://github.com/Hieromon/AutoConnect/files/5519671/log.dump.txt)

PS the instruction causing dump is:
  File param = SPIFFS.open(PARAM_FILE, "w");
  Portal.aux(AUX_MAILSETTING)->saveElement(param);     <==== THIS
  param.close();

Hieromon, 2020-11-11:
I found two possibilities for your error at first sight.

1. Which module of ESP32 are you using?
    > `[AC] PSRAM not found, JSON buffer allocates to the heap.`
    
    PSRAM cannot be detected from the ESP module. Nevertheless you have specified to use PSRAM.
    However, this error is not directly related to the crash.

2. The JSON description seems correct, but ArduinoJson is detecting a deserialization error.
    > `[AC] Deserialize:InvalidInput`
    > `[AC]   on hands`
    
    As a result, your sketch has lost AutoConnectAux. After that, you refer to its AutoConnectAux as you mention by the following, it becomes an undefined memory address reference:
    > Portal.aux(AUX_MAILSETTING)->saveElement(param);

    It will trigger the fatal errors as follows. (Quoted from the [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html) published by Espressif):
    > [**LoadProhibited, StoreProhibited**]()
    > This CPU exception happens when application attempts to read from or write to an invalid memory location.
    
    You have two deserialization errors, so maybe your sketch was trying to load JSON into `/mailsettings` and `/mail_save`?
    I have never seen your sketch, so it is not possible to pinpoint the exact cause by a diagnosis based solely on the information presented. I expect you are **not using the API correctly** for [loading JSON to AutoConnectAux](https://hieromon.github.io/AutoConnect/apiaux.html#load). (probably both `/mailsettings` and `/mail_save`)

I recommend you review the sketch and check the return code from [**AutoConnectAux::load**](https://hieromon.github.io/AutoConnect/apiaux.html#load). Probably the cause of the crash is there.

Di3s3l, 2020-11-11:
Thank you for the precius info, i have tryed to put the aux page code into the sketch and it works better. I think it's some related with my tool/ide that upload json into spiffs (will try to reformat it or build into the sketch a file uploader to be sure to be legacy to library spiffs.h im using to compile).

Thank you again for your great work on this software and forgive me to have bother you :)

closed 2020-11-11 by Di3s3l

Hieromon, 2020-11-11:
Since then, I checked your Sketch code, MRFC_v6.ino. It has three errors.

1. The correct notation for the HTTP content-type is not a backslash.
    ```cpp
    Server.send (200,"text\plain", content);
    ```
    It should be `text/plain`.

2. Wrong API sequence for MailSetting.load(AUX_MAILSETTING);
    ```cpp
    #define AUX_MAILSETTING "/mail_setting"
    ...
    MailSetting.load(AUX_MAILSETTING);
    ```
    Your code has specified a "/mail_setting" string as JSON document. This is the cause of the JSON deserialization error and is causing the crash.
    Refer to [AutoConnectAux::load](https://hieromon.github.io/AutoConnect/apiaux.html#load) function. You cannot specify a file name as an argument. It can be specified only JSON document itself as follows:
    ```cpp
    bool load(const String& in)
    ```
    ```cpp
    bool load(PGM_P in)
    ```
    ```cpp
    bool load(const __FlashStringHelper* in)
    ```
    ```cpp
    bool load(Stream& in)
    ```
3. `MailSave.load(AUX_MAILSAVE);` as same as above.

Di3s3l, 2020-11-11:
You are very kindly!
I'll have to pay more attention to writing the code next time.