Hieromon/AutoConnect #271: How put saveElement array into variable?

nikanet, 2020-11-08:
I would like put saveElement array into variable. 
page_data->saveElement(data_file, {**"Motor_speed", "angle", "hostname"**});
Especially get that array from another array. How do that? I spent couple hours but without success. I will be grateful for your help.

By the way - great project!

Hieromon, 2020-11-09:
I cannot understand what you mentioned that put saveElement array into variable.
Do you want to store the values of "Motor_speed", "angle", and "hostname" that exist as AutoConnectElements to another variables of the Sketch?
If so, You can find such an example at mqttRSSI.ino.

nikanet, 2020-11-09:
Thank's for quick answer.

I want put text "Motor_speed", "angle", "hostname" etc into lets say my global variables in separated file. How put this text into saveElement?

Hieromon, 2020-11-21:
> I want put text "Motor_speed", "angle", "hostname" etc into lets say my global variables in separated file

What is the separated file? source code as .ino?

Hieromon, 2021-01-04:
Closed as there has been no activity for over a month.
It will reopen with new posts.

closed 2021-01-04 by Hieromon