Hieromon/AutoConnect #272: When I use AutoConnectUpdate I can not send any request to the server !! ??

Abedi98, 2020-11-12:
When I put the following 2 lines of code for update by the server
`AutoConnectUpdate update("update.tik-box.ir",8001,"update");`

```
void setup(){
.....
update.attach(Portal); 
}
```

 I can no longer send any requests to the server and it seems that the module has no access to send information to the server.

sample of send GET request to server :

```
void sendLocalIP(){  
    std::unique_ptr<BearSSL::WiFiClientSecure>client(new BearSSL::WiFiClientSecure);
    client->setInsecure();
    HTTPClient https;
    https.setUserAgent("TIKBOX");
    String urlcr = baseUrl+"/api/saveLocalIP?api_token=" + String (token)+"&localip=" + IpAddress2String(WiFi.localIP())+"&ssid="+ String(WiFi.SSID());
    https.begin(*client, urlcr);
    int httpCode = https.GET();
    if(httpCode == 200){
      Serial.println("IP SEND TO SERVER");
    }else{
      Serial.println("NO SEND IP TO SERVER");
    }
}
```

Abedi98, 2020-11-13:
@Hieromon

Hieromon, 2020-11-21:
The entity of AutoConnectUpdate is the ESPHttpUpdate class, which has been really tested. Your problem may occur from not be the prerequisite environment for the ESPHttpUpdate class.
Check out the [documentation](https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html#http-server) scenarios. Also, it also contains a simple but fully functional update server script provided by the ESP8266 Aruino core team. You can easily try it.

Therefore, please you identify your problem. I'll address library-specific issues, but the content your post depends entirely on the mechanics you've built. I can't deal such post. You should isolate the problem and post with clear requirements for AutoConnect.
This post is just a kind of debug work request.

Hieromon, 2021-01-04:
Closed as there has been no activity for over a month.
It will reopen with new posts.

closed 2021-01-04 by Hieromon