Hieromon/AutoConnect #273: AP & STA MODE TOGETHER

Abedi98, 2020-11-20:
How to access ap mode when esp connected to wifi 

I want to access AP mode forever.

thanks

Abedi98, 2020-11-20:
```
if(WiFi.SSID() == ""){
    configtone();
    digitalWrite(D2, HIGH);
    digitalWrite(D3, HIGH);
    config.immediateStart = true;
    config.autoRise = true;
    
  }else{ 
     
    
    rootPage();
    Serial.println("after wireless availability : "+WiFi.SSID());

    loadParams(PARAM_FILE);     // Load the parameters in advance

    Serial.print("Setting soft-AP configuration ... ");
    Serial.println(WiFi.softAPConfig(local_IP, gateway, subnet) ? "Ready" : "Failed!");
  
    Serial.print("Setting soft-AP ... ");
    Serial.println(WiFi.softAP("ESPsoftAP_01","password",1,false,4) ? "Ready" : "Failed!");
  
    Serial.print("Soft-AP IP address = ");
    Serial.println(WiFi.softAPIP());

  }

  portal.config(config);
  portal.begin();          
```

But access point(ESPsoftAP_01) are not displayed in the WiFi list

Hieromon, 2020-11-21:
It's simple. Use WiFi.enableAP(true) and WiFi.softAP(local_IP, gateway, subnet) after AutoConnect::begin regardless of the return condition from AutoConnect::begin.

Abedi98, 2020-11-21:
> It's simple. Use WiFi.enableAP(true) and WiFi.softAP(local_IP, gateway, subnet) after AutoConnect::begin regardless of the return condition from AutoConnect::begin.

thanks @Hieromon 👍 

I want a new page to be displayed when I connect to the "NewAP-Device" access point and I do not want the AutoConnect portal to be displayed.

```
     portal.begin();
    
    WiFi.enableAP(true);
    Serial.print("Setting soft-AP configuration ... ");
    Serial.println(WiFi.softAPConfig(IPAddress(172,217,1,20), IPAddress(172,217,1,20), IPAddress(255,255,255,0)) ? "Ready" : "Failed!");

    Serial.print("Setting soft-AP ... ");
    Serial.println(WiFi.softAP("NewAP-Device","password") ? "Ready" : "Failed!");

```

Hieromon, 2020-11-21:
> I want a new page to be displayed when I connect to the "NewAP-Device" access point and I do not want the AutoConnect portal to be displayed.

What mean? Please clarify the specifications more. I cannot understand the specification you want.
In the first place, is it related to improving the AutoConnect library?

Hieromon, 2021-01-04:
Closed as there has been no activity for over a month.
It will reopen with new posts.

closed 2021-01-04 by Hieromon