Hieromon/AutoConnect #274: Captive portal ip and notification

chr1s692, 2020-11-20:
Hi,
currently i'm testing your library and it looks really great :).
I set a custom IP (192.168.4.1) for the captive portal, but then my mobile phone has not been redirected any more. Is that behaviour right or do i miss something?
Kind regards

Hieromon, 2020-11-21:
Which your mobile phone OS?

chr1s692, 2020-11-21:
I'm using Android 9.
Do you need a snippet of my test sketch or further details? I'm using the last / stable version 1.1.7 of autoconnect

Hieromon, 2020-11-21:
> Do you need a snippet of my test sketch or further details?

Please post to make sure. However, probably the cause is the below.
https://github.com/Hieromon/AutoConnect/issues/85#issuecomment-522297724
https://github.com/esp8266/Arduino/issues/7185
https://github.com/esp8266/Arduino/pull/5529

Now, Android has 192.168.0.0 is out of the captive portal detection range.

chr1s692, 2020-11-21:
Here's the snippet, i used one of your examples as template :):
```
void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();
  SPIFFS.begin();

  loadParams(mqtt_setting, args);
  
  if (uniqueid.checked) {
    config.apid = String("ESP") + "-" + String(GET_CHIPID(), HEX);
    Serial.println("apid set to " + config.apid);
  }

  config.apip = IPAddress(192,168,4,1);
  //config.gateway = IPAddress(192,168,4,1); //needed?
  //config.netmask = IPAddress(255,255,255,255);  //needed?
  config.menuItems = AC_MENUITEM_CONFIGNEW | AC_MENUITEM_OPENSSIDS | AC_MENUITEM_DISCONNECT ;
  portal.config(config);

  // Join the custom Web pages and register /mqtt_save handler
  portal.join({ mqtt_setting, mqtt_save });
  portal.on(AUX_SETTING_URI, loadParams);
  portal.on(AUX_SAVE_URI, saveParams);

  Serial.print("WiFi ");
  if (portal.begin()) {
    config.bootUri = AC_ONBOOTURI_HOME;
    Serial.println("connected:" + WiFi.SSID());
    Serial.println("IP:" + WiFi.localIP().toString());
  }
  else {
    Serial.println("connection failed:" + String(WiFi.status()));
    while (1) {
      delay(100);
      yield();
    }
  }
  WiFiWebServer&  webServer = portal.host();
  webServer.on("/", handleRoot);
  //webServer.on(AUX_CLEAR_URI, handleClearChannel);
}

void loop() {
  portal.handleClient();
}
```

chr1s692, 2020-11-21:
Ok, the notfication work when i set the ip to 192.169.4.1. Seems the problem is 192.168.0.0 on Android :/

Hieromon, 2020-11-21:
Thank you post the code. It has no problem.
In the past, I have seen the official announcement about changing the captive portal detection range since Android 7. I'm looking for it now, but I'm missing. I think it was a security consideration in my memory.
In any case, 192.168.0.0/16 cannot be used on Android now.

Hieromon, 2020-11-21:
This issue is caused by the Android OS and is not a subject that the library needs to handle. So I close.

closed 2020-11-21 by Hieromon