Hieromon/AutoConnect #276: Manage multiple AutoConnect devices on the same page

myhrmans, 2020-11-24:
Hey, are there any good solution using AutoConnect to manage multiple ESP32 connected over ethernet to the same network from the same page?
Like to see current status, upload ota update etc.

Thanks

Hieromon, 2020-11-24:
It's interesting to think of using WiFi connections to monitor or manage multiple ESP32 nodes scattered across an Ethernet-based local network, but each ESP32 node uses two different connections with the same protocol. It's uselessly redundant.

I think ESP-NOW is suitable for your requirement.
After all, AutoConnect is a connection-oriented server application that runs at a lower layer to provide the ability to connect to access points. I just happened to implement it assuming the HTTP model. 
In that case, it would be smarter to use ESP-NOW as the connectionless model to exchange the necessary data between the manager node and the edge node. I think the required data frame size fits within the 250 bytes constraint if only condition monitoring. It's not a big deal to implement.
What's more, the OTA updates feature is even simpler. When the node receives the MESH frame notifying the update, it simply downloads the new firmware from the update server. Such examples are included in the ESP8266 and ESP32 libraries.
You don't have to bother to complicate your implementation with AutoConnect.

Hieromon, 2020-11-24:
The content of the topic is not specific, and I answered. Close.

closed 2020-11-24 by Hieromon