Hieromon/AutoConnect #277: ColorPicker ?

JanHBade, 2020-11-26:
Hello,

I want to have an Color Picker like this: https://github.com/kitesurfer1404/WS2812FX/blob/master/examples/esp8266_webinterface/main.js.cpp in my root page of the portal.

Is that possible?

A dream of mine is the support as AutoConnect element

Hieromon, 2020-11-27:
I'm not sure what the UI will look like as the reference model you posted, but you can still create a color picker UI with a similar interface.

We have two choices for implementing the Color Picker UI.

1. Make a request handler
Make a page with an HTTP request handler according to the ESP8266WebServer class interface  specification. You can find it in the ESP8266WebServer library examples.
AutoConnect is not involved in this way. You just send the HTML containing JavaScript that realizes the color picker UI in response to the HTTP request.
I recommend you take this way If your root page doesn't have AutoConnect menu navigation. Its page will be suitable caused the legacy HTML page.
Also, AutoConnect documentation mentions how to navigate your legacy page from the AutoConnect menu.
    ```cpp
    #include <Arduino.h>
    #include <ESP8266WiFi.h>
    #include <ESP8266WebServer.h>
    #include <AutoConnect.h>

    ESP8266WebServer	server;
    AutoConnect	portal(server);
    AutoConnectAux	color("/color", "Color");

    const char HTML[] PROGMEM = R"(
    Here,
    You have to place an HTML and JavaScript for forming a web page as
    the color picker and can script it using native HTML and Javascript.
    If the script communicates with the server via XMLHttpRequest like
    the WS2812FX, you also need to program the handler to respond to it
    using the ESP8266WebServer request handler.
    )";

    void handleColor() {
      server.send_P(200, "text/plain", HTML);	
    }

    void setup() {
      server.on("/color", handleColor);
      portal.join({ color });
      portal.begin();
    }

    void loop() {
      portal.handleClient();
    }

    ```

2. Use AutoConnectElement
One of the easiest ways to implement a color picker is to use `input type="color"`. AutoConnectElement allows you to define HTML tags natively, so your sketch can have a custom web page with an `input type="color"` tag. But I'm not sure if you like the UI.
You can obtain the value of `input type="color"` defined by AutoConnectElement using the [PageArgument::arg](https://github.com/Hieromon/PageBuilder#string-pageargumentargstring-name) of the [PageBuilder](https://github.com/Hieromon/PageBuilder) function in the AutoConnectAux page handler.
    ```cpp
    #include <Arduino.h>
    #include <ESP8266WiFi.h>
    #include <AutoConnect.h>

    ESP8266WebServer  server;
    AutoConnect portal(server);
    AutoConnectConfig config;
    AutoConnectAux  picker;

    static const char COLOR_PICKER[] PROGMEM = R"(
    {
      "uri": "/color",
      "title": "Color",
      "menu": true,
      "element": [
        {
          "name": "text",
          "type": "ACText",
          "value": "COLOR",
          "style": "font-weight:bold;font-size:20px;color:#"
        },
        {
          "name": "picker",
          "type": "ACElement",
          "value": "<input type='color' id='picker' name='color'>"
        },
        {
          "name": "apply",
          "type": "ACSubmit",
          "value": "APPLY",
          "url": "/color"
        }
      ]
    }
    )";

    String onPicker(AutoConnectAux& aux, PageArgument& arg) {
      if (arg.hasArg("color")) {
        AutoConnectText&  text = aux["text"].as<AutoConnectText>();
        String  color = arg.arg("color");
        text.style.remove(text.style.indexOf('#'));
        text.style.concat(color);
      }
      return "";
    }

    void setup() {
      delay(1000);
      Serial.begin(115200);
      Serial.println();

      picker.load(FPSTR(COLOR_PICKER));
      picker.on(onPicker);
      portal.join({ picker });

      config.homeUri = "/color";
      config.bootUri = AC_ONBOOTURI_HOME;
      portal.config(config);
      portal.begin();
    }

    void loop() {
      portal.handleClient();
    }
    ```

JanHBade, 2020-11-27:
thank you, 2. looks nice and clean, thanks for the help

closed 2020-11-27 by JanHBade