Hieromon/AutoConnect #279: Ability for access AutoConnect portal when connected to Wifi?

FinlandApollo, 2020-11-29:
Hi all!

I must say, this has to be best implementation I've seen of this type, and best looking too! However, I must ask one thing, which isn't an issue: Am I able to have captive portal remain open that I could access through the IP while it's connected to my WLAN, e.g. future update of certain settings and etc? I didn't really understand or get it from the manual (advance use and FAQ).

If yes, does it then require any other codes than these in the `void setup():` 
`Config.retainPortal = true;`
`Portal.config(Config);`
`Portal.begin();`

And parts in the `void loop()`
`Portal.handleClient();`

Thank you and cheers!

Hieromon, 2020-12-01:
I'm not sure that what case you want SoftAP access, but your intended use case must fit into the [AutoConnect::begin logic sequence](https://hieromon.github.io/AutoConnect/lsbegin.html). As the document shows, the conditions for starting SoftAP are limited. Also, the conditions for leaving SoftAP are limited.
If you want SoftAP to work under other conditions, you need to incorporate the process into your sketch yourself.
Also, if you want to prevent the ESP module from becoming completely standalone, just monitor wl_status_t with a loop function and start SoftAP based on WiFi conditions. Its implementation is not difficult.

I have received many pointing out like this kind. However, all the indications misunderstand the use of SoftAP. I can't imagine a use case that requires two different IP connections to access one ESP module. (Except ESP-MESH, ESP-NOW)
What kind of case is it? Can you explain it to me?

FinlandApollo, 2020-12-01:
The case what I'm looking for, is a GUI where I can setup few parameters which will be used later on the code. So far I've tried to acchieve through AutoConnect's custom pages:
InfluxDB settings and fan speed settings.

The idea is that I can set those values up, then set the Wifi with AutoConnect's default page. Then I'd like to be able to access the Wifi page and also these custom pages (InfluxDB and Fan Speed) so for example, I could just change the database or IP address of the InfluxDB what will be used by the chip without having to disconnect or having to turn off the Wifi just to do that.

Am I look for a wrong solution for my use case?

Hieromon, 2020-12-02:
>  just change the database or IP address of the InfluxDB what will be used by the chip without having to disconnect or having to turn off the Wifi just to do that.

I'm not sure about this condition. What does it mean that without having to disconnect? A disconnect between where and where?
Does that mean you don't make the ESP module completely standalone? (In other words, do you always want to keep a WiFi route that can access the ESP module?, And does the route need to be open all the time?)

The transitions in the network diagram below represent the connection states that the ESP module may fall into. Which situation do you envision?

#### 1st Stage:

ESP module does not have any known access points.
It opens the captive portal and is in the WIFI_AP_STA state.
You can open the captive potal and configure the WiFi connection.

```
Network A                  NetWork B
~~~~~~~~~                  ~~~~~~~~~
<AP-a>===<Router>          <AP-b>===<Router>
  .     192.168.0.0/16             172.16.0.0/12
  .       |         |                |        |
  .   <InfluxDB>    |           <MQTT Broker> |
  .                 |                         |
  .             <Client-a>                <Client-b>
<ESP8266>
WIFI_STA: Not assigned
WIFI_AP: 172.217.28.1
  |
<Client-X> e.g. Your cellphone
WiFi Client for ESP8266.
```

#### 2nd Stage:

You had configured the WiFi connection with navigated by AutoConnect Config NEW menu.
ESP module joined to the Network A and is reachable to the InfluxDB over TCP or UDP via `Router-a`.
In this state, the ESP module has an IP address as a WiFi client and an IP address as a WiFi access point.
You can access the ESP module from `Client-a`, also from `Client-x` via SoftAP.
But it is not reachable to InfluxDB from `Client-x` directly.

```
Network A                  NetWork B
~~~~~~~~~                  ~~~~~~~~~
<AP-a>===<Router-a>        <AP-b>===<Router-b>
  |     192.168.0.0/16              172.16.0.0/12
  |       |         |                 |
  |   <InfluxD B>   |            <MQTT Broker>
  |                 |
  |             <Client-a>
<ESP8266>
WIFI_STA:192.168.1.10
WIFI_AP: 172.217.28.1
  |
<Client-x>
WiFi Client for ESP8266.
```

#### 3rd Stage:

ESP module lost IP connection of WIFI_STA. There are two possible causes. It either lost the WIFI signal between the ESP module and `Router-a` or disconnected by the ESP module itself. But WIFI_AP is still alive in this state.
A network diagram is the same as the 1st stage.

#### 4th Stage:

This stage is the most steady state of operation.
The ESP module participates in `Network A` and is accessible from `Client-a`. e.g. Your cell phone which to use the AutoConnect menu.
In this stete, You can remove the ESP module from the `Network A` and connect it to the `AP-b` using `AutoConnect Config NEW` or `Open SSIDs` menu via `Client-a`.

```
Network A                  NetWork B
~~~~~~~~~                  ~~~~~~~~~
<AP-a>===<Router-a>        <AP-b>===<Router-b>
[... 91 lines cut ...]

The above trap code will enable SoftAP, but automatic pop-up does not work because DNS remains down.
If the meaning you mentioned "the captive portal remains open" means "automatic pop-up works whenever you access the SoftAP launched by the ESP module", you misunderstand the meaning of the captive portal.
The role of the captive portal is to block network access until the client authenticates the user. It has nothing to do with the browser's automatic pop-up. AutoConnect simply realized its mechanism with a combination of SoftAP and DNS. Automatically popping up the browser is a useful feature of the client device OS, which is left to the implementation of that OS. Some OSs do not implement automatic browser pop-ups in the captive portal state.

Alternatively, allow the ESP module to be reset. After rebooting, the module will open the captive portal. Also automatic pop-up the browser.

```cpp
    else if (millis() - tm > 3*60*1000) {
      if (!(WiFi.mode() & WIFI_AP)) {
        ESP.reset();
      }
      else
        tm = 0;
    }
```

#### So, I ask questions again.

1. What situation do you think the ESP module should be in the captive portal state when the ESP module is in which connection state of WiFI? 

2. In the captive portal state, the ESP module cannot access the network. Just make an IP connection with the client device. It's different from SoftAP.
Does your requirement be an automatic browser pop-up with SoftAP is active? Does it have anything to do with network transparency?

3. Is there any reason you can't reset the ESP module?

Hieromon, 2021-01-04:
Closed as there has been no activity for over a month.
It will reopen with new posts.

closed 2021-01-04 by Hieromon