Hieromon/AutoConnect #281: Change the gateway page

Dwang7285, 2020-12-04:
Hi, 

Is that possible I can change the gateway page to config new IP page? So I can enter the SSID and Password at the beginning

Dora

Hieromon, 2020-12-04:
The API provided by the AutoConnect library as standard allows you to change the automatic pop-up page in the captive portal using the [AutoConnectConfig::bootUri](https://hieromon.github.io/AutoConnect/adnetwork.html#assign-user-sketchs-home-path) option. However, a bootUri setting changes affect AutoConnect as a whole.

If you want to change the page only when it pops up automatically with accessing to the captive portal, need change the library source code directly:

https://github.com/Hieromon/AutoConnect/blob/39ef333c42cd432e6a7ed697cce10f2193305500/src/AutoConnect.cpp#L1092

#### Change to:

```cpp
String location = String(F("http://")) + _webServer->client().localIP().toString() + String(AUTOCONNECT_URI_CONFIG);
```

**Note:**
This change is unreasonable measures.
In the captive portal state, the client device repeatedly issues HTTP requests to confirm the internet transparency. it will cause repeated WiFi scans to search for nearby SSIDs, which will interfere with the connection with the client device during the captive portal.
This behavior will cause you to lose the Config New page from your client device (which is probably your cell phone) in the captive portal state.
That's why I avoided displaying the Config New page on the auto-pop screen. But if you really want to do it, I have no reason to impede. (Cannot be a library standard)

tali, 2020-12-07:
Isn't it possible to already scan for nearby SSIDs when or even before opening the captive portal? This should be possible as soon as the device decides that no known SSID is available. Then the configuration page only needs to show this cached list.
For me it makes a lot of sense to directly point the user to the configuration page so that they can connect to their WLAN as soon as possible.

Hieromon, 2020-12-08:
> For me it makes a lot of sense to directly point the user to the configuration page so that they can connect to their WLAN as soon as possible.

@tali I agree and already have evaluated a prototype that was according to the primary purpose of this intention, as everyone thinks, prior to the full-fledged implementation. And I had two nasty problems as follows and got stuck with a complicated task:

1. ESP8266 shares the same WiFi channel with STA and AP, and ESP32 same too.

    [Active scanning](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#scan-type) is a fitting way to probe nearby WiFi signals. This behavior is a part of the featured component of STA. By default, channels 1 to 11 use active scans, and channels 12 to 13 use passive scans. 

    On the other hand, the device holding the captive portal maintains an IP connection to the AP component of the ESP module. Its IP connection depends on the WiFi data link on the particular channel specified in the AutoConnect::channel settings. And the AP component is not involved in WiFi active scanning by STA.

    Therefore, the STA's active scan may disconnect the AP's WiFi connection due to channel transition by scanning.

    The figure below is quoted from the [ESP32 Wi-Fi Programming Model](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#esp32-wi-fi-programming-model):

    [image]

    As you can see from the above figure, the AutoConnect model is consistently passive. It's natural because it's a server model.

    During the activity of AutoConnct, the event that the TCP stack catches is an HTTP request from the client. And there are two reasons why a client device will issue an HTTP request:

    a. The client device OS repeatedly will issue HTTP requests asynchronously to verify Internet transparency. Then, AutoConnect will trap its HTTP request and redirects it to the URL `/_ac`.

    b. A user will issue a request navigating the AutoConnect menu manually from a browser in the captive portal state. And one of them is the `Config New` menu that will trigger a WiFi scan.

    If we change the redirect destination of HTTP requests issued by the device OS to verify internet transparency from `/_ac` to `/_ac/config`, that means a WiFi scan triggered by `Config New` will transition the channel during an HTTP session, and it will induce a WiFi radio signal losing. It is desirable to separate the above two requirements to avoid WiFi radio channel conflicts.

    Therefore, it is desirable for AutoConnect to complete the active scan when the client device is not holding the captive portal. This is exactly what you pointed out.

2. How does AutoConnect know when to trigger an active scan

    AutoConnect will execute a WiFi scan inside AutoConenct::begin before it begins responding to the captive portal. This is equivalent to the "already scan for nearby SSIDs when or even before opening the captive portal?" as you mentioned.

    The result of a scan inside AutoConnect::begin will cover a page view of automatic pops up at 1st time. But how does AutoConnect know when to start the second and subsequent scans. AutoConnect can not know if the device OS has issued an HTTP request (which verifies transparency to the Internet).

    Another approach is to use asynchronous WiFi scanning. The `Config New` page always just references the WiFi scan result structure that the SDK has. This seemed to work fine with the AutoConnect prototype. However, it triggered an asynchronous WiFi scan every few seconds. As result, the browser in the captive portal state on the client device lost the session when the asynchronous scan came into collision with an HTTP request issued by the client device OS.

After all, I'd not come up with a way to know when to perform a WiFi scan safely and decided for an automatic pop-up page (which doesn't trigger a WiFi scan) in simple HTML.
If you have a solution to the problem, I would like to consider it.