Hieromon/AutoConnect #282: AUTOCONNECT_NOUSE_JSON compile option broken

TKTF50, 2020-12-04:
Hi Hieromon,

I'm glad to see a new release of your AutoConnect library. Thank you.

It looks like the `AUTOCONNECT_NOUSE_JSON` compile time option was broken in the latest release. I've been using this option to try to keep the AutoConnect build size a little smaller. When I compile (I'm using PlatformIO) I get these errors:

<pre>
`.pio\libdeps\esp07\AutoConnect\src\AutoConnectElementBasisImpl.h: In member function 'virtual const String AutoConnectInputBasis::toHTML() const':
.pio\libdeps\esp07\AutoConnect\src\AutoConnectElementBasisImpl.h:152:24: error: 'AUTOCONNECT_JSON_VALUE_NUMBER' was not declared in this scope
       applyType = PSTR(AUTOCONNECT_JSON_VALUE_NUMBER);
                        ^
.pio\libdeps\esp07\AutoConnect\src\AutoConnectElementBasisImpl.h:155:24: error: 'AUTOCONNECT_JSON_VALUE_PASSWORD' was not declared in this scope
       applyType = PSTR(AUTOCONNECT_JSON_VALUE_PASSWORD);
                        ^
.pio\libdeps\esp07\AutoConnect\src\AutoConnectElementBasisImpl.h:159:24: error: 'AUTOCONNECT_JSON_VALUE_TEXT' was not declared in this scope
       applyType = PSTR(AUTOCONNECT_JSON_VALUE_TEXT);
                        ^
Compiling .pio\build\esp07\lib4a8\AutoConnect\AutoConnectPage.cpp.o`
</pre>

Hieromon, 2020-12-05:
Hello TKTF50,

I was able to finally release your PR. Thank you for your contribution.

This issue you found is my fault. I missed the AUTOCONNECT_NOUSE_JSON condition from the validation case. I regret that it was a bug that I could find in one shot just by compiling, really sorry.
I will release the patch immediately.

Hieromon, 2020-12-05:
@TKTF50, I released [v1.2.1](https://github.com/Hieromon/AutoConnect/releases/tag/v1.2.1) to fix the issue.

TKTF50, 2020-12-05:
Hi @Hieromon,

Thanks for the patch. I updated my local copy and it is compiling correctly now.

As to my PR: I was happy to be able to make a contribution to your library.

closed 2020-12-05 by TKTF50