Hieromon/AutoConnect #284: OTA with firmware.bin not writing

brad-colbert, 2020-12-06:
I'm trying to use the default _ac and Update menu tab to update the firmware but it's not completing.  I'm not sure why.  I'm able to update files in SPIFFS using the Update option (thank you for adding that capability).  Anything I can do to debug?

Hieromon, 2020-12-06:
AutoConnectOTA has been fully tested and should work in a typical environment.
You can enable the [AC_DEBUG](https://hieromon.github.io/AutoConnect/faq.html#3-turn-on-the-debug-log-options) option and monitor what is happening inside AutoConnect with a serial monitor. We may be able to improve your situation with a more information.
Can you post the information that is needed for problem diagnose according to the [post items](https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository) in the FAQ of the document as clues?

brad-colbert, 2020-12-10:
Hello Hieromon.  My apologies for the late reply.

Details and update on status:
1. MELIFE ESP32 ESP-32S Development Board (WROOM 32) 4MB flash and Adafruit HUZZAH32 – ESP32
2. I'm using PlatformIO Core 5.0.3·Home 3.3.1 but I'm unsure which arduino packages it is using
3. MacOS Catalina 10.15.7
4. Smartphone... not really the issue
5. Linksys WRT1900AC chan auto
6. lwIP?
7. See start of this thread but more info below.
8. Stack dump.  I don't think it's crashing
9. Not really an issue with the sketch code, I don't think
10. Working on the debug messages.

My partition table looks like:
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x200000,
spiffs,   data, spiffs,  0x210000,0xF0000,

I don't think that platformIO is building the firmware.bin as LittleFS but it is building the data partition (named spiffs strangely enough) as LittleFS.

brad-colbert, 2020-12-10:
Crashes and reboots.  Here is the log:

brad-colbert, 2020-12-10:
Starting networking and HTTP server...
```
[AC] Current:Trebloc
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin().....................................................................................................time out IP:0.0.0.0
[AC] autoReconnect, 5 network(s) found, Trebloc(RECENT) loaded
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin(Trebloc,InYourFace1)..established IP:192.168.1.212
[AC] http server started
WiFi connected: 192.168.1.212 surfomatic
Starting mDNS...
Getting time from NTP servers...
[AC] s_rc placed on /_ac/update
[AC] cap placed on /_ac/update
[AC] bin placed on /_ac/update
[AC] update placed on /_ac/update
[AC] js placed on /_ac/update
[AC] bin placed on /_ac/update_act
[AC] result placed on /_ac/update_act
[AC] dvo placed on /_ac/update_act
[AC] rc placed on /_ac/update_act
[AC] dvc placed on /_ac/update_act
[AC] /_ac/update on hands
[AC] /_ac/update_act on hands
[AC] Host:,/_ac/update,generated:/_ac/update, allocated
[AC] Host:surfomatic.local,/_ac/update,already allocated
[AC] Host:surfomatic.local,/_ac/update_act,generated:/_ac/update_act, allocated
[AC] Host:surfomatic.local,/_ac/update_act,already allocated
[AC] Host:surfomatic.local,/_ac/update_act,already allocated
[AC] /_ac/update requests upload to /_ac/update_act
[AC] ACFile bin accepted firmware.bin, handler enabled
[AC] /firmware.bin updating start
[AC] Host:surfomatic.local,/_ac/update_act,already allocated
[AC] Host:surfomatic.local,/_ac/update_act,already allocated
[AC] Host:surfomatic.local,/_ac/update_act,already allocated
[AC] Host:surfomatic.local,/_ac/update_act,already allocated
[AC] Host:surfomatic.local,/_ac/update_act,already allocated
[AC] Host:surfomatic.local,/_ac/update_act,already allocated
abort() was called at PC 0x401623e5 on core 1

Backtrace: 0x4008c674:0x3ffb1890 0x4008c8a5:0x3ffb18b0 0x401623e5:0x3ffb18d0 0x40086e37:0x3ffb18f0 0x400f2085:0x3ffb1910 0x401852ee:0x3ffb1930 0x401854d5:0x3ffb1950 0x400e9f24:0x3ffb1970 0x400defa5:0x3ffb1990 0x4019a766:0x3ffb19d0 0x400e232e:0x3ffb19f0 0x400db34d:0x3ffb1a70 0x4019a03e:0x3ffb1a90 0x40182be1:0x3ffb1ab0 0x400d90ae:0x3ffb1ae0 0x400d9d11:0x3ffb1b10 0x400dac1d:0x3ffb1d00 0x400d8785:0x3ffb1f20 0x400de31b:0x3ffb1f70 0x400d3644:0x3ffb1f90 0x400f48d5:0x3ffb1fb0 0x40088dbd:0x3ffb1fd0

Rebooting...
```

Hieromon, 2020-12-10:
@brad-colbert Thank you for the log.
I added a test pattern with reference to your partition scheme and reproduced the issue.
However, the module you are using is an ESP32, so choosing a filesystem has no relation to this issue. The ESP32 core is still SPIFFS. Only ESP8266 is prompting the transition to LittleFS.

Analyzing the trace dump, it seems that ESP-IDF's std::regex_match is the cause. It's a new implementation part for the regular file upload to enhanced with v120. This seems to have caused the regression. Sorry.

I'll make a patch soon.

brad-colbert, 2020-12-10:
@Hieromon wow!  Thank you for the quick results.  Fantastic work, fantastic support!

Hieromon, 2020-12-12:
@brad-colbert 
I released the fix as [v.1.2.2](https://github.com/Hieromon/AutoConnect/releases/tag/v.1.2.2).
In this release, I changed the indication of the file extension that allows OTA from a regular expression to a fixed string. The default is still `.bin`, but you can't specify it with a regular expression by default. If you want to use a regular expression for the file extension to identify a firmware, need to define `AUTOCONNECT_UPLOAD_ASFIRMWARE_USE_REGEXP` macro. Please confirm the [documentation](https://hieromon.github.io/AutoConnect/otabrowser.html#regular-file-uploading-using-autoconnectotaenhanced-wv120).

However, it is highly recommended to leave it as a fixed string. I have taken a fairly tricky approach to solve this issue. The `std::regex` implemented by ESP-IDF is unsophisticated.

https://github.com/Hieromon/AutoConnect/blob/0d02b06a3b476371a9f23bae045176af8705fc4b/src/AutoConnectDefs.h#L271-L282

Also, the partition table you used may insufficient for OTA. It seems to have no app area to stage a new one.

brad-colbert, 2020-12-14:
Thank you for the quick response.

I still get an immediate crash:

Hieromon, 2020-12-14:
@brad-colbert 
Which partition table are you using?
The trace below you provided suggests that the crash was intentionally caused by the Arduino core. The Arduino core had check whether the writing address is safe prior to erasing the flash and determines that there is not the OTA staging area.

```
abort() was called at PC 0x40165a65 on core 1

Backtrace: 0x4008c674:0x3ffb1870 0x4008c8a5:0x3ffb1890 0x40153b8d:0x3ffb18b0 0x40086e37:0x3ffb18d0 0x400f1c41:0x3ffb18f0 0x40177e32:0x3ffb1910 0x40178015:0x3ffb1930 0x400ee524:0x3ffb1950 0x400dd229:0x3ffb1970 0x40193aa6:0x3ffb19b0 0x400e3c4a:0x3ffb19d0 0x400d9839:0x3ffb1a50 0x40193146:0x3ffb1a70 0x400d8631:0x3ffb1a90 0x400d5e22:0x3ffb1ac0 0x400d6ce9:0x3ffb1af0 0x400d7de2:0x3ffb1cf0 0x400d552d:0x3ffb1f20 0x400dc4bf:0x3ffb1f70 0x400d20c6:0x3ffb1f90 0x400f42a9:0x3ffb1fb0 0x40088dbd:0x3ffb1fd0
  #0  0x4008c674:0x3ffb1870 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x4008c8a5:0x3ffb1890 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x40153b8d:0x3ffb18b0 in is_safe_write_address at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:126
  #3  0x40086e37:0x3ffb18d0 in spi_flash_erase_sector at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:204 (discriminator 2)
  #4  0x400f1c41:0x3ffb18f0 in EspClass::flashEraseSector(unsigned int) at C:\Users\hieromon\.platformio\packages\framework-arduinoespressif32\cores\esp32/Esp.cpp:203
  #5  0x40177e32:0x3ffb1910 in UpdateClass::_writeBuffer() at C:\Users\hieromon\.platformio\packages\framework-arduinoespressif32\libraries\Update\src/Updater.cpp:360
  #6  0x40178015:0x3ffb1930 in UpdateClass::write(unsigned char*, unsigned int) at C:\Users\hieromon\.platformio\packages\framework-arduinoespressif32\libraries\Update\src/Updater.cpp:360
  #7  0x400ee524:0x3ffb1950 in AutoConnectOTA::_write(unsigned char const*, unsigned int) at C:\Users\hieromon\Documents\Arduino\libraries\AutoConnect\src/AutoConnectOTA.cpp:254
```

Maybe here:
`~/.platformio/packages/framework-arduinoespressif32/libraries/Update/src/Updater.cpp`

If you keep the partition definition that has posted yet, it is probably the cause of the crash. Strictly speaking, it's not a crash, the Update class (which is in the esp32 Arduino library) has interrupted the writing to the flash because there is no staging area for the OTA, it's an **intentional ABORT**.

I commented [above](https://github.com/Hieromon/AutoConnect/issues/284#issuecomment-744117291):

> Also, the partition table you used may insufficient for OTA. It seems to have no app area to stage a new one.

At least two app partitions are required for successful OTA. (usually app0 and app1)
Could you change the partition table to `default.csv` or `min_spiffs.csv` and try again?
(Please note that the v.1.2.2 fixed a bug found while diagnosing the traces you provided)

brad-colbert, 2020-12-14:
My partitions were the final problem.  Thank you very much.  Below is the partition table that I found to be successful:

```
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   , 0x180000,
app1,     app,  ota_1,   , 0x180000,
spiffs,   data, spiffs,  , 0xF0000,

```

Hieromon, 2020-12-15:
Glad to hear that. It solved.
Close.

closed 2020-12-15 by Hieromon